RIOT-OS/RIOT #16411: border_router: significant packet loss when sending out packets using USB cdc-ecm on nrf52

haukepetersen, 2021-04-28:
#### Description
This evening I setup a border router using a `nrf52840dongle` board. After I got the setup configured correctly, I noticed a significant packet loss when pinging between RIOT nodes and my Linux host. As far as I can tell, the packet loss happens between the border router RIOT node and the host on the wired upstream interface.

The border router is configure to use the USB `cdc-ecm` as upstream interface, while my lowpan network is running IP-over-BLE. But the results are independent from BLE, running the same node as IEEE802.15.4 border router (default configuration for the `gnrc_border_router`), the results are exactly the same.

Example pings from border router node to Linux host over the wired interface:
```
2021-04-28 20:29:47,694 # --- fe80::1 PING statistics ---
2021-04-28 20:29:47,694 # 200 packets transmitted, 162 packets received, 19% packet loss
2021-04-28 20:29:47,695 # round-trip min/avg/max = 0.611/0.657/1.125 ms

2021-04-28 20:41:35,283 # --- fd5f:6d79:a254:0:5d5a:5818:8db1:864d PING statistics ---
2021-04-28 20:41:35,283 # 100 packets transmitted, 74 packets received, 26% packet loss
2021-04-28 20:41:35,284 # round-trip min/avg/max = 0.629/0.827/1.200 ms

2021-04-28 20:50:23,611 # --- fd5f:6d79:a254:0:5d5a:5818:8db1:864d PING statistics ---
2021-04-28 20:50:23,612 # 200 packets transmitted, 161 packets received, 19% packet loss
2021-04-28 20:50:23,613 # round-trip min/avg/max = 0.625/0.668/1.046 ms
...
```
This picture repeats itself, independent from ping interval etc. Packet loss is always around 15%-25%...

The connectivity between the BR node and some other RIOT node is just fine:
```
2021-04-28 20:29:18,991 # --- fd23:affe:0:2:dbe4:f4ff:fec2:21da PING statistics ---
2021-04-28 20:29:18,997 # 200 packets transmitted, 200 packets received, 0% packet loss
2021-04-28 20:29:19,001 # round-trip min/avg/max = 19.836/39.221/70.505 ms
```

Also pinging the border router node from Linux seems not to be impacted:
```
--- fd23:affe:0:2:dbe4:f4ff:fec2:21da ping statistics ---
60 packets transmitted, 60 received, 0% packet loss, time 59643ms
rtt min/avg/max/mdev = 0.776/0.996/1.848/0.227 ms
```

Did anyone notice this behavior before? Or does anyone have any idea what could be causing this?

#### Steps to reproduce the issue
Flash `examples/gnrc_border_router` with the following settings:
```
BOARD=nrf52840dongle STATIC_ROUTES=1 UPLINK=cdc-ecm make ...
```
Then simply ping your linux host, doesn't matter if you use link local or global addresses...

#### Expected results
No packet loss when pinging, we are talking about a fast, wired interface after all...

#### Actual results
Between 15% and 25% packet loss when pinging, independent of the ping interval etc.

#### Versions
- RIOT Master
- LInux Mint 20.1
- arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103

haukepetersen, 2021-04-29:
Additional finding: it seems that the ICMP echo requests all arrive at the Linux host and are replied properly:
RIOT node:
```
2021-04-29 10:18:51,166 # 12 bytes from fd5f:6d79:a254:0:5b49:8eff:4aad:7ad7: icmp_seq=187 ttl=64 time=0.675 ms
2021-04-29 10:18:51,366 # 12 bytes from fd5f:6d79:a254:0:5b49:8eff:4aad:7ad7: icmp_seq=191 ttl=64 time=0.672 ms
2021-04-29 10:18:51,416 # 12 bytes from fd5f:6d79:a254:0:5b49:8eff:4aad:7ad7: icmp_seq=192 ttl=64 time=0.658 ms
2021-04-29 10:18:51,466 # 12 bytes from fd5f:6d79:a254:0:5b49:8eff:4aad:7ad7: icmp_seq=193 ttl=64 time=0.633 ms
2021-04-29 10:18:51,516 # 12 bytes from fd5f:6d79:a254:0:5b49:8eff:4aad:7ad7: icmp_seq=194 ttl=64 time=0.649 ms
2021-04-29 10:18:51,566 # 12 bytes from fd5f:6d79:a254:0:5b49:8eff:4aad:7ad7: icmp_seq=195 ttl=64 time=0.645 ms
2021-04-29 10:18:51,616 # 12 bytes from fd5f:6d79:a254:0:5b49:8eff:4aad:7ad7: icmp_seq=196 ttl=64 time=0.660 ms
2021-04-29 10:18:51,716 # 12 bytes from fd5f:6d79:a254:0:5b49:8eff:4aad:7ad7: icmp_seq=198 ttl=64 time=0.657 ms
2021-04-29 10:18:51,766 # 12 bytes from fd5f:6d79:a254:0:5b49:8eff:4aad:7ad7: icmp_seq=199 ttl=64 time=0.657 ms
```
-> seq 188-190 and 197 are missing

Linux host for the same sequence:
```
10:18:51.164936 IP6 fe80::e465:e9ff:fe0e:b10 > schelm: ICMP6, echo request, seq 187, length 12
10:18:51.164948 IP6 schelm > fe80::e465:e9ff:fe0e:b10: ICMP6, echo reply, seq 187, length 12
10:18:51.214942 IP6 fe80::e465:e9ff:fe0e:b10 > schelm: ICMP6, echo request, seq 188, length 12
10:18:51.214954 IP6 schelm > fe80::e465:e9ff:fe0e:b10: ICMP6, echo reply, seq 188, length 12
10:18:51.264977 IP6 fe80::e465:e9ff:fe0e:b10 > schelm: ICMP6, echo request, seq 189, length 12
10:18:51.264989 IP6 schelm > fe80::e465:e9ff:fe0e:b10: ICMP6, echo reply, seq 189, length 12
10:18:51.315011 IP6 fe80::e465:e9ff:fe0e:b10 > schelm: ICMP6, echo request, seq 190, length 12
10:18:51.315023 IP6 schelm > fe80::e465:e9ff:fe0e:b10: ICMP6, echo reply, seq 190, length 12
10:18:51.365041 IP6 fe80::e465:e9ff:fe0e:b10 > schelm: ICMP6, echo request, seq 191, length 12
10:18:51.365053 IP6 schelm > fe80::e465:e9ff:fe0e:b10: ICMP6, echo reply, seq 191, length 12
10:18:51.415079 IP6 fe80::e465:e9ff:fe0e:b10 > schelm: ICMP6, echo request, seq 192, length 12
10:18:51.415090 IP6 schelm > fe80::e465:e9ff:fe0e:b10: ICMP6, echo reply, seq 192, length 12
10:18:51.465113 IP6 fe80::e465:e9ff:fe0e:b10 > schelm: ICMP6, echo request, seq 193, length 12
10:18:51.465125 IP6 schelm > fe80::e465:e9ff:fe0e:b10: ICMP6, echo reply, seq 193, length 12
10:18:51.515138 IP6 fe80::e465:e9ff:fe0e:b10 > schelm: ICMP6, echo request, seq 194, length 12
10:18:51.515150 IP6 schelm > fe80::e465:e9ff:fe0e:b10: ICMP6, echo reply, seq 194, length 12
10:18:51.565168 IP6 fe80::e465:e9ff:fe0e:b10 > schelm: ICMP6, echo request, seq 195, length 12
10:18:51.565179 IP6 schelm > fe80::e465:e9ff:fe0e:b10: ICMP6, echo reply, seq 195, length 12
10:18:51.615212 IP6 fe80::e465:e9ff:fe0e:b10 > schelm: ICMP6, echo request, seq 196, length 12
10:18:51.615223 IP6 schelm > fe80::e465:e9ff:fe0e:b10: ICMP6, echo reply, seq 196, length 12
10:18:51.665243 IP6 fe80::e465:e9ff:fe0e:b10 > schelm: ICMP6, echo request, seq 197, length 12
10:18:51.665255 IP6 schelm > fe80::e465:e9ff:fe0e:b10: ICMP6, echo reply, seq 197, length 12
10:18:51.715273 IP6 fe80::e465:e9ff:fe0e:b10 > schelm: ICMP6, echo request, seq 198, length 12
10:18:51.715284 IP6 schelm > fe80::e465:e9ff:fe0e:b10: ICMP6, echo reply, seq 198, length 12
10:18:51.765309 IP6 fe80::e465:e9ff:fe0e:b10 > schelm: ICMP6, echo request, seq 199, length 12
10:18:51.765321 IP6 schelm > fe80::e465:e9ff:fe0e:b10: ICMP6, echo reply, seq 199, length 12
````
-> all packets arrive and responses seem to be properly sent.

So the problem seems to be on the receiving end...

haukepetersen, 2021-04-29:
Played around some more, but now I am stuck: I am running `examples/gnrc_networking` on a `nrf52840dongle`, only adding the modules `auto_init_usbus` and `usbus_cdc_ecm`. On top, I run a simple UDP echo server on my linux host. When sending a number of UDP packets to this echo server, I observe the same behavior then with pinging the host: all UDP packets arrive just fine at the echo server and responses are sent out fine by the host. However, a random? number of these UDP packets is not received correctly by the RIOT node.

Printing the number of received bytes in `_store_frame_chunk()` in `sys/usb/usbus/cdc/ecm/cdc_ecm.c` reveled that (randomly?) incoming network packets are reported to be too short. In my example, I am sending 66 byte UDP packets from the RIOT node to a UDP echo server running on my host machine. For the response packets that are received OK, I get 
```
2021-04-29 11:37:46,949 # _frame_chunk: in:64 len:0                                                                    
2021-04-29 11:37:46,949 # _frame_chunk: in:2 len:64
--> totaling the expected 66 bytes
```
But for some response packets I get:
```
2021-04-29 11:37:47,950 # _frame_chunk: in:2 len:0                                                                     
2021-04-29 11:37:47,950 # _frame_chunk: in:2 len:2 
-> totalling just 4 bytes
```
These 4 bytes are then handed to the netdev layer, and of course dropped somewhere higher in the network stack... I can with certainty say, that these 4 byte packets are the UDP response packets sent by my UDP server.

So it seems to me, there is something buggy, either (i) deeper in the USB stack, or (ii) in the usbdev driver for the `nrf52`. On first sight I was not able to pin down this issue any further, but then I am not familiar with the USB code...

It would be great if someone with deeper knowledge of the USB code could take a closer look!

haukepetersen, 2021-04-29:
Found my `samr21-xpro` and decided to check if I can reproduce the issue on that platform. For this I tried to configure the `samr21-xpro` as close to the `nrf52840dongle` as I can, so also using `stdio_cdc_acm` for the shell (build with `BOARD=samr21-xpro USEMODULE="auto_init_usbus usbus_cdc_ecm stdio_cdc_acm"`).

Turns out the faulty behavior does not show on this board:
```
2021-04-29 12:22:35,489 # --- fe80::1 PING statistics ---
2021-04-29 12:22:35,490 # 2000 packets transmitted, 2000 packets received, 0% packet loss
2021-04-29 12:22:35,490 # round-trip min/avg/max = 0.855/0.902/1.008 ms
```

-> So my, by now informed, guess would be that the problem lays somewhere in the `nrf52` usbus driver?!

fjmolinas, 2021-08-03:
I'm trying to reproduce this but seem to be unable, I'm using a `nrf52840-mdk-dongle` and when pinging every 100ms I get <1% packet loss (pinging over 1hop), note that for this test I'm using `uhcp`.

```
sudo ping6 -i .1 -c 1000 2001:db8::d82a:b8ff:fee0:ab50
...
...
64 bytes from 2001:db8::d82a:b8ff:fee0:ab50: icmp_seq=996 ttl=63 time=36.6 ms
64 bytes from 2001:db8::d82a:b8ff:fee0:ab50: icmp_seq=997 ttl=63 time=83.3 ms
64 bytes from 2001:db8::d82a:b8ff:fee0:ab50: icmp_seq=998 ttl=63 time=207 ms
64 bytes from 2001:db8::d82a:b8ff:fee0:ab50: icmp_seq=999 ttl=63 time=106 ms
64 bytes from 2001:db8::d82a:b8ff:fee0:ab50: icmp_seq=1000 ttl=63 time=150 ms

--- 2001:db8::d82a:b8ff:fee0:ab50 ping statistics ---
1000 packets transmitted, 998 received, 0.2% packet loss, time 101615ms
rtt min/avg/max/mdev = 7.634/96.986/227.568/48.565 ms, pipe 3
```

haukepetersen, 2021-08-04:
Interesting. I will try this using a different nrf board again later today, would be great if its 'just' a hardware issue :-)

haukepetersen, 2021-08-04:
Wait, you pinged the RIOT BR from the Linux host, right? Could you maybe try to ping the Linux host from the RIOT node, because this is where I noticed the packet loss...

fjmolinas, 2021-08-04:
> Wait, you pinged the RIOT BR from the Linux host, right? Could you maybe try to ping the Linux host from the RIOT node, because this is where I noticed the packet loss...

Yes, I'll try that as well, maybe you were using a specific payload size?

Pinging from the BR to the Linux host:

```
ping6 fd00:dead:beef::1 -i 100 -c 100
2021-08-04 10:00:19,530 # ping6 fd00:dead:beef::1 -i 100 -c 100
2021-08-04 10:00:19,654 # 12 bytes from fd00:dead:beef::1: icmp_seq=0 ttl=63 time=118.583 ms
...
2021-08-04 09:58:10,503 # 12 bytes from fd00:dead:beef::1: icmp_seq=98 ttl=64 time=0.722 ms
2021-08-04 09:58:10,603 # 12 bytes from fd00:dead:beef::1: icmp_seq=99 ttl=64 time=0.747 ms
2021-08-04 09:58:10,603 # 
2021-08-04 09:58:10,604 # --- fd00:dead:beef::1 PING statistics ---
2021-08-04 09:58:10,604 # 100 packets transmitted, 100 packets received, 0% packet loss
2021-08-04 09:58:10,604 # round-trip min/avg/max = 0.667/0.758/0.941 ms
```

Pinging over one hop to the Linux host:

```
ping6 fd00:dead:beef::1 -i 100 -c 100
2021-08-04 10:00:19,530 # ping6 fd00:dead:beef::1 -i 100 -c 100
...
2021-08-04 10:00:29,553 # 12 bytes from fd00:dead:beef::1: icmp_seq=98 ttl=63 time=220.195 ms
2021-08-04 10:00:29,628 # 12 bytes from fd00:dead:beef::1: icmp_seq=99 ttl=63 time=195.449 ms
2021-08-04 10:00:29,628 # 
2021-08-04 10:00:29,632 # --- fd00:dead:beef::1 PING statistics ---
2021-08-04 10:00:29,637 # 100 packets transmitted, 100 packets received, 0% packet loss
2021-08-04 10:00:29,642 # round-trip min/avg/max = 29.362/134.784/220.525 ms
```

fjmolinas, 2021-08-04:
I also tried with a `nrf52840dongle` clone, the performance is similar.

fjmolinas, 2021-08-04:
*note latest test was actually with DHCP

haukepetersen, 2021-08-04:
I assume you use the current master?

haukepetersen, 2021-08-05:
I did some testing using a `nrf52840dongle` and RIOT master on my office host, and indeed I too can't reproduce this issue anymore. At the same time I run some tests on my already deployed RIOT BR in my home network (running RIOT masters from ~Apr to May), and there I can still see this issue. So there is a good chance that this issue has resolved itself implicitly by changes in master. That would be great.

I will run some final tests in my home office environment where I initially discovered this issue. I am hopeful :-)

haukepetersen, 2021-08-06:
The issues seems indeed to at least have improved in current master. I ran some ping tests using the exact environment and hardware that I used when I discovered the issue above, and the results are much better :-)

Did a number of ping runs and all look identical, right like they should:
```
2021-08-06 10:14:37,183 # ping ff02::1%9 -c 2000 -i 100
...
2021-08-06 10:17:55,280 # --- ff02::1 PING statistics ---                                                              
2021-08-06 10:17:55,281 # 2000 packets transmitted, 2000 packets received, 0% packet loss             
2021-08-06 10:17:55,282 # round-trip min/avg/max = 0.701/1.282/2.300 ms 
```

As none of us is able to reproduce the faulty behavior anymore, I consider this issue solved.

closed 2021-08-06 by haukepetersen

miri64, 2021-08-06:
If someone wants to be super-diligent it would be nice, however, if the fix could be bisected, so we know what did the trick ;-).

fjmolinas, 2021-08-06:
> I did some testing using a `nrf52840dongle` and RIOT master on my office host, and indeed I too can't reproduce this issue anymore. At the same time I run some tests on my already deployed RIOT BR [...]

Do you maybe have a faulty commit? I checked quickly 2021.01 and 2021.04 and it was working.

Ivrawn, 2025-03-05:
Hi there!

I wonder how you made the ping work from your Dongle to the Linux machine, since I am struggling with that issue in a similar scenario. Specifically, I have a nRF52840DK board running gnrc_networking and the Dongle running gnrc_border_router. I get then a global IPv6 address for both of them from the prefix of a new interface in my PC generated by the border router. However, although I can ping between the board and the dongle using their global addresses, I cannot ping from the Dongle to my machine (I mean, to the address of the interface generated).

Thanks in advance :)