RIOT-OS/RIOT #16413: ztimer: Removal with reporting

chrysn, 2021-04-29:
#### Description

Whenever something is passed in to a timer that's supposed to be called exactly once (esp. everything that `free`s up its own resources), pulling that out early is tricky.

Since the [introduction of `ztimer_is_set`](https://github.com/RIOT-OS/RIOT/pull/15780) this is easier, but the necessary dance is

* disable interrupts
* query ztimer_is_set
* remove ztimer
* enable interrupts
* run the callback if it was actually set

which is not immediately obvious, and likely overkill in terms of interrupt disabling (given that all ztimer functions need to do that too).

AIU this pattern (as well as any integrated form of it that could come of this issue) has the slight quirk that a result of "it was not set" would *usually* mean it fired, but could *also* mean that it was fired but did not execute (or did not execute to completion). That's OK for how I'd use it, but needs to be made very clear in the documentation.

The alternative to admitting this possibility would be to make reporting removal unavailable from interrupts -- and that wouldn't be so great.

#### Roads towards fixing this

* Just make ztimer_remove return whether it did remove something
  * void -> bool API change -- is that OK?
  * Does this increase any constrained parameter? (I have the vague hope that the `_is_set` result is stored on the stack anyway and that returning it just means it gets put into a different register, but there'd be nontrivial comparisons of assembly involved to check whether that's the case)
* Introduce a `ztimer_remove_checking` (better name?) function that does ztimer_is_set and remove in a single efficient step

In the course of this I'd also like to look at the use cases for `is_set` -- I have a gut feeling that there may be cases when users are tempted to use it that are racy, and a "checking removal" would be the better thing to do.

#### Useful links

* https://github.com/RIOT-OS/RIOT/issues/14390 would be very hard to resolve without this (it could be resolved now that ztimer_is_set exists ... but hey it'll need porting over to ztimer in the first place)
* http://rustdoc.etonomy.org/riot_wrappers/ztimer/struct.ZTimer.html could offer something useful here (again, it's possible now with ztimer_is_set but why do it the complex way). Rust *really* insists that if a callback is "call once only", then it be called once only, and for some setups *exactly once* is a requirement punishable by UB.

kaspar030, 2021-04-29:
> In the course of this I'd also like to look at the use cases for `is_set` -- I have a gut feeling that there may be cases when users are tempted to use it that are racy, and a "checking removal" would be the better thing to do.

I didn't want to initially expose this for that reason.

We should also consider that the current implementation fails if

- the timer was set, but not set on the specified clock (as it might have a next pointer) -> false positive
- the timer was not set and the struct not cleaned (so random next pointer) -> possible false positive

for the ztimer_remove() case, returning the value of ìs_set()` might thus not tell the truth, if used incorrectly (meaning, on a timer that was previously set on that exact clock). We'd need to use the to-be-added bool return of `_del_entry_from_list(clock, &timer->base)`.

> `ztimer_remove_checking()`

I'd rather make `ztimer_remove()` return bool.

maribu, 2021-04-29:
Maybe making a `ztimer_check_api_use` (pseudo-)module that also stores the clock within a `ztimer_t` it was added to an added `expect(clock == ztimer->clock)` would make sense?

I don't really think that adding unconditionally the overhead in terms of both RAM and CPU cycles is justified by checking for API misuse and broken code. But being able to add it "on demand" could be pretty helpful. We might even consider `ztimer_check_api_use` (or whatever name is used) to become a default module with develhelp enabled.

closed 2021-12-17 by chrysn