RIOT-OS/RIOT #16417: w5100 driver's improvement

liyue75, 2021-04-30:
Last time, I found a bug in w5100  #15447, that will cause wrong pointer position when its tx ring buffer is full and cause wrong data to be sent.

I found there is anoter problem in drivers/w5100/w5100.c, in **send** function:
```
static int send(netdev_t *netdev, const iolist_t *iolist)
{
    w5100_t *dev = (w5100_t *)netdev;
    int sum = 0;
    spi_acquire(dev->dev->p.spi, dev->p.cs, SPI_CONF, dev->p.clk);
    uint16_t pos = raddr(dev, S0_TX_WR0, S0_TX_WR1);

    if (pos == 0) {
        pos = S0_TX_BASE;
    }

    for (const iolist_t *iol = iolist; iol; iol = iol->iol_next) {
        size_t len = iol->iol_len;
        pos = tx_upload(dev, pos, iol->iol_base, len);
        sum += len;
    }

    waddr(dev, S0_TX_WR0, S0_TX_WR1, pos);
     ......
```
should be replaced by
```
static int send(netdev_t *netdev, const iolist_t *iolist)
{
    w5100_t *dev = (w5100_t *)netdev;
    int sum = 0;
    spi_acquire(dev->dev->p.spi, dev->p.cs, SPI_CONF, dev->p.clk);
    uint16_t tx_wr = raddr(dev, S0_TX_WR0, S0_TX_WR1);
    uint16_t pos = (tx_wr & S0_MASK) + S0_TX_BASE;

    if (pos == 0) {
        pos = S0_TX_BASE;
    }

    for (const iolist_t *iol = iolist; iol; iol = iol->iol_next) {
        size_t len = iol->iol_len;
        pos = tx_upload(dev, pos, iol->iol_base, len);
        sum += len;
    }

    waddr(dev, S0_TX_WR0, S0_TX_WR1, tx_wr + sum);
     ......
```
Values in TX_WR register should be in range of 0~0xFFFF, if I don't modify this send function,
it will cause the value of TX_WR to vary between 0~0x2000.
This bug will cause about 5ms delay every time the send ring buffer is full.

fjmolinas, 2021-04-30:
Hi @liyue75 thanks for catching this, do you think you can open a PR with the proposed fix?

liyue75, 2021-04-30:
@fjmolinas could you please  help me to do that?

fjmolinas, 2021-04-30:
Sure, I would suggest first going through https://github.com/RIOT-OS/RIOT/blob/master/CONTRIBUTING.md#contributing-code, it gives a nice overview of the contributing process. Is there anything specific you want or need help with?

fjmolinas, 2021-04-30:
ups wrong button

liyue75, 2021-04-30:
@fjmolinas thank you.

fjmolinas, 2021-05-06:
Hi @liyue75 any updates on this?

liyue75, 2021-05-10:
@fjmolinas Hi, when I running static-test, an error seems to occur:
`Required command 'spatch' for './dist/tools/coccinelle/check.sh' not found in PATH *`

jeandudey, 2021-05-14:
> @fjmolinas Hi, when I running static-test, an error seems to occur:
> `Required command 'spatch' for './dist/tools/coccinelle/check.sh' not found in PATH *`

You need `coccinelle` to be installed on your system

PR #16474 (2021-10-04): drivers/w5100/w5100.c: fix TX_WR register's value

### Contribution description

drivers/w5100/w5100.c
fix w5100's TX_WR register's value. Let the value exceed 0x2000.

### Testing procedure

### Issues/PRs references

Fixes #16417
closed 2021-10-04 by maribu