RIOT-OS/RIOT #16419: SX127x Join procedure failed (The Things Network)

JaccoVeldscholten, 2021-04-30:
#### Description
Unable to join the The Things Network (TTN) [https://console.thethingsnetwork.org/](url) with SX1276 Chip. Using default Lorawan Example.
Development board: Nucleo-l452re.
The communication between the SX1276 is working fine. Tested The following code: https://github.com/RIOT-OS/RIOT/tree/master/tests/driver_sx127x with HackRF and i was able to see the chirp signal. On the logic analyzer the communication with SPI Is fine.

#### Steps to reproduce the issue
Using the following example code: https://github.com/RIOT-OS/RIOT/tree/master/examples/lorawan
Using DevEUI / AppEUI / AppKey from TTN filled in makefile in BIG-Endian
[image]

Pin setup from SX1276 to Nucleo-l452re
MISO -> PA6
MOSI -> PA7
SCK -> PA5
NSS -> PA10
RESET -> PA11
DIO0 -> PA12
DIO1 -> PA13
GND -> GND
VCC -> 3V3

#### Expected results
Expected Join Request to be succesfull

#### Actual results
Failed Join Procedure
[image]

#### Versions
ARM Toolchain with Latest RIOT OS Repo Clone

iseatea, 2021-05-03:
为什么你是同性恋。**SX127x** 去他妈的你自己 nucleo 一堆菜鸟 
兵兵奉

iseatea, 2021-05-03:
中国共产党永远赢

miri64, 2021-05-03:
@iseatea please use English on this platform. Furthermore, your comments do not align with our code of conduct, so I deleted them. Further such comments will not be tolerated.

akshaim, 2021-05-03:
Hey @JaccoVeldscholten . Can you try `ABP` method using this [example](https://github.com/RIOT-OS/RIOT/tree/master/examples/gnrc_lorawan) ?

JaccoVeldscholten, 2021-05-03:
> Hey @JaccoVeldscholten . Can you try `ABP` method using this [example](https://github.com/RIOT-OS/RIOT/tree/master/examples/gnrc_lorawan) ?

Im getting up-link but when im trying to run IFconfig im receiving 0 gateways. A few blocks away there is an gateway for TTN. And im outside....

```
)> ifconfig 3
ifconfig 3
Iface  3  HWaddr: 26:01:1E:EA  Frequency: 88299987Hz  RSSI: 99  BW: 125kHz  SF: 12  CR: 4/5  Link: up 
           TX-Power: 14dBm  State: TX  Demod margin.: 0  Num gateways.: 0 
          L2-PDU:2286  
```

So the device will get in Up-link method but is not receiving any gateways nearby? That might be also the problem that OTAA is not working? But its kinda strange, my arduino module that im heaving here (Heltec LoRa V2) is working fine.

akshaim, 2021-05-04:
> Im getting up-link but when im trying to run IFconfig im receiving 0 gateways. A few blocks away there is an gateway for TTN. And im outside....

You may enable linkcheck `ifconfig 3 link_check ` and then do an uplink.

>  But its kinda strange, my arduino module that im heaving here (Heltec LoRa V2) is working fine.

Can you try the same example using OTAA ? 

OTAA needs RX which further requires correct mapping of DIO1. Also, while performing OTAA do you see the **join request and join accept** in the TTN console?

iseatea, 2021-05-10:
> @iseatea please use English on this platform. Furthermore, your comments do not align with our code of conduct, so I deleted them. Further such comments will not be tolerated.

issa sorry

JaccoVeldscholten, 2021-05-10:
> You may enable linkcheck `ifconfig 3 link_check ` and then do an uplink.
>
>
> Can you try the same example using OTAA ?
>
> OTAA needs RX which further requires correct mapping of DIO1. Also, while [...]

After some days i purchased my own Gateway for LoRa and binded with TTN. Currently im seeing that the device is trying to join. TTN Approves, but than the Join failed.

**Activation TTN**
[image]
**Gateway Info TTN**
[image]

akshaim, 2021-05-12:
Did you check the `DIOx` connections ? Also, are the pins for `sx1276` configured correctly in board.h ?

```
#define SX127X_PARAM_RESET                  GPIO_PIN(PORT_x, PIN_NO)
#define SX127X_PARAM_DIO0                   GPIO_PIN(PORT_x, PIN_NO)
#define SX127X_PARAM_DIO1                   GPIO_PIN(PORT_x, PIN_NO)
#define SX127X_PARAM_DIO2                   GPIO_PIN(PORT_x, PIN_NO)
#define SX127X_PARAM_DIO3                   GPIO_PIN(PORT_x, PIN_NO)
```

JaccoVeldscholten, 2021-05-14:
> Did you check the `DIOx` connections ? Also, are the pins for `sx1276` configured correctly in board.h ?
>
> ```
> #define SX127X_PARAM_RESET                  GPIO_PIN(PORT_x, PIN_NO)
> #define [...]

Yes this should be good. 
I followed the folling instructions from: https://github.com/RIOT-OS/RIOT/blob/master/drivers/sx127x/include/sx127x_params.h
#define SX127X_PARAM_SPI_NSS                GPIO_PIN(1, 6)       /* D10 */
#define SX127X_PARAM_RESET                  GPIO_PIN(0, 0)       /* A0 */
#define SX127X_PARAM_DIO0                   GPIO_PIN(0, 10)      /* D2 */
#define SX127X_PARAM_DIO1                   GPIO_PIN(1, 3)       /* D3 */
#define SX127X_PARAM_DIO2                   GPIO_PIN(1, 5)       /* D4 */

JaccoVeldscholten, 2021-05-16:
@akshaim 

The pinout should be correct. I followed the Arduino connector information in the manual:
https://www.st.com/resource/en/user_manual/dm00105823-stm32-nucleo64-boards-mb1136-stmicroelectronics.pdf

And from the RIOT: https://github.com/RIOTOS/RIOT/blob/73ccd1e2e721bee38f958f8906ac32e5e1fceb0c/drivers/sx127x/include/sx127x_params.h

Im getting DIO1 / DIO2 Unkown state issues when im outside.

jia200x, 2021-05-17:
Hi @JaccoVeldscholten 
> Steps i took in GNRC (Keys stripped):

This won't work if the device is not receiving downlink packets, since this information is carried there. In order to receive link_checks, the device must be associated. 

> After some days i purchased my own Gateway for LoRa and binded with TTN. Currently im seeing that the device is trying to join. TTN Approves, but than the Join failed.

Does TTN receive uplinks when using ABP? And downlinks? (For TTN the rx2_dr must be configured to DR3, otherwise you won't receive downlinks). With GNRC LoRaWAN, you can run the following:
```
ifconfig 3 set addr <DEVADDR>
ifconfig 3 set nwkskey <NWKSKEY>
ifconfig 3 set appskey <APPKSKEY>
ifconfig 3 set rx2_dr 3
ifconfig 3 up

/* ifconfig wil show "Link: Up" because there's no handshake using ABP and the device is assumed to be joined */
```
Then try to schedule uplinks and downlinks from the network server.
```
E.g 
send 3 Test
```

This could reveal quite some useful information in order to track down the issue, so please let us know how this goes.

JaccoVeldscholten, 2021-05-17:
> Hi @JaccoVeldscholten
>
>
> This won't work if the device is not receiving downlink packets, since this information is carried there. In order to receive link_checks, the device must be associated. [...]

Thanks for responding. My gateway is not showing the data. I setted up 2 gateways and i verified that they are working with Heltec V2 LoRa module. 
[image]

With Heltec OTAA:
[image]

jia200x, 2021-05-17:
> I have to note that i got ABP worked sometimes. Few days it sended Hello from RIOT! Randomly to the TTN console. But after rebooting the device it stopped.

It seems to be a problem with the transmission power. Could you try the first proposal [here](https://forum.riot-os.org/t/example-lorawan-on-bluepill-board/3183/29)? (Setting PA_BOOST).

JaccoVeldscholten, 2021-05-17:
> It seems to be a problem with the transmission power. Could you try the first proposal [here](https://forum.riot-os.org/t/example-lorawan-on-bluepill-board/3183/29)? (Setting PA_BOOST).

Thanks for helping. I tried to set PA_BOOST in common/nucleo64/board.h but it didn't seem to work :(

jia200x, 2021-05-17:
> Update:
After trying to rejoin and send stuff i got again some data in the console! (ABP)

It's something!

Could you send me transmissions stats? (SNR, RSSI, etc). You should be able to see that details if you click the message.

> Trying to join with OTAA. But the gateway is to slow to send out the Join Accept. Is there a way to extend the RX ?

Yes, this is possible:
```
/**
 * @defgroup net_gnrc_lorawan_conf GNRC LoRaWAN compile configurations
 * @ingroup net_gnrc_conf
 * @{
 */
/**
 * @brief maximum timer drift in per mille
 *
 * @note this is only a workaround to compensate inaccurate timers.
 *
 * E.g a value of 1 means there's a positive drift of 0.1% (set timeout to
 * 1000 ms => triggers after 1001 ms)
 */
#ifndef CONFIG_GNRC_LORAWAN_TIMER_DRIFT
#define CONFIG_GNRC_LORAWAN_TIMER_DRIFT 10
#endif

/**
 * @brief the minimum symbols to detect a LoRa preamble
 */
#ifndef CONFIG_GNRC_LORAWAN_MIN_SYMBOLS_TIMEOUT
#define CONFIG_GNRC_LORAWAN_MIN_SYMBOLS_TIMEOUT 30
#endif
/** @} */
```

Try to change CONFIG_GNRC_LORAWAN_TIMER_DRIFT in order to adjust the drift. You can extend the RX window time with  CONFIG_GNRC_LORAWAN_MIN_SYMBOLS_TIMEOUT.

In any case, if there's a timing issue, joining with a lower DR shouldn't affect the timings. Could you try joining e.g with DR0?
```
ifconfig 3 set dr 0
ifconfig 3 set deveui <DEVEUI>
ifconfig 3 set appeui <APPEUI>
ifconfig 3 set appkey <APPKEY>
ifconfig 3 up
/* Wait 7 seconds */
ifconfig /*<-- should show Link: Up */
```
```

JaccoVeldscholten, 2021-05-17:
> It's something!
>
> Could you send me transmissions stats? (SNR, RSSI, etc). You should be able to see that details if you click the message.
>
>
> Yes, this is possible:
>
> ```
> /**
>  * @defgroup [...]

Many many many many many thanks. It works! You're are hero mate!

closed 2021-05-17 by JaccoVeldscholten

jia200x, 2021-05-17:
Good to hear!! Glad it worked.

Let us know if you have more questions.

Cheers and have fun!