RIOT-OS/RIOT #16451: sam0_eth: extremely long time to RX (frames stuck in buffer?)

benpicco, 2021-05-05:
#### Description
If `same54-xpro` is connected to a network with some random (broadcast?) network traffic, after a while it will get very slow to respond to incoming frames. 

#### Steps to reproduce the issue
Flash `examples/gnrc_networking` onto the `same54-xpro`

```
2021-05-05 21:49:59,921 # Iface  5  HWaddr: FC:C2:3D:0D:2D:1F 
2021-05-05 21:49:59,925 #           L2-PDU:1500  MTU:1500  HL:64  RTR  
2021-05-05 21:49:59,927 #           RTR_ADV  
2021-05-05 21:49:59,930 #           Source address length: 6
2021-05-05 21:49:59,932 #           Link type: wired
2021-05-05 21:49:59,938 #           inet6 addr: fe80::fec2:3dff:fe0d:2d1f  scope: link  VAL
2021-05-05 21:49:59,941 #           inet6 group: ff02::2
2021-05-05 21:49:59,943 #           inet6 group: ff02::1
2021-05-05 21:49:59,947 #           inet6 group: ff02::1:ff0d:2d1f
2021-05-05 21:49:59,948 #           
2021-05-05 21:49:59,951 #           Statistics for Layer 2
2021-05-05 21:49:59,955 #             RX packets 49273  bytes 5847245
2021-05-05 21:49:59,960 #             TX packets 18032 (Multicast: 15)  bytes 2129894
2021-05-05 21:49:59,963 #             TX succeeded 18032 errors 0
2021-05-05 21:49:59,966 #           Statistics for IPv6
2021-05-05 21:49:59,970 #             RX packets 18838  bytes 1962063
2021-05-05 21:49:59,975 #             TX packets 18032 (Multicast: 15)  bytes 1877446
2021-05-05 21:49:59,978 #             TX succeeded 18032 errors 0
```

ping the link-local address from a Linux machine

```
ping fe80::fec2:3dff:fe0d:2d1f%eno1
PING fe80::fec2:3dff:fe0d:2d1f%eno1(fe80::fec2:3dff:fe0d:2d1f%eno1) 56 data bytes
…
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=151 ttl=64 time=0.668 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=152 ttl=64 time=0.329 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=153 ttl=64 time=0.324 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=154 ttl=64 time=0.312 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=155 ttl=64 time=0.338 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=156 ttl=64 time=0.321 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=157 ttl=64 time=0.309 ms
```

Now flood the network with broadcast traffic

```
sudo ping -f ff02::1%eno1
```

#### Expected results

We might see some disruption, but the when ending the broadcast ping, everything should return to normal.

#### Actual results

The broadcast storm triggers an issue where frames get seemingly stuck in the RX buffer of the ethernet peripheral.
They are delivered to the applications with a delay of several seconds.

```
4 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=178 ttl=64 time=0.319 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=179 ttl=64 time=0.315 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=180 ttl=64 time=11.7 ms # broadcast storm starts
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=181 ttl=64 time=7.07 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=182 ttl=64 time=3.52 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=183 ttl=64 time=95.4 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=184 ttl=64 time=4091 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=185 ttl=64 time=4128 ms # broadcast storm ends
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=186 ttl=64 time=4070 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=188 ttl=64 time=3985 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=189 ttl=64 time=4032 ms
…
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=312 ttl=64 time=5081 ms # device never recoveres
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=313 ttl=64 time=5955 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=314 ttl=64 time=5952 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=315 ttl=64 time=6246 ms
64 bytes from fe80::fec2:3dff:fe0d:2d1f%eno1: icmp_seq=316 ttl=64 time=6053 ms
```

packets sent from RIOT with `udp send` are delivered instantly, but packets sent with

    nc -u -6 fe80::fec2:3dff:fe0d:2d1f%eno1

take several seconds to be received.

#### Versions
RIOT master

benpicco, 2021-07-14:
Not sure if this is related, but this is something odd:

TTL of ICMPv6 ping replies goes up to 255 shortly after reset:

```
> ping fe80::a4ed:29ff:feec:cbf3%eno1 
PING fe80::a4ed:29ff:feec:cbf3%eno1(fe80::a4ed:29ff:feec:cbf3%eno1) 56 data bytes
64 bytes from fe80::a4ed:29ff:feec:cbf3%eno1: icmp_seq=7 ttl=64 time=0.690 ms
64 bytes from fe80::a4ed:29ff:feec:cbf3%eno1: icmp_seq=8 ttl=255 time=0.300 ms
64 bytes from fe80::a4ed:29ff:feec:cbf3%eno1: icmp_seq=9 ttl=255 time=0.332 ms
64 bytes from fe80::a4ed:29ff:feec:cbf3%eno1: icmp_seq=10 ttl=255 time=0.343 ms
64 bytes from fe80::a4ed:29ff:feec:cbf3%eno1: icmp_seq=11 ttl=255 time=0.296 ms
# reset
64 bytes from fe80::a4ed:29ff:feec:cbf3%eno1: icmp_seq=15 ttl=64 time=0.732 ms
64 bytes from fe80::a4ed:29ff:feec:cbf3%eno1: icmp_seq=16 ttl=64 time=0.312 ms
64 bytes from fe80::a4ed:29ff:feec:cbf3%eno1: icmp_seq=17 ttl=255 time=0.303 ms
64 bytes from fe80::a4ed:29ff:feec:cbf3%eno1: icmp_seq=18 ttl=255 time=0.352 ms
64 bytes from fe80::a4ed:29ff:feec:cbf3%eno1: icmp_seq=19 ttl=255 time=0.315 ms
64 bytes from fe80::a4ed:29ff:feec:cbf3%eno1: icmp_seq=20 ttl=255 time=0.317 ms
64 bytes from fe80::a4ed:29ff:feec:cbf3%eno1: icmp_seq=21 ttl=255 time=0.323 ms
```

dylad, 2021-07-17:
@benpicco I cannot reproduce this when I plug my SAME54-XPRO directly on my laptop.
TTL value remains constant even if  I reset the board while pinging from the host side.

Did you plug your board directly on your laptop ? Is there a gateway between ?

benpicco, 2021-07-17:
Both my machine and the `same54-xpro` are connected to a switch that is connected to the rest of my LAN.

dylad, 2021-07-19:
> Both my machine and the same54-xpro are connected to a switch that is connected to the rest of my LAN.

I did try to connect both my laptop and my board to a local switch in my office, no issue so far with TTL. I still cannot reproduce :(

PR #16683 (2021-07-27): cpu/sam0: improve ethernet driver resilience
### Contribution description
This PR provides a better resilience for the SAM0 ethernet driver in case of heavy traffic in the network.

In some case interrupts fire faster than the netdev thread can process them and we run out of buffers quickly. With this PR, we now check if we don't have buffers available, so we can flush everything and restart reception properly even if we did drop a few in the operation.

### Testing procedure
See #16451 

### Issues/PRs references
Fixes #16451
closed 2021-07-27 by benpicco