RIOT-OS/RIOT #16471: pkg/lvgl: LoadProhibitedCause exception on BOARD=esp32-wrover-kit

jeandudey, 2021-05-13:
#### Description

On latest RIOT master the code compiles for the `tests/pkg_lvgl` test and flashes correctly using:

- `make -C tests/pkg_lvgl BOARD=esp32-wrover-kit flash`

However nothing is displayed and instead outputs the following error:

```
jeandudey@jean ~/D/R/t/pkg_lvgl> make BOARD=esp32-wrover-kit term
/home/jeandudey/Dev/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyUSB1" -b "115200"  
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2021-05-13 17:17:23,103 # Connect to serial port /dev/ttyUSB1
Welcome to pyterm!
Type '/exit' to exit.
2021-05-13 17:17:27,072 # ets Jun  8 2016 00:22:57
2021-05-13 17:17:27,072 # 
2021-05-13 17:17:27,073 # rst:0x7 (TG0WDT_SYS_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
2021-05-13 17:17:27,073 # configsip: 0, SPIWP:0xee
2021-05-13 17:17:27,087 # clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
2021-05-13 17:17:27,088 # mode:DOUT, clock div:2
2021-05-13 17:17:27,088 # load:0x3fff0018,len:4
2021-05-13 17:17:27,089 # load:0x3fff001c,len:4004
2021-05-13 17:17:27,090 # load:0x40078000,len:7304
2021-05-13 17:17:27,090 # load:0x40080000,len:7224
2021-05-13 17:17:27,091 # entry 0x40080350
2021-05-13 17:17:27,103 # W (29) boot: PRO CPU has been reset by WDT.
2021-05-13 17:17:27,104 # W (29) boot: WDT reset info: PRO CPU PC=0x400881b8
2021-05-13 17:17:27,106 # W (29) boot: WDT reset info: APP CPU PC=0xb7eb2555
2021-05-13 17:17:27,215 # 
2021-05-13 17:17:27,233 # EXCEPTION!! exccause=28 (LoadProhibitedCause) @800d01ea excvaddr=00000004
2021-05-13 17:17:27,234 # heap: 309808 (used 696, free 309112) [bytes]
2021-05-13 17:17:27,234 # 
2021-05-13 17:17:27,235 # register set
2021-05-13 17:17:27,249 # pc      : 400d01d7	ps      : 00060c30	exccause: 0000001c	excvaddr: 00000004
2021-05-13 17:17:27,252 # epc1    : 400d01d7	epc2    : 00000000	epc3    : 00000000	epc4    : 00000000
2021-05-13 17:17:27,253 # epc5    : 00000000	epc6    : 00000000	epc7    : 00000000	
2021-05-13 17:17:27,265 # a0      : 800d01ea	a1      : 3ffb1dd0	a2      : 00000000	a3      : 3ffb5740
2021-05-13 17:17:27,267 # a4      : 400a0000	a5      : 400a0000	a6      : 3f4048c0	a7      : 00000000
2021-05-13 17:17:27,280 # a8      : 3ffb0ee4	a9      : 3ffb1db0	a10     : 00000000	a11     : 00000000
2021-05-13 17:17:27,282 # a12     : 7fffffff	a13     : 000a2200	a14     : ffffffff	a15     : 00000001
2021-05-13 17:17:27,284 # lbeg    : 400e2738	lend    : 400e275f	lcount  : 00000000
```

Will further investigate since last time the LoadProhibitedCause error occurred due to an error in the priorities. See #15428

#### Steps to reproduce the issue

- `make -C tests/pkg_lvgl BOARD=esp32-wrover-kit flash`

#### Expected results

The test outputs the results on the screen.

#### Actual results

LoadProhibitedCause exception is thrown.

#### Versions

Latest RIOT master.

aabadie, 2021-05-19:
I don't have the hardware for testing this. Can you bisect to find the problematic commit ? I suspect that it comes from #14339 where the lvgl thread is started from a dedicated function:

https://github.com/RIOT-OS/RIOT/blob/9c749bc21406b8e2d9f4ca5633fa2169e57251ec/pkg/lvgl/contrib/lvgl.c#L161-L166

This function must be called from the main function of an application:

https://github.com/RIOT-OS/RIOT/blob/9c749bc21406b8e2d9f4ca5633fa2169e57251ec/tests/pkg_lvgl/main.c#L124

The lvgl thread creation was cutted out from `lvgl_init` because this function is now called from auto_init:

https://github.com/RIOT-OS/RIOT/blob/9c749bc21406b8e2d9f4ca5633fa2169e57251ec/sys/auto_init/screen/auto_init_lvgl.c#L46-L64

So before main is called.

It works perfectly on ARM, I don't know why this doesn't work ESP.

aabadie, 2021-05-20:
@jeandudey, can you try #16486 and tell of it fixes the problem on esp32 ?

PR #16488 (2021-05-20): boards/esp32-wrover-kit: add dependency to ILI9341
### Contribution description

When `disp_dev` is present, pull the ili9341 driver to allow auto-initialization. Previously initialization of LVGL failed because no display was found.

### Testing procedure

- `make -C tests/pkg_lvgl BOARD=esp32-wrover-kit flash`. See the system monitor test 😉 

### Issues/PRs references

Fixes #16471