RIOT-OS/RIOT #16500: Flow control for STDIO CDC ACM

ant9000, 2021-05-26:
#### Description

When using the USB CDC ACM serial interface for STDIO, it is quite easy to overflow the internal buffer. It is possible to enlarge the buffer, of course, but for some use cases it would be preferrable to implement a flow control mechanism that blocks the host until the received characters are fully processed.

### Useful links

I've created a [PoC](https://github.com/ant9000/test_usb_uart) to exhibit the behavior described. Ideally, I'd like to see the host pause whenever the board does - so that no byte gets lost.

fjmolinas, 2021-11-10:
@bergzand this somehow relates to whet we iscussed offline

maribu, 2023-05-17:
@bergzand: Ping?