RIOT-OS/RIOT #16512: "Context before hardfault:" when calling msg_try_send()

lvjiamei, 2021-05-31:
## The description of the bug:
Based on RIOT, I am try to build a project that running TCP over BLE with lwIP stack and Nimble. I use two nrf52840 (Flash memory: 1024 kB , RAM:256 kB) boards. 
However, I encounter a problem that the program often crashes when calling msg_send(). The corresponding log is "Context before hardfault:". There is no more useful log about the bug. The log is shown in the screenshot. 
[image].

How can I determine the real cause of the program crash? Or there is any useful advice  that can provide me with more information about the "hardfault"? Thanks!!

kaspar030, 2021-05-31:
Can you try running `BOARD=nrf52840dk make debug` on one of the failing boards? The debugger might be able to show a backtrace.

lvjiamei, 2021-06-01:
> Can you try running `BOARD=nrf52840dk make debug` on one of the failing boards? The debugger might be able to show a backtrace.

Thanks for your advice! I try to run `BOARD=nrf52840dk make debug`.  However, there is a error —"Cannot find bounds of current function" when I debug the code step by step. the output is as below.
[image]

What' worse, I found my program crashes earlier after I made some changes (I forget what I have changed :() .  The log is
  '
  > Stack pointer corrupted, reset to top of stack
  FSR/FAR:
   CFSR: 0x00000092
   HFSR: 0x40000000
   DFSR: 0x00000000
   AFSR: 0x00000000
  MMFAR: 0x20000018
  Misc
  EXC_RET: 0xfffffff1
  '
The screenshot is：
[image]
I have add ‘CFLAGS += -DDEBUG_ASSERT_VERBOSE’ in the Makefile and there seems no more information.
How can I determine where the error occurred? Thanks a lot!!!

jeandudey, 2021-06-01:
>Thanks for your advice! I try to run BOARD=nrf52840dk make debug. However, there is a error —"Cannot find bounds of current function" when I debug the code step by step. the output is as below.

Try running the code without stepping manually, I would usually do:

```
>monitor reset
>continue
# Hardfault should be reached, now show backtrace
> bt
```

Other thing to consider is the message `Context before hardfault`  should probably print a message that says:

>Attempting to reconstruct state for debugging...

Containing some instructions for you to insert on GDB to inspect where the hardfault happened.

closed 2022-04-18 by stale[bot]