RIOT-OS/RIOT #16517: Cannot assign a pointer when interrupted

sada45, 2021-06-01:
Hey!
     I am using the nimble l2cap example to connect multiple server nodes. Now I want to record every node the master has connected, so I have a struct named `remotedevice`. 
```c
struct remotedevice{
	struct remotedevice *next;
	uint8_t data_type;
	ble_addr_t addr;
	char *name;
	thread_t *devthread;
	uint16_t handle;
	struct ble_l2cap_chan *coc;
	uint8_t *receive_buf;
	uint8_t *write_buf;
};
```
At start up, the node will call `remote_device_init()` to create a Dummy-Head linked list `remotedevice_list`. Once the master node has discover a node (in function `_filter_and_connect()`), it will call `remote_device_add()` to the list. But when assign the pointer of new device to `remtedevice_list->next` it has a hardfault, but no further information has been displayed. It seems that, the formatted output function are failed, so the status of register can not be printed (I use SEGGER RTT as the stdio). The code is shown as blow:
```c
static void _filter_and_connect(struct ble_gap_disc_desc *disc)
{
    bluetil_ad_t ad;
    int res = 0;

    bluetil_ad_init(&ad, (uint8_t *)disc->data,
                    (size_t)disc->length_data, (size_t)disc->length_data);
    bluetil_ad_data_t field;
    if (bluetil_ad_find(&ad, BLE_GAP_AD_NAME, &field) == BLUETIL_AD_OK){
	if (memcmp(field.data, "BLEdge_Light_Sensor", field.len) == 0){
	    remote_device_add(DEV_PERIODIC, field.data, field.len, &disc->addr);
	    res = ble_gap_disc_cancel();
	    expect(res == 0);
	    res = ble_gap_connect(nimble_riot_own_addr_type, &disc->addr,
				     BLE_HS_FOREVER, NULL, _on_gap_evt, &disc->addr);
	    expect(res == 0);
	    thread_flags_set(_connection_thread, FLAG_GAP_DONE);
    }
    else{
           // Continue to scan
    }
}
```

```c
/* add the device to device list */
struct remotedevice *remotedevice_list;
int dev_counter = 0;

/* Implementation of the module */

void remote_device_init(void){
	remotedevice_list = (struct remotedevice *)malloc(sizeof(struct remotedevice));
	memset(remotedevice_list, 0, sizeof(struct remotedevice));
}

int remote_device_add(uint8_t data_type, uint8_t *name, size_t name_len, ble_addr_t *addr){
	struct remotedevice dev;
	memset(&dev, 0, sizeof(struct remotedevice));
	memcpy(dev.name, name, name_len);
	dev.data_type = data_type;
	memcpy(&dev.addr, addr, sizeof(ble_addr_t));
	struct remotedevice *ptr = remotedevice_list;
	while (ptr->next){
	    if (memcmp(ptr->next->name, dev.name, name_len) == 0){
	        return dev_counter;
	    }
	    ptr = ptr->next;
	}
        /* Where the error is 
          * If I delete this,  
          * the program works fine */
	ptr->next = &dev;  // Error "Context before hardfault:"
	return ++dev_counter;
}
```

I have try the GDB, but I can not understand the output
[image]

Thanks!

jeandudey, 2021-06-01:
On:

```c
/* add the device to device list */
struct remotedevice *remotedevice_list;
```

The pointer is not valid and is likely NULL (`0x00000000`), removing the pointer may fix it. Let me know if this fixes the issue :-D

---

You may also want to take a look at [`utlist`](https://doc.riot-os.org/group__sys__ut.html) which provides some common operations for linked lists. For example, appending, deleting and iterating over a linked list.

sada45, 2021-06-01:
Thank for your quick reply!
But I not quite sure what's your mean.
The pointer `remotedevice_list` should be initialized at function `remote_device_init()`
A quick runs at `main()` works fine.
```c
int main(void)
{
	ble_addr_t addr = {0x01, {0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD}};
	remote_device_init();
	remote_device_add(1, "test", 5, &addr);
	printf("name = %s\n", remotedevice_list->next->name);
}
```
[image]

jeandudey, 2021-06-01:
It fails here:

```c
        /* Where the error is 
          * If I delete this,  
          * the program works fine */
	ptr->next = &dev;  // Error "Context before hardfault:"
```

Because `dev` is allocated on the stack, when the `remote_device_add` function returns `ptr->next` contains a pointer to garbage memory, you may want to allocate it on the heap or do use static memory for this. Thats why it only works the first time, calling it a second/third time might not work properly.

sada45, 2021-06-01:
Yes, you are right. I use `malloc` to allocate the struct in the heap and it works fine. 
Thanks a lot!

jeandudey, 2021-06-01:
Closing since the issue was solved, thanks for asking!

closed 2021-06-01 by jeandudey