RIOT-OS/RIOT #16520: Reference errors when using multiple folders in C.

JaccoVeldscholten, 2021-06-01:
#### Description
Im receiving reference errors when im trying to use multiple folders in RIOT OS. What is the correct way to use multiple folders with software?
Currently my root structure is:

- tests.h
- tests.c
- Unity
- - unity.h
- - unity.c

Im trying to include unity.h from tests.h. With the following code:

**tests.h**
`#include "Unity/unity.h"`

But im getting references errors from the compiler that it does not reconigze the functions that from unity.h that are called from tests.h.

I did not edit the makefile, This might be a issue?

#### Steps to reproduce the issue
Create root structure with functions:

- tests.h
- tests.c
- Unity
- - unity.h
- - unity.c

**tests.h:**
```
#include "Unity/unity.h"

void test_true_should_be_true(void);
int TestSuite(void);

void setUp(void);
void tearDown(void);
```

**tests.c**
`#include "tests.h"

void setUp(void){}
void tearDown(void){}

void test_true_should_be_true(void){
    TEST_ASSERT_TRUE(1);
}

int TestSuite(void){
    UNITY_BEGIN();
    RUN_TEST(test_true_should_be_true);
    return UNITY_END();
}`

The Unity file is from Unity tests (http://www.throwtheswitch.org/unity)

#### Versions
arm-none-aebi 9.2.1 Toolchain Riot Docker Container

jeandudey, 2021-06-03:
Hello @JaccoVeldscholten, by default RIOT compiles only files in your current directory, other directories aren't included by default so that's why you get the 'undefined reference' errors, because the linker cannot find the functions it needs for the program to work.

On the on coming version `2020.07` (and on `master` branch) you can use `EXTERNAL_MODULE_DIRS` to add folders where you want to find your modules, for example:

```
EXTERNAL_MODULE_DIRS += libraries
USEMODULE += Unity
```

And on a file `libraries/Unity/Makefile`:

```
include $(RIOTBASE)/Makefile.base
```

Please note that the module name (the one you add to `USEMODULE`) must be equal to the name of the folder containing the module.

JaccoVeldscholten, 2021-06-03:
Thanks. That solved thie issue.

closed 2021-06-03 by JaccoVeldscholten