RIOT-OS/RIOT #16529: auto_init: needs cleanup

miri64, 2021-06-05:
As noted in https://github.com/RIOT-OS/RIOT/pull/16528, the order of `auto_init` seems arbitrary and actually seems to cause some issues. My proposal: initialize devices first, then go up the network stack(s) and then initialize application specific stuff. Throw in in the middle whatever is needed by other modules first.

kaspar030, 2021-06-06:
Let's use XFA and its "priority" value.

miri64, 2021-06-06:
> Let's use XFA and its "priority" value.

How about we clean-up first and then deploy the experimental feature already years in the making ;-). Then there is little discussion needed as well on the priority value and people who might need or want to backport to an older version of RIOT can still do so with a clean `auto_init` :-).

miri64, 2022-04-18:
I guess this was fixed with #17794? If not, please re-open @fabian18.

fabian18, 2022-04-18:
The default order is still the same. It is possible to `#define` the priorities beforehand but it is not very convenient, because the goal was to keep it simple. So it is kind of half fixed. I think it could be reopened if someone still experiences problems with the current order.