RIOT-OS/RIOT #16541: TCP client  cannot  send read only data

kex2017, 2021-06-09:
#### Description
TCP client  cannot  send read only data

#### Steps to reproduce the issue
hardware:stm32f407ze
usemodule:stm32_eth
test ipv4 tcp client send const data like this "const char sendbuf[] = "Hello!";"
The test code is as follows：
`
const char sendbuf[] = "Hello!";
int test_tcp_client(void)
{
    int res = 0;
    sock_tcp_t sock;
    sock_tcp_ep_t remote = SOCK_IPV4_EP_ANY;

    remote.port = SERVER_PORT;//SERVER_PORT=12345

    ipv4_addr_from_str((ipv4_addr_t *)&remote.addr, SERVER_IP_ADDR);//SERVER_IP_ADDR=192.168.12.89
    xtimer_sleep(2);

    if (sock_tcp_connect(&sock, &remote, 0, 0) < 0)
    {
        puts("Error connecting sock");
        return 1;
    }

    puts("Sending \"Hello!\"");
    if ((res = sock_tcp_write(&sock, sendbuf, sizeof(sendbuf))) < 0)
    {
        puts("Errored on write");
    }
    puts("send over...");
`
However, if the data sent is modified to  "char sendbuf[] = "Hello!";", without  "const",it will be sent successfully

#### Expected results

#### Actual results

#### Versions
Operating System Environment
----------------------------
         Operating System: "Ubuntu" "18.04.5 LTS (Bionic Beaver)"
                   Kernel: Linux 5.4.0-73-generic x86_64 x86_64
             System shell: /bin/dash (probably dash)
             make's shell: /bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.0.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.10.2
                 cppcheck: missing
                  doxygen: 1.8.13
                      git: git version 2.17.1
                     make: GNU Make 4.1
                  openocd: missing
                   python: Python 2.7.17
                  python2: Python 2.7.17
                  python3: Python 3.6.9
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing

kaspar030, 2021-06-09:
stm32_eth is using DMA for sending: https://github.com/kaspar030/RIOT/blob/88da1421be9401c6ce0b6152ce502547e8d1efc3/cpu/stm32/periph/eth.c#L468.
The nrf DMA controller can only transfer from/to RAM, maybe stm32 has a similar restriction?

@maribu Maybe you can take a look?

maribu, 2021-06-09:
According to the reference manual, reading from internal flash is fine: https://www.st.com/resource/en/reference_manual/dm00031020-stm32f405415-stm32f407417-stm32f427437-and-stm32f429439-advanced-armbased-32bit-mcus-stmicroelectronics.pdf

> ### 2.1.6 Ethernet DMA bus
>
> This bus connects the Ethernet DMA master interface to the BusMatrix. This bus is used by the Ethernet DMA to load/store data to a memory. The targets of this bus are data memories: internal SRAMs (SRAM1, SRAM2, SRAM3), internal Flash memory, and external memories through the FSMC/FMC.

maribu, 2021-06-09:
I used the following patch

```diff
diff --git a/examples/gnrc_networking/udp.c b/examples/gnrc_networking/udp.c
index 54f00afd7e..e72e026c63 100644
--- a/examples/gnrc_networking/udp.c
+++ b/examples/gnrc_networking/udp.c
@@ -36,7 +36,7 @@ static gnrc_netreg_entry_t server = GNRC_NETREG_ENTRY_INIT_PID(GNRC_NETREG_DEMUX
                                                                KERNEL_PID_UNDEF);
 
 
-static void send(char *addr_str, char *port_str, char *data, unsigned int num,
+static void send(char *addr_str, char *port_str, const char *data, unsigned int num,
                  unsigned int delay)
 {
     gnrc_netif_t *netif = NULL;
@@ -149,7 +149,7 @@ static void stop_server(void)
 int udp_cmd(int argc, char **argv)
 {
     if (argc < 2) {
-        printf("usage: %s [send|server]\n", argv[0]);
+        printf("usage: %s [send|server|const]\n", argv[0]);
         return 1;
     }
 
@@ -169,6 +169,23 @@ int udp_cmd(int argc, char **argv)
         }
         send(argv[2], argv[3], argv[4], num, delay);
     }
+    else if (strcmp(argv[1], "const") == 0) {
+        uint32_t num = 1;
+        uint32_t delay = 1000000;
+        if (argc < 4) {
+            printf("usage: %s send <addr> <port> [<num> [<delay in us>]]\n",
+                   argv[0]);
+            return 1;
+        }
+        if (argc > 4) {
+            num = atoi(argv[4]);
+        }
+        if (argc > 5) {
+            delay = atoi(argv[5]);
+        }
+        const char *data = "Hallo world!";
+        send(argv[2], argv[3], data, num, delay);
+    }
     else if (strcmp(argv[1], "server") == 0) {
         if (argc < 3) {
             printf("usage: %s server [start|stop]\n", argv[0]);
```

flashed `examples/gnrc_networking` to a Nucleo-F767ZI and did the following:

```
2021-06-09 09:17:47,217 # udp const fe80::.... 12345
2021-06-09 09:17:47,223 # Success: sent 12 byte(s) to [fe80::....]:12345
```

With Wireshark, I can see a valid UDP datagram being received that contains "Hello world!" as payload. The Ethernet peripheral for STM32 is as far as I know identical for all STM32 MCUs. So that should also work on F4.

@kex2017 Could you provide the exact source code so that I can try to reproduce? If it is based on an example program, you can just provide the output of `git diff` here.

kex2017, 2021-06-10:
I use my board to test that UDP can send read-only data successfully, but if the same data is sent by TCP, it will fail.
Blocking when sending TCP data.
My code refers to the following file:
https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/sock/tcp.h#L191
https://github.com/RIOT-OS/RIOT/blob/master/tests/lwip_sock_tcp/main.c#L493

`**main.c**:
/*-----------------------------------------------------------------------------------------------*/
#include <stdio.h>

#include "tcp_udp_test.h"

int main(void)
{
    update_dev_net_info_to_eth();

    test_tcp_client();
    // test_udp_client();
    // test_tcp_server();

    return 0;
}
/*-----------------------------------------------------------------------------------------------*/

**tcp_udp_test.c**
/*-----------------------------------------------------------------------------------------------*/
#include "xtimer.h"
#include <stdio.h>

#include "lwip/ip4_addr.h"
#include "lwip/netif.h"

#include "net/sock/udp.h"

#include "net/sock/tcp.h"

#define DEV_IP_ADDR ("192.168.12.206")
#define DEV_NETMASK ("255.255.255.0")
#define DEV_GATEWAY ("192.168.12.1")

#define SERVER_IP_ADDR ("192.168.12.89")
#define SERVER_PORT (12345)

const char sendbuf[] = "Hello!";

/**********************************************************************/
void update_dev_net_info_to_eth(void)
{
    struct netif *netif = netif_get_by_index(netif_name_to_index("ET")); //or:netif_find("ET");
    ip4_addr_t ipaddr = {0}, netmask = {0}, gw = {0};

    ipaddr.addr = ipaddr_addr(DEV_IP_ADDR);
    netmask.addr = ipaddr_addr(DEV_NETMASK);
    gw.addr = ipaddr_addr(DEV_GATEWAY);

    netif_set_addr(netif, &ipaddr, &netmask, &gw);
}

/***********************test udp client***********************/
int test_udp_client(void)
{
    int ret = 0;
    sock_udp_ep_t local = SOCK_IPV4_EP_ANY;
    sock_udp_t sock;
    local.port = 0xabcd;

    if (sock_udp_create(&sock, &local, NULL, 0) < 0)
    {
        puts("Error creating UDP sock");
        return 1;
    }
    sock_udp_ep_t remote = {.family = AF_INET, .port = SERVER_PORT};
    ipv4_addr_from_str((ipv4_addr_t *)&remote.addr, SERVER_IP_ADDR);
    xtimer_sleep(2);

    while (1)
    {
        if ((ret = sock_udp_send(&sock, sendbuf, sizeof(sendbuf), &remote)) < 0)
        {
            printf("ret is %d\r\n", ret);
            puts("Error sending message");
            sock_udp_close(&sock);
            return 1;
        }
        xtimer_sleep(2);
    }
    return 0;
}

/***********************test tcp client***********************/
int test_tcp_client(void)
{
    int res = 0;
    sock_tcp_t sock;
    sock_tcp_ep_t remote = SOCK_IPV4_EP_ANY;

    remote.port = SERVER_PORT;

    ipv4_addr_from_str((ipv4_addr_t *)&remote.addr, SERVER_IP_ADDR);
    xtimer_sleep(2);

    if (sock_tcp_connect(&sock, &remote, 0, 0) < 0)
    {
        puts("Error connecting sock");
        return 1;
    }

    while (1)
    {
        puts("Sending \"Hello!\"");
        if ((res = sock_tcp_write(&sock, sendbuf, sizeof(sendbuf))) < 0)
        {
            puts("Errored on write");
        }
        puts("send over...");
    }

    return 0;
}

/***********************test tcp server***********************/
#define SOCK_QUEUE_LEN (1U)
sock_tcp_t sock_queue[SOCK_QUEUE_LEN];
uint8_t buf[128] = {0};

int test_tcp_server(void)
{
    sock_tcp_ep_t local = SOCK_IPV4_EP_ANY;
    sock_tcp_queue_t queue;

    local.port = SERVER_PORT;

    if (sock_tcp_listen(&queue, &local, sock_queue, SOCK_QUEUE_LEN, 0) < 0)
    {
        puts("Error creating listening queue");
        return 1;
    }
    puts("Listening on port 12345");
    while (1)
    {
        sock_tcp_t *sock;

        if (sock_tcp_accept(&queue, &sock, 0) < 0)
        {
            puts("Error accepting new sock");
        }
        else
        {
            int read_res = 0;

            puts("Reading data");
            while (read_res >= 0)
            {
                read_res = sock_tcp_read(sock, &buf, sizeof(buf),
                                         SOCK_NO_TIMEOUT);
                if (read_res < 0)
                {
                    puts("Disconnected");
                    break;
                }
                else
                {
                    int write_res;
                    printf("Read: \"");
                    for (int i = 0; i < read_res; i++)
                    {
                        printf("%c", buf[i]);
                    }
                    puts("\"");
                    if ((write_res = sock_tcp_write(sock, &buf,
                                                    read_res)) < 0)
                    {
                        puts("Errored on write, finished server loop");
                        break;
                    }
                }
            }
            sock_tcp_disconnect(sock);
        }
    }
    sock_tcp_stop_listen(&queue);
    return 0;
}
/*-----------------------------------------------------------------------------------------------*/
`
@maribu

crasbe, 2026-10-30:
Claude wrote a short reproducer:

```
/*
 * Reproducer for RIOT-OS/RIOT#16541: TCP client writes const (flash) data,
 * RAM data or per-write unique sequence data back to back via sock_tcp.
 */
#include <errno.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include "net/ipv4/addr.h"
#include "net/sock/tcp.h"
#include "shell.h"
#include "ztimer.h"

#include "lwip/netif.h"

/* exactly as in the issue: lives in .rodata (flash) */
const char sendbuf_const[] = "Hello!";
/* same content, but in RAM */
char sendbuf_ram[] = "Hello!";

static sock_tcp_t _sock;

static int _cmd_tcpc(int argc, char **argv)
{
    if (argc < 4) {
        printf("usage: %s <ipv4> <port> <const|ram|seq> [num] [delay_ms]\n", argv[0]);
        return 1;
    }
    sock_tcp_ep_t remote = SOCK_IPV4_EP_ANY;
    if (ipv4_addr_from_str((ipv4_addr_t *)&remote.addr.ipv4, argv[1]) == NULL) {
        puts("bad address");
        return 1;
    }
    remote.port = atoi(argv[2]);
    const char *mode = argv[3];
    unsigned num = (argc > 4) ? strtoul(argv[4], NULL, 0) : 10;
    unsigned delay = (argc > 5) ? strtoul(argv[5], NULL, 0) : 0;

    int res = sock_tcp_connect(&_sock, &remote, 0, 0);
    if (res < 0) {
        printf("connect failed: %d\n", res);
        return 1;
    }
    printf("connected, mode=%s num=%u delay=%u ms\n", mode, num, delay);

    uint32_t start = ztimer_now(ZTIMER_MSEC);
    unsigned ok = 0;
    for (unsigned i = 0; i < num; i++) {
        char seqbuf[16];
        const void *data;
        size_t len;

        if (strcmp(mode, "const") == 0) {
            data = sendbuf_const;
            len = sizeof(sendbuf_const);
        }
        else if (strcmp(mode, "ram") == 0) {
            data = sendbuf_ram;
            len = sizeof(sendbuf_ram);
        }
        else {
            len = snprintf(seqbuf, sizeof(seqbuf), "S%08u\n", i);
            data = seqbuf;
        }
        res = sock_tcp_write(&_sock, data, len);
        if (res < 0) {
            printf("write %u failed: %d\n", i, res);
            break;
        }
        if ((size_t)res != len) {
            printf("write %u short: %d/%u\n", i, res, (unsigned)len);
        }
        ok++;
        if (delay) {
            ztimer_sleep(ZTIMER_MSEC, delay);
        }
    }
    printf("done: %u/%u writes ok in %" PRIu32 " ms\n", ok, num,
           ztimer_now(ZTIMER_MSEC) - start);
    ztimer_sleep(ZTIMER_MSEC, 500);
    sock_tcp_disconnect(&_sock);
    puts("disconnected");
    return 0;
}

static int _cmd_where(int argc, char **argv)
{
    (void)argc; (void)argv;
    printf("sendbuf_const @ %p, sendbuf_ram @ %p\n",
           (void *)sendbuf_const, (void *)sendbuf_ram);
    return 0;
}

static const shell_command_t _cmds[] = {
    { "tcpc", "TCP client test for #16541", _cmd_tcpc },
    { "where", "print buffer addresses", _cmd_where },
    { NULL, NULL, NULL }
};

int main(void)
{
    puts("repro #16541");
    char line_buf[SHELL_DEFAULT_BUFSIZE];
    shell_run(_cmds, line_buf, SHELL_DEFAULT_BUFSIZE);
    return 0;
}
```

```
# Reproducer for RIOT-OS/RIOT#16541 (TCP client cannot send read-only data)
APPLICATION = repro_16541
BOARD ?= nucleo-f207zg
RIOTBASE ?= $(CURDIR)/../RIOT

USEMODULE += lwip
USEMODULE += lwip_ipv4
USEMODULE += lwip_arp
USEMODULE += lwip_dhcp_auto
USEMODULE += lwip_tcp
USEMODULE += sock_tcp
USEMODULE += ipv4_addr
USEMODULE += netdev_default
USEMODULE += shell
USEMODULE += shell_cmds_default
USEMODULE += shell_cmd_lwip_netif
USEMODULE += ps
USEMODULE += ztimer_msec

DEVELHELP ?= 1
QUIET ?= 1

include $(RIOTBASE)/Makefile.include
```

We can run the following commands on the board (a `nucleo-l207zg` in my case, but the network controller is the same):
```
> ifconfig
2026-10-30 10:28:08,394 # ifconfig
2026-10-30 10:28:08,399 # Iface ET0 HWaddr: da:40:ca:ee:aa:91 Link: up State: up
2026-10-30 10:28:08,402 #         Link type: wired
2026-10-30 10:28:08,408 #         inet addr: 134.28.42.168 mask: 255.255.255.0 gw: 134.28.42.42
> tcpc 134.28.42.150 12345 seq 2000
2026-10-30 10:28:49,133 # tcpc 134.28.42.150 12345 seq 2000
2026-10-30 10:28:49,137 # connected, mode=seq num=2000 delay=0 ms
2026-10-30 10:28:49,413 # done: 2000/2000 writes ok in 272 ms
2026-10-30 10:28:49,914 # disconnected
```

And this on the machine. We expect a continuous sequence, which is generated with the `seq` command.
```
chris@WorkPringle:~$ ifconfig
enp4s0f2np2: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 134.28.42.150  netmask 255.255.255.0  broadcast 134.28.42.255
        inet6 fe80::44ab:6d4a:4fe9:8601  prefixlen 64  scopeid 0x20<link>
        ether 84:a9:3e:85:2b:27  txqueuelen 1000  (Ethernet)
        RX packets 12644542  bytes 11319270440 (11.3 GB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 8209539  bytes 2476573962 (2.4 GB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

chris@WorkPringle:~$ nc -l 12345 > out.bin
chris@WorkPringle:~$ seq -f 'S%08g' 0 1999 > exp.txt
chris@WorkPringle:~$ cmp out.bin exp.txt && echo OK
out.bin exp.txt sind verschieden: Byte 997, Zeile 100

chris@WorkPringle:~$ sed -n '98,102p' out.bin
S00000097
S00000098
S00000100
S00000100
S00000101
```

With the fix from #23037 applied, we get this:
```
chris@WorkPringle:~$ nc -l 12345 > out.bin
chris@WorkPringle:~$ cmp out.bin exp.txt && echo OK
OK
```

I also tried it with 20000 sequences and it worked just the same with no errors. I can't reproduce the hang locally, but that might have been fixed by https://github.com/RIOT-OS/RIOT/pull/17485 in the meantime.

closed 2026-11-02 by crasbe