RIOT-OS/RIOT #16589: drivers/at24cxxx: at24cxxx_params.h doesn't compile in C++

nicoHarel, 2021-06-28:
Hello everyone,

#### Description

 Cannot compile _drivers/at24cxxx/include/at24cxxx_params.h_ while using C++ because of at24cxxx_params initialization. 
```
RIOT/drivers/at24cxxx/include/at24cxxx_params.h:96:1: sorry, unimplemented: non-trivial designated initializers not supported
 };
```
Problem is there's no missing fields at initialization in the structure but they're not in the right order, so G++ can't understand it. A problem that I have not encountered while using other various drivers because the parameter structures are initialized correctly.

Fix for is very simple but I can never remember if we are supposed to open a bug repport issue in riot's github or we can go straight ahead and open a PR. 

#### Steps to reproduce the issue

Include at24cxxx driver in a C++ project and compile. 

Have a good day

dylad, 2021-06-28:
> Fix for is very simple but I can never remember if we are supposed to open a bug repport issue in riot's github or we can go straight ahead and open a PR.

Hello &Thanks for the report !
You can directly open a PR for this kind of fix, as this is pretty straightforward.

dylad, 2021-07-07:
I'm closing this as #16619 was merged earlier.
Thanks for the fix !

closed 2021-07-07 by dylad