RIOT-OS/RIOT #16649: Initialization failing for BME680 sensor

thanosmanolis, 2021-07-14:
Hello,

I am trying to use BME680 sensor with an ESP32 board. However, even [this](https://github.com/RIOT-OS/RIOT/tree/master/tests/driver_bme680) test, prints this error:

> Initialize BME680 sensor ... failed

I am quite sure the hardware connection is not false on my side since it's pretty simple (only I2C supported by the sensor, so just SDA,SCL,VCC,GND pins).

The thing is, my BME680 sensor is not the Bosch's one, but [this](https://shop.pimoroni.com/products/bme680-breakout) one (pimoroni). So is it possible that it is just not supported by RIOT, or should I search for another solution?

Thank you.

wosym, 2021-07-23:
Just a sidenote: You Pimoroni is the manufacturer of the breakout board. The chip itself is still made by Bosch.
It even said so on the page you linked: 
> Bosch BME680 temperature, pressure, humidity, air quality sensor

Regarding your issue: Maybe setting `#define ENABLE_DEBUG 0` to `1` in `/drivers/bme680/bme680.c` can shed some more light on the issue you are experiencing?

thanosmanolis, 2021-07-23:
You are right, thanks for the clarification. 

I set the DEBUG setting to 1, and now I get this error:

> Initialize BME680 sensor 0 ... [bme680]: Failed to get ID
> failed

which derives from the call of `bme680_init_internal` function on `/drivers/bme680/bme680.c`

thanosmanolis, 2021-07-24:
I solved it by setting `BME680_PARAM_I2C_ADDR` to `BME680_I2C_ADDR_1`, which is 0x76, instead of the default one that was 0x77. I had already tried to do that from the very beginning but maybe I didn't do it correctly.

I am now closing this issue since it's solved, thank you for your answer. :blush:

closed 2021-07-24 by thanosmanolis