RIOT-OS/RIOT #16677: dhcpv6_client: mrd calculation for renew and rebind is wrong

kfessel, 2021-07-23:
#### Description

while reviewing #16668, hat a look at all of dhcpv6_clients time(r) calls i was not sure about the handling of rebind_time.

there are errors  in the use of rebind time and valid_time for the `mrd` calculation 

a printf in l.892 after the PR reveals the issue if one uses short T1 T2

```
printf("DHCPv6 client: mrd %"PRIu32" , etime %"PRIi16"\n", mrd, _get_elapsed_time() );
```

mrd ever increasing 

the issue is there before the PR and after the PR (they are generated by unchanged lines) i got no intend to search for them before PR

#### Steps to reproduce the issue

setup the dhcpv6 to issue short leases (there is a kea config with the dhcp client test)

and a printf to show MRD 

#### Expected results

the mrd printed should be ~constant 

#### Actual results

mrd is incresing ( it is the time the last lease was recived)

#### Versions

current and PR 16668

kfessel, 2021-07-23:
```
2021-07-23 15:30:42,370 # DHCPv6 client: mrd 6 , etime 0
2021-07-23 15:31:02,375 # DHCPv6 client: mrd 26 , etime 0
2021-07-23 15:31:22,379 # DHCPv6 client: mrd 46 , etime 0
2021-07-23 15:31:42,383 # DHCPv6 client: mrd 66 , etime 0
2021-07-23 15:32:02,387 # DHCPv6 client: mrd 86 , etime 0
2021-07-23 15:32:22,392 # DHCPv6 client: mrd 106 , etime 0
2021-07-23 15:32:42,396 # DHCPv6 client: mrd 126 , etime 0
2021-07-23 15:33:02,400 # DHCPv6 client: mrd 146 , etime 0
2021-07-23 15:33:22,405 # DHCPv6 client: mrd 166 , etime 0
2021-07-23 15:33:42,410 # DHCPv6 client: mrd 186 , etime 0
2021-07-23 15:34:02,414 # DHCPv6 client: mrd 206 , etime 0
2021-07-23 15:34:22,418 # DHCPv6 client: mrd 226 , etime 0
2021-07-23 15:34:42,422 # DHCPv6 client: mrd 246 , etime 0
2021-07-23 15:35:02,427 # DHCPv6 client: mrd 266 , etime 0
2021-07-23 15:35:22,432 # DHCPv6 client: mrd 286 , etime 0
2021-07-23 15:35:42,436 # DHCPv6 client: mrd 306 , etime 0
2021-07-23 15:36:02,440 # DHCPv6 client: mrd 326 , etime 0
2021-07-23 15:36:22,444 # DHCPv6 client: mrd 346 , etime 0
2021-07-23 15:36:42,449 # DHCPv6 client: mrd 366 , etime 0
2021-07-23 15:37:02,454 # DHCPv6 client: mrd 386 , etime 0
2021-07-23 15:37:22,458 # DHCPv6 client: mrd 406 , etime 0
```

miri64, 2021-07-23:
I am not sure you are seeing MRDs for the REBIND: "Remaining time until valid lifetimes of all leases in all IAs have expired", according to RFC 8415, section 18.2.5, which at least from the code seems to be just that, unless the values returned by the prefix list are wrong (can be checked with `nib prefix` and they are decresing). Would have been nice to also see the message type in your output ;-). For RENEW it is supposed to be "Remaining time until earliest T2", according to RFC 8415, section 18.2.4. Given that `rebind_time` is the time the T2 was received + T2, getting the correct value should just be a matter of adding the current time into the mix:

```diff
diff --git a/sys/net/application_layer/dhcpv6/client.c b/sys/net/application_layer/dhcpv6/client.c
index 0e92501e36..c6c6f75a0e 100644
--- a/sys/net/application_layer/dhcpv6/client.c
+++ b/sys/net/application_layer/dhcpv6/client.c
@@ -839,7 +839,7 @@ static void _request_renew_rebind(uint8_t type)
         case DHCPV6_RENEW:
             irt = DHCPV6_REN_TIMEOUT;
             mrt = DHCPV6_REN_MAX_RT;
-            mrd = rebind_time - t2;
+            mrd = (rebind_time - t2) - _now_sec();
             break;
         case DHCPV6_REBIND: {
             irt = DHCPV6_REB_TIMEOUT;
```

kfessel, 2021-07-23:
without the - t2

miri64, 2021-07-23:
Right! So

```diff
diff --git a/sys/net/application_layer/dhcpv6/client.c b/sys/net/application_layer/dhcpv6/client.c
index 0e92501e36..c6c6f75a0e 100644
--- a/sys/net/application_layer/dhcpv6/client.c
+++ b/sys/net/application_layer/dhcpv6/client.c
@@ -839,7 +839,7 @@ static void _request_renew_rebind(uint8_t type)
         case DHCPV6_RENEW:
             irt = DHCPV6_REN_TIMEOUT;
             mrt = DHCPV6_REN_MAX_RT;
-            mrd = rebind_time - t2;
+            mrd = rebind_time - _now_sec();
             break;
         case DHCPV6_REBIND: {
             irt = DHCPV6_REB_TIMEOUT;
```

kfessel, 2021-07-23:
and "dhcpv6_client_prefix_valid_until" does do what its name says but not what its description says 
the function return the time in second until a lease is valid on the msec (ev)timer

miri64, 2021-07-23:
> and "dhcpv6_client_prefix_valid_until" does do what its name says but not what its description says
> the function return the time in second until a lease is valid on the msec (ev)timer

AFAIK this can just be converted by dividing by 1000, since it is an offset to that timers time. Of course, due to the time being in 32-bit scale, there could be a cut-of, but that's an issue with GNRC's NIB, not the DHCPv6 client implementation.

closed 2021-07-23 by miri64