RIOT-OS/RIOT #16680: cpu/nrf52/periph/i2c

nicoHarel, 2021-07-26:
#### Description

Hello everyone.
After my fix for cpp on the at24cx driver I tried using it in my app which is powered by a NRF52 MCU.

Big problem: the i2C periph of that MCU doesn't like it supports 16-bit register addressing (see:  periph/i2c.c line 172)

Any reason why ?

Also a few lines later: 
> the nrf52's TWI device does not support to do two consecutive transfers
>      * without a repeated start condition in between. So we have to put all data
>      * to be transferred into a temporary buffer

That doesn't seem to be true: 
See this link: 
https://devzone.nordicsemi.com/f/nordic-q-a/5267/i2c-eeprom-with-16bit-addressing

I could use a use a 2KB AT24C chip but yeah... It's a bit small and ancient =)

benpicco, 2021-07-26:
> Any reason why?

My guess would be that nobody has gotten around to implementing it. If you want to take a stab at it, that would be great! :smiley:

nicoHarel, 2021-07-27:
> My guess would be that nobody has gotten around to implementing it. If you want to take a stab at it, that would be great! smiley

I'll have a look and give it a go as soon as I have a bit of time :wink:

nicoHarel, 2021-08-04:
Ok so a little update on this: - I won't open a PR just now because, unfortunately, I fail to see results with what I am trying to do. Write operation seems to be working (hard to confirm with my hardware and available tools at my disposal), but read operation is giving me nothing. 

I did a bit of reading: 
- The original PR: https://github.com/RIOT-OS/RIOT/pull/11929
- [The AT24C01/02/04/08/16 datasheet:](http://ww1.microchip.com/downloads/en/devicedoc/doc0180.pdf)
- [The AT24C256C datasheet ](https://www.mouser.fr/datasheet/2/268/AT24C256C-I2C-Compatible-Serial-EEPROM-256-Kbit-20-1398091.pdf)
- [The NRF52 EasyDMA datasheet](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.ps.v1.1%2Ftwim.html)

Mostly I'm just posting this here so the links are there and to you guys updated. I'll of course open a PR as soon as I see results.

closed 2021-08-06 by benpicco