RIOT-OS/RIOT #16690: Larger project for RIOT OS on STM32 device

ubik-valis, 2021-07-29:
Hi together,

I'm using a STM32 nucleo board and so far RIOT OS is amazing for realizing my IoT projects.

Currently I'm tackling a larger project that requires libraries like NTL and GMP to work properly. The application runs on my laptop quite fine, but that I have no idea how I would port this to my nucleo board with RIOT OS.

Has anyone run into this problem and could please provide some tips on how to approach this.

Thank you all for your help!

Citrullin, 2021-08-02:
I am not sure, if those libraries are save to use in a microcontroller environment. Regarding malloc etc.That is probably something you have to find out yourself.
Speaking of how to move forward and how to bring into RIOT OS. Quite easy. Make a [package for it](https://api.riot-os.org/group__pkg.html). 
If you need inspirations, you can also just take a look into existing packages. It may be a little tricky for some libraries, but it is possible to port any library. Good luck.

forellendev, 2021-08-10:
Thank you for your answer! I'll try to port the package and maybe this one could be interesting for the community too.

closed 2022-04-19 by stale[bot]