RIOT-OS/RIOT #16698: sys/ztimer: Potential footgun when mixing timestamps of ztimer clocks

maribu, 2021-08-03:
#### Description

Each ztimer clock can potentially be backed by different hardware. E.g. on a given system `ZTIMER_USEC` could use `periph_timer` as backend, while `ZTIMER_MSEC` could use `periph_rtt` as backend. A call to `ztimer_now()` will give the current counter value of the underlying hardware (after first applying frequency conversion and bit extension as needed). When comparing timestamps of different clocks this can be problematic (that is, after converting them to the same time unit), because:

1. Each hardware clock's epoch might be different. E.g. after a warm boot the RTT might resume counting from the value it had, while the high frequency timer will be reinitialized with zero.
2. Clocks will drift in regard to each other

#### Steps to reproduce the issue

```C
static ztimer_now_t point_in_time_a_us;
static ztimer_now_t point_in_time_b_ms;
static ztimer_now_t point_in_time_c_ms;

static void event_a_callback(void) {
    point_in_time_a_us = ztimer_now(ZTIMER_USEC);
}

static void event_b_callback(void) {
    point_in_time_b_ms = ztimer_now(ZTIMER_MSEC);
}

static void event_c_callback(void) {
    point_in_time_c_ms = ztimer_now(ZTIMER_MSEC);
    ztimer_now_t duration_a_b_us = point_in_time_b_ms * USEC_PER_MSEC - point_in_time_a_us;
    ztimer_now_t duration_b_c_ms = point_in_time_c_ms - point_in_time_b_ms;
}
```

#### Expected results

Both `duration_a_b_us` and `duration_b_c_ms` are correct.

#### Actual results

If `ZTIMER_MSEC` and `ZTIMER_USEC` are not provided by the same hardware clock, `duration_a_b_us` will likely be incorrect.

#### Versions

Current `master`.

maribu, 2021-08-03:
Crossposting from https://github.com/RIOT-OS/RIOT/pull/16697#issuecomment-891255179 :

> More generally, I wonder if we should reconsider whether allowing `ztimer_now()` on clocks with a different epochs is a good idea. IMO thisis a non-obvious footgun that is just waiting to blast off.
> 
> The alternatives that come to my mind are:
> 
> 1. Allow only one "main" clock to provide timestamps
>     * + Absolutely foot-gun free
>     * + Trivial to implement
>     * - Which should be the "main" clock? What when two modules have a conflicting requirements regarding the "main" clock?
>     * - manually converting between time scales is really meh, especially since ztimer implements the conversion logic fine
> 2. Allow all clocks derived from the "main" hardware clock to provide timestamps
>     * + a bit more convenient than 1.
>     * - replaces one footgun by another (IMO it is quite surprising that making use of an RTT could result in some clocks loosing their timestamping features)
> 3. Synchronise the ztimer backends
>     * - a horrible idea IMO, since clocks will drift. Mixing timestamps from different sources will result in different durations being reported for the same real duration depending on which clock was used for the start and which for the stop timestamp.
>     * - overhead / complexity
> 4. Always use the same hardware clock for timestamps. (E.g even if ZTIMER_USEC would use periph_timer for timers, still use periph_rtt and frequency conversion for timestamps.)
>     * + no big footguns and relatively easy
>     * - different resolutions for timers and timestamps
>     * - clock drift between timestamps and timers. (If timer duration are dynamically computed from durations measured using the delta of two timestamps, that could be an issue).
> 5. Drop access to timestamps altogether. Instead, allow only to measure durations by providing two functions (for start and stop) which operate on an opaque `struct`. With develhelp the struct would contain a pointer to the clock it was started on and the stop function `assert()`s the same being used there. The stop function would be allowed to be called multiple times for the same starting value and a convenience "epoch" starting value for each clock could be provided for when timestamps are needed.
>     * + No loss of functionality
>     * + No overhead when not using DEVELHELP
>     * + Fixes the footgun
>     * + The convenience epoch values could be compile time constants to allow them to be garbage collected at link time. Also, they could be randomly chosen at built time to ensure mixing "timestamps" from different clocks fails soon and noisy.
>     * - More verbose code
>     * - Some overhead (one pointer and an `asserr()`) when using DEVELHELP

I would like to elaborate a bit more on how option 5 could work:

```C
typedef struct {
    ztimer_now_t value;
#if DEVELHELP
    ztimer_clock_t *clock;
#endif
} ztimer_timepoint_t;

extern const ztimer_timepoint_t ztimer_epoch_msec;
extern const ztimer_timepoint_t ztimer_epoch_usec;
extern const ztimer_timepoint_t ztimer_epoch_sec;

static inline ztimer_assert_tp_is_from_clock(const ztimer_clock_t *clock, const ztimer_timepoint_t *tp)
{
    (void)clock;
#if DEVELHELP
    assert(clock == tp->clock);
#endif
}

static inline ztimer_assert_tps_same_clock(const ztimer_timepoint_t *tp_a, const ztimer_timepoint_t *tp_b)
{
    (void)tp_a;
    (void)tp_b;
#if DEVELHELP
    assert(tp_a->clock == tp_b->clock);
#endif
}

static inline ztimer_timepoint_t ztimer_timepoint(ztimer_clock_t *clock)
{
    ztimer_timepoint_t result = {
        .value = ztimer_now(clock),
#if DEVELHELP
        .clock = clock
#endif
    };
    return result;
}

static inline ztimer_now_t ztimer_duration(const ztimer_timepoint_t *start, const ztimer_timepoint_t *stop)
{
    ztimer_assert_tps_same_clock(start, stop);
    return stop->value - start->value;
}

static inline ztimer_now_t ztimer_duration_now(ztimer_clock_t *clock, const ztimer_timepoint_t *tp)
{
    ztimer_assert_tp_is_from_clock(clock, tp);
    return ztimer_now(clock) - tp->value;
}
```

If one would need many timestamps (so that using `ztimer_timepoint_t` is not practical due to the 4 bytes overhead for each value), one could call `ztimer_duration_now(ZTIMER_MSEC, &ztimer_epoch_msec);` to yield the timestamps without addition RAM consumption (at the expense of one additional subtraction compared to now). By having the epoch values randomly initialized (at compile time, so that the epochs can be garbage collected at link time), mixing those timestamps would directly blow up, even when all clocks are backed by the same hardware. That should result in buggy code being easily detected.

miri64, 2021-08-03:
Here is [my reply](https://github.com/RIOT-OS/RIOT/pull/16697#issuecomment-891288507) from the original discussion:

> In case this won't happen, here are some of my concerns:
> 
>  1. Can lead to potential granularity problems. Alternatively, too high granularity if we actually only need e.g. a scale of minutes for the time stamps (potential for too rapid overflows).
> 
>  2. Same problem, if I understand it correctly.
> 
>  3. I agree with your statement that this is a horrible idea ;-).
> 
>  4. As you pointed out: same problem as with 1 and 2
> 
>  5. Not doable if you want to derive e.g. networking functionality. Sometimes (e.g. in 6LoWPAN and IPv6 reassembly buffer) time of arrival needs to be stored. Also can be quite cumbersome if applications want to e.g. set certain timings for sensors / actuators.  Some notion of "now" also is required for e.g. stuff like [SenML](https://datatracker.ietf.org/doc/html/rfc8428#section-4.5.3).

maribu, 2021-08-03:
> Some notion of "now" also is required for e.g. stuff like [SenML](https://datatracker.ietf.org/doc/html/rfc8428#section-4.5.3).

Note that points in time are always given as a duration since some reference point (epoch). E.g. most humans tend to specify points in time as duration since the birth of guy whose mother was - according to Christian mythology - still a virgin.

The suggestions in 5 makes the reference time point an explicit API argument (rather than an implicit), which I believe would increase awareness that those reference points are actually independent. And if bugs still get in, the `assert()`s will ease debugging significantly.

jue89, 2021-08-03:
In #16327 I proposed to introduce a stop watch that also might be used as a wall clock (which is a stop watch running since 1.1.1970 ;)). This approach also solves an important power management problem: currently ztimer doesn't know if someone is measuring time differences.

So I'd like to opt for @maribu's alternative 5. All other alternatives will introduce drawbacks in terms of power management and/or complexity.

maribu, 2021-08-03:
I think that the API as proposed in 5. wouldn't solve the power management problem (which IMO is more important than the footgun issue), since the API proposed is not concerned about starting / stopping the underlying clock.

But the other way round: If we are replacing the `ztimer_now()` API with a stopwatch API to assist power management, it should be relatively straight forward to add optional `assert()`s to counter confusing clocks for measurements of durations.

miri64, 2021-08-03:
> E.g. most humans tend to specify points in time as duration since the birth of guy whose mother was - according to Christian mythology - still a virgin.

<ot>We should go back to reckoning from when a random Mediterranean city was allegedly build by a twin who killed his brother.</ot>

kfessel, 2021-08-03:
I would like to add some cents:
in #16697 the issue was that having one 64bit ztimer makes all of them 64bit, 
in #16327 I said that i would like to see some "neither ztimer nor xtimer, but maybe using them" system that provides a clock of application selected  granularity approaching  {uint64_t sec;uint64_t ns;} (ithink it is posix -struct time) (this is @maribu 3. in hard i think)

maybe if the stop watch / wall clock hat its own types to extend beyond 32 Bit this could solve most of our issues

@jue89 is there an example for your wall clock / stop watch

jue89, 2021-08-04:
> I think that the API as proposed in 5. wouldn't solve the power management problem (which IMO is more important than the footgun issue), since the API proposed is not concerned about starting / stopping the underlying clock.

Using a dedicated API to measure time differences can be designed to be aware of the current measurement has been finished or still is ongoing. We can solve both problems here! :)

> @jue89 is there an example for your wall clock / stop watch

Not written down, yet. Just in my head. But let's change that!

I am thinking of 4 methods:

```c
void ztimer_stopwatch_start(ztimer_clock_t * clock, ztimer_stopwatch_t * stopwatch);
// Starts the stopwatch.

void ztimer_stopwatch_stop(ztimer_clock_t * clock, ztimer_stopwatch_t * stopwatch);
// Stops the stopwatch.

void ztimer_stopwatch_set(ztimer_clock_t * clock, ztimer_stopwatch_t * stopwatch, ztimer_now_t value);
// Sets the current value of the stopwatch. Setting value `0` can be used to reset the stopwatch. It can be called in started and stopped state.

ztimer_now_t ztimer_stopwatch_read(ztimer_clock_t * clock, ztimer_stopwatch_t * stopwatch);
// Returns the current value of the stopwatch. It can be called in started and stopped state.
```

Measuring time differences:

```c
void measure_time()
{
    ztimer_stopwatch_t sw = ZTIMER_STOPWATCH_INIT(ZTIMER_MSEC);

    ztimer_stopwatch_start(ZTIMER_MSEC, &sw);
    do_heavy_workload();
    ztimer_stopwatch_stop(ZTIMER_MSEC, &sw);

    printf("Heavy workload took %dms.\n", ztimer_stopwatch_read(ZTIMER_MSEC, &sw));
}
```

Using it as a wallclock:

```c
void main()
{
    ztimer_stopwatch_t unixtime = ZTIMER_STOPWATCH_INIT(ZTIMER_SEC);
    ztimer_stopwatch_start(ZTIMER_SEC, &unixtime);
    ztimer_stopwatch_set(ZTIMER_SEC, &unixtime, sntp_get_time("0.de.pool.ntp.org"));
    printf("Current unix time: %d.\n", ztimer_stopwatch_read(ZTIMER_MSEC, &unixtime));
}
```

kaspar030, 2021-08-18:
Is "don't compare timestamps of different clocks" an option?

miri64, 2021-08-24:
> Is "don't compare timestamps of different clocks" an option?

Given that multiple people already tried that and opened PRs on that it is only a matter of time until this footgun is fired even if we go with this option ;-).

chrysn, 2021-09-29:
I'm with @kaspar030 here. If that means that ztimer_now needs to get some friendly bright letters to it that unless you're doing one of N things you should probably do it differently, or even gets unpublished at all, I'd be fine with that.

But this is IMO more a case of unrealistic expectations (that there's no clock drift; that all times use the same epoch) and not of a problem in the APIs. Possibly their documentation, though.

miri64, 2021-09-29:
Maybe there could be both: `ztimer_now()` for low-level stuff (low level meaning also updating DTLS- or DHCPv6-level timeouts), but also something more in the vain that @jue89 proposed on top of that. I'd like to have a timer in application space at least at some point. Something that can handle a more standardized notion of “now”, so stuff like SenML or just setting an alarm clock would not need to rely on `periph_rtc`.

chrysn, 2021-09-29:
A higher-level stopwatch API would be a nice addition.

For low-level stuff I easily see ztimer_now usable -- it just needs to come with warnings like "You can only compare values from the same ztimer, *and* need to consider overflow, *and* forming differences only ever makes sense if you have forced the timer to run continuously between when the compared timestamps were taken". (The latter is *really* tricky to use correctly; if I set a timer, it fired and freed a mutex ready then switched off, I'd have to first get the "now", then look at the mutex, and if it's still locked I can use the time, and otherwise I'll have to assume that it's just "larger" because actually the timer fired before I was fully done).

chrysn, 2021-09-30:
Just a wild idea for footgun mitigation: Can we make the behavior of ztimer_now *worse* deliberately, just to ensure things don't happen to work? (Like, you wouldn't point a gun at your foot if it randomly gave off a shot every few seconds when securities are off).

I'm thinking "initialize each timer with a random value" and "report a sentinel value when queried while the value is undefined" eg. while the timer is not running (both: just while DEVELHELP=1). (I briefly thought about aborting the program when queried while undetermined, but that'd be overly harsh and possibly racy -- you *may* query it, you just may have to check later that the query was justified).

maribu, 2023-01-05:
Fixed by documenting the footgun visibly.

closed 2023-01-05 by maribu