RIOT-OS/RIOT #16714: test/pkg_u8g2 using SDL is failing

kfessel, 2021-08-05:
#### Description

test/pkg_u8g2 using SDL does not work as expected

#### Steps to reproduce the issue

install SDL2 dev packages
```
tests/pkg_u8g2$ make TEST_OUTPUT=2
tests/pkg_u8g2$ make  term 

and start the test
```

#### Expected results

SDL window showing "THIS" "IS" "RIOT"

#### Actual results

terminal showing:
```
...
<RIOT>/tests/pkg_u8g2/bin/native/tests_pkg_u8g2.elf: not implemented
...
<RIOT>/tests/pkg_u8g2/bin/native/tests_pkg_u8g2.elf: not implemented
Segmentation fault (core dumped)
```

the "not implemented" message seems to come from cpu/native/syscalls.c : getpid() which seem to be called from SDL

make debug having a breakpoint there yields:
```
#0  getpid () at <RIOT>/cpu/native/syscalls.c:421
#1  0xf7ef9015 in ?? () from /lib/i386-linux-gnu/libSDL2-2.0.so.0
#2  0xf7ef9201 in ?? () from /lib/i386-linux-gnu/libSDL2-2.0.so.0
#3  0xf7eddd36 in ?? () from /lib/i386-linux-gnu/libSDL2-2.0.so.0
#4  0xf7e26a3a in ?? () from /lib/i386-linux-gnu/libSDL2-2.0.so.0
#5  0x5655b8c1 in u8g_sdl_init (width=width@entry=128, height=height@entry=64) at <RIOT>/build/pkg/u8g2/sys/sdl/common/u8x8_d_sdl_128x64.c:100
#6  0x5655bced in u8x8_d_sdl_128x64 (u8g2=0x56568d80 <main_stack+11616>, msg=9 '\t', arg_int=0 '\000', arg_ptr=0x0) at <RIOT>/build/pkg/u8g2/sys/sdl/common/u8x8_d_sdl_128x64.c:274
#7  0x5655b6e7 in u8x8_SetupMemory (u8x8=0x56568d80 <main_stack+11616>) at <RIOT>/build/pkg/u8g2/csrc/u8x8_display.c:95
#8  0x5655bf27 in u8x8_Setup_SDL_128x64 (u8x8=0x56568d80 <main_stack+11616>) at <RIOT>/build/pkg/u8g2/sys/sdl/common/u8x8_d_sdl_128x64.c:372
#9  0x5655bf71 in u8g2_SetupBuffer_SDL_128x64_4 (u8g2=0x56568d80 <main_stack+11616>, u8g2_cb=0x5655eaa0 <u8g2_cb_r0>) at <RIOT>/build/pkg/u8g2/sys/sdl/common/u8x8_d_sdl_128x64.c:390
#10 0x5655670f in main () at <RIOT>/tests/pkg_u8g2/main.c:122
```

#### Versions
RIOT master (commit 1b0152bd73b93efd6a723640c873aa8918f4ac80)
libSDL 2.0.12+dfsg1-4 (current UBUNTU/groovy)

benpicco, 2021-08-09:
Looks like this is the same issue as #13501

kfessel, 2021-10-07:
Fixed the main issue with #16954

application may need increased stack size see pkg/u8g2/doc.txt

closed 2021-10-07 by kfessel