RIOT-OS/RIOT #16749: particle: error while flashing using Docker build and DFU mode

nicoHarel, 2021-08-16:
#### Description

Hello everyone. I've decided to switch to Docker recently since I'm tired of using VMs. It's awesome that RIOT offers that possibility. 

However I've been running into a problem:
When I try to flash my particle board (Xenon) using the command under, I get a missing file error.
`make flash BUILD_IN_DOCKER=1 BOARD=particle-xenon PARTICLE_MONOFIRMWARE=1`

I know in this mode the checksum needs to be verified at startup so it of course makes sense that this file is needed. Yet it simply isn't there when building with Docker. 

#### Steps to reproduce the issue

- Go to examples/hello-world
- Use the following command: `make flash BUILD_IN_DOCKER=1 BOARD=particle-xenon PARTICLE_MONOFIRMWARE=1`

#### Expected results
Make flash runs without issues. 

#### Actual results
The error: 
`Could not open file /my_file_path/RIOT/examples/hello-world/bin/particle-xenon/hello-world.bin-checksummed for reading: No such file or directory
`

I've checked, the file doesn't exist. It is there if I build without Docker.

Have a good week everybody.

dylad, 2021-09-14:
@nicoHarel Are you on Windows ?
I can't reproduce your issue with Linux Mint 19.3.

nicoHarel, 2021-09-15:
> @nicoHarel Are you on Windows ?
> I can't reproduce your issue with Linux Mint 19.3.

No, I'm on Fedora. And it is probably only occurring on Fedora.
I'm also unable to reproduce on Linux Mint.

dylad, 2021-09-15:
I was able to reproduce on Linux Mint after all, I had to run make clean before and the same error appears.
@nicoHarel could you check if  #16857 fixes this issue please ?

closed 2021-09-15 by chrysn

nicoHarel, 2021-09-16:
Yeap, perfect. Thanks.

PR #16857 (2021-09-15): makefiles/docker.inc.mk: pass PARTICLE_MONOFIRMWARE to docker env vars

### Contribution description

This PR allows to pass `PARTICLE_MONOFIRMWARE` to `DOCKER_ENV_VARS`
This will allow the generation of the bin-checksummed file needed by `dfu-util` flashtool and thus allow to flash the all particle-based board with RUN_IN_DOCKER=1

### Testing procedure

You can test it without owning the hardware.
`make BUILD_IN_DOCKER=1 BOARD=particle-xenon PARTICLE_MONOFIRMWARE=1 -C examples/hello-world clean `
`make BUILD_IN_DOCKER=1 BOARD=particle-xenon PARTICLE_MONOFIRMWARE=1 -C examples/hello-world fla [... cut]