RIOT-OS/RIOT #16756: Manual init for ztimer

ant9000, 2021-08-19:
#### Description

I would like to have a simple way for manually initializing ztimer: a simple call to `ztimer_init()` for instance.

#### Use case

The RTC subsystem of CPUs in SAM0 family is endowed with a few 32 bit registers marked as "general purpose" - they are meant for users to store data in (not all CPUs in the family sport them; existence can be tested with `#ifdef RTC_GPR_NUM`).

My specific use case is to store state that persist after deep sleep; which is called BACKUP mode on SAML21/SAMR34 CPUs. In backup mode the RTC clock is still powered - so any data in those registers will be available when waking up again.

Now, for the problem: resetting the RTC wipes those registers, and in RIOT this happens in on RTT/RTC initialization.

For solving this, I added

```
DISABLE_MODULE += periph_init_rtt
```
to my application Makefile, and persist manually the values at boot:

```
#ifdef RTC_GPR_NUM
    uint32_t persist[RTC_GPR_NUM];
    for (int i = 0; i < RTC_GPR_NUM; i++) persist[i] = RTC->MODE0.GP[i].reg;
    rtt_init();
    for (int i = 0; i < RTC_GPR_NUM; i++) RTC->MODE0.GP[i].reg = persist[i];
#else
    rtt_init();
#endif
```

This approach works well with the `periph_rtt` module, but I cannot find a way to produce the same solution with `ztimer`. Which is like my fault: I can't really understand how ztimer auto init can be disabled, and how to manually init the timers.

ant9000, 2021-08-19:
A different solution, only for my specific use case, would be to introduce a config variable, something like `CONFIG_SAM0_PERIPH_RTT_PERSIST_GPR`, and wrapping the RTC reset code with the snippet above.
I could easily provide a PR for this one.

kaspar030, 2021-08-19:
> A different solution, only for my specific use case, would be to introduce a config variable, something like `CONFIG_SAM0_PERIPH_RTT_PERSIST_GPR`, and wrapping the RTC reset code with the snippet above.
> I could easily provide a PR for this one.

You mean, wrap but inside rtt_init()? Sounds best to me.

closed 2021-10-01 by ant9000