RIOT-OS/RIOT #16801: [GPIO] - External interrupt

ksadura, 2021-09-02:
Hi everyone. Because I wasn't able to post my issue on [RIOT forum](https://forum.riot-os.org/), I put it here. I've got two questions about **```int gpio_init_int()```** function.
1. Is there any infinite loop implemented in this method, that makes a program working constantly? After I put this method into the ```int main()``` function, my programme was kept alive.
2. Second question, I think more crucial to me, is whether a callback function (passed as an argument to this method) executes on the different thread than the rest of code or not. I had a case, when in the callback function I was sending an IPC message - ```msg_t``` to the another thread, let's call it **THREAD2**. When my pin was interrupted for the first time, callback function executed on the thread with proccess ID (```kernel_pid_t```) equaled **1**. Then, after second thread received IPC message and executed its code on the thread with ID = **2**, any next pin interruption caused, that callback function always executed on the same thread as **THREAD2** with ID = **2**. Why there was no thread disitinction between those two processes?

Thanks in advance

jeandudey, 2021-09-16:
Hello  @ksadura !

>Is there any infinite loop implemented in this method, that makes a program working constantly? After I put this method into the int main() function, my programme was kept alive.

Depends on the board/cpu you're using, sometimes using a pin that is not supported to be initialized as an interrupt will an `assert` checking for that will be reached.

> Second question, I think more crucial to me, is whether a callback function (passed as an argument to this method) executes on the different thread than the rest of code or not. I had a case, when [...]

The callback function is executed on an ISR context which is a good idea to keep these callbacks simple enough to not introduce any latency into the system and keep real-time guarantees.

The PID for messages sent from an ISR is [`THREAD_PID_ISR`](https://api.riot-os.org/group__core__msg.html#gadba0f57bc141ce3a7ec95fbd4bf26429) which is calculated at compile time, however if you try to get the pid of an ISR with `thread_getpid` you'll probably get the PID of the last scheduled thread (I'm not sure though).

To send a [`msg_t`](https://api.riot-os.org/structmsg__t.html) from an ISR you can use [`msg_send_int`](https://api.riot-os.org/group__core__msg.html#ga2762c096056c5eac7ffe38c8727e033a).

maribu, 2023-05-17:
I think @jeandudey did answer your question. If you still have further questions, feel free to ask them or to reopen the issue.

closed 2023-05-17 by maribu