RIOT-OS/RIOT #16832: can webusb support be added

pathnirvana, 2021-09-09:
#### Description
connecting a dev board to a computer and be able to configure the app via javascript (nodejs or website) is very attractive to beginners. I am not sure if it is technically possible to add webusb to riot nor such a feature matches with the philosopy of this project.
If not feel free to close this issue.

### Useful links
Zephyr demo app https://docs.zephyrproject.org/latest/samples/subsys/usb/webusb/README.html

chrysn, 2021-09-13:
This should be well within what's possible with the current USB stack, and primarily a matter of setting the right descriptors.

My gut feeling is that this will even be more effort on the browser side (to get a WebUSB demo application to run with it) than on the RIOT side.

pathnirvana, 2021-09-13:
There are many web demos written in JS that can connect to devices already. It is pretty straightforward on the browser side.

chrysn, 2021-09-13:
Is there a particular demo application you'd suggest for whoever'd start implementing this? Zephyr points to <https://github.com/finikorg/webusb-sample>, is that a good demo?

pathnirvana, 2021-09-13:
Yes it is a good demo. Also the TinyUSB demo https://example.tinyusb.org/webusb-serial/
They both do the same thing, send something to the mcu and mcu echoes it back to the browser.

chrysn, 2021-09-13:
I won't have time to work on this right now, but expecting it to be straightforward enough, I've marked this as "good first issue", and offer to help with questions arising if you (or any other novice contributor) want to give it a try.

pathnirvana, 2021-09-13:
I myself am a web developer who got interested in the embedded recently. I would be happy to help on the JS front to write a demo app if needed.

chrysn, 2021-09-13:
Having looked around a bit, there are applications for [DFU](https://github.com/devanlai/webdfu) and [serial](https://example.tinyusb.org/webusb-serial/) as you linked, along with a lot of custom ones. Do you happen to know if anyone has ventured into implementing the host side of CDC-ECM (USB Ethernet) on the Javascript / WASM side?

(That could make this a useful thing for setting up security in RIOT based systems even when the user does not have root access on the host; CDC-ACM aka serial might work too but that'd mean that the embedded device would need two separate services).

chrysn, 2021-09-13:
What I meant was that given there are implementations of USB Serial in JavaScript, there could also be an implementation of Ethernet in JavaScript. And like USB Serial over JavaScript should work immaterial of whether the OS deals with serial devices well, doing USB Ethernet in JavaScript could allow some limited use of USB Ethernet devices on Windows or other badly configured systems. (It wouldn't route out, but a typical demo application might listen for route advertisements or send pings down a network of which the operating system doesn't even notice it's there).

pathnirvana, 2021-09-13:
JS running on the browser have limited access to operating system unlike other programming languages. As long as there is an ip address JS can send requests and receive responses.

chrysn, 2021-09-13:
A JS implementation of CDC-ECM would use its own IP stack, like a has become fashion in some areas with Unikernels (and inside this "virtual" world be allowed to send arbitrary messages) -- but let's leave that discussion to when there's actually basic WebUSB support (which should be the easy part).

pathnirvana, 2021-09-13:
I think this would work with nodejs but not inside the browser. Take a look at this
https://stackoverflow.com/questions/17999950/communicating-with-arduino-over-ethernet-using-node-js