RIOT-OS/RIOT #16855: gcoap: API revamp - Abstracting nanocoap

miri64, 2021-09-14:
During the summit [we discussed a general API overhaul for our CoAP infrastructure](https://forum.riot-os.org/t/coap-api-revamp/3356). While the overall goal is a complete redesign of the gCoAP API that would for now fork from the original gCoAP, there is also work to be done for good old gCoAP:

> * Compatibly (with deprecations over release cycles) get rid of direct member access in nanocoap (using static inline accessors instead).

This issue serves as a tracking issue towards this goal and how I imagine the road forward there:

- [ ] Provide `gcoap`-based getter/setter functions for `nanocoap` data types.
- [ ] Replace direct access to members of those `nanocoap` data types getter/setter functions.
- [ ] Deprecate direct assess to members of `nanocoap` data types
- [ ] Move gcoap-specific members of `nanocoap` data types to new `gcoap` types that inherit from the original `nanocoap` types. In an in-between step this will most likely require a lot of casting, since the `gcoap` functions, for now, should keep their (`nanocoap`-based) parameter types.
- [ ] Provide new `gcoap` functions and types that use the new `gcoap` types, provide wrappers for old `nanocoap`-types-based functions and types.
- [ ] Deprecate `nanocoap`-types-based wrappers

chrysn, 2021-09-16:
One detail: I'd rather think that the getter/setters would first be introduced in nanocoap (static inline; we have `coap_code` and would gain `coap_set_token` (or _rebuild_header? will become clear from usage)).

Then, we'd see which of these can have trivial wrappers (technically all b/c gcoap-1.0 is not going away from nanocoap) that just cast-and-call (again all static inline, so in the end it's member access again), and which access members that are now moved.

miri64, 2021-09-16:
> One detail: I'd rather think that the getter/setters would first be introduced in nanocoap (static inline; we have `coap_code` and would gain `coap_set_token` (or _rebuild_header? will become clear from usage)).

Not sure this fits into this discussion. These getters and setters are more about nanocoap functionality and API. This tracking issue is mostly about `gcoap`'s usage of `nanocoap` and how to go forward to clean-out `gcoap`-stuff out of `nanocoap`.