RIOT-OS/RIOT #16866: Flashing of STM32 Nucleo device doesn't work

ubik-valis, 2021-09-16:
Hello there, I'm currently trying to flash my nucleo-f446re device with the hello-world example, but run into the following error:

```
### Flashing Target ###
Open On-Chip Debugger 0.11.0+dev-00331-g53556fcde-dirty (2021-08-19-14:30)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_assert_srst

Info : clock speed 2000 kHz
Info : STLINK V2J30M19 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.024852
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 0
Info : Listening on port 35683 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f4x.cpu       hla_target little stm32f4x.cpu       reset

Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800067c msp: 0x20000200
Info : device id = 0x10006421
Info : flash size = 512 kbytes
auto erase enabled
wrote 16384 bytes from file /home/alexstrauch/Documents/RIOT/examples/hello-world/bin/nucleo-f446re/hello-world.elf in 0.646139s (24.762 KiB/s)

verified 9256 bytes in 0.104261s (86.696 KiB/s)

Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
shutdown command invoked
Done flashing
/home/alexstrauch/Documents/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM0" -b "115200"  
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2021-09-16 21:20:33,066 # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type '/exit' to exit.
```
The problem is that I get no hello world message, the program is just stuck there. Could you please explain to me whether this problem occurred in the past?

Thank you all

kfessel, 2021-09-17:
flashing worked
```
wrote 16384 bytes from file ...
verified 9256 bytes in 0.104261s (86.696 KiB/s)
```

The device just booted faster and showed the message before the pyterm connected.

the nucleos have reset switches just press it while pyterm is running  and you will seen the program output

ubik-valis, 2021-09-19:
@kfessel thank you! That was it :)

closed 2021-09-19 by ubik-valis