RIOT-OS/RIOT #16873: Kconfig/tinydtls: Unable to compile `examples/dtls-sock` with `CONFIG_DTLS_ECC` enabled

miri64, 2021-09-21:
#### Description
When trying to build `examples/dtls-sock` with the ECC cyphersuit, the build fails. I am not sure if this is a `tinydtls` (configuration) problem or a `Kconfig` problem.

#### Steps to reproduce the issue
Add the following `app.config` to the `examples/dtls-sock` directory:

```Kconfig
CONFIG_KCONFIG_USEPKG_TINYDTLS=y
CONFIG_DTLS_ECC=y
CONFIG_DTLS_PSK=n
```

and try to build

```sh
make -C examples/dtls-sock/ flash -j
```

#### Expected results
The application should build.

#### Actual results
The build fails due to the `IS_ACTIVE(CONFIG_DTLS_PSK)` branch in `dtls-client.c` being taken at one point:

```
examples/dtls-sock/dtls-client.c: In function ‘client_send’:
examples/dtls-sock/dtls-client.c:169:24: error: ‘credential1’ undeclared (first use in this function); did you mean ‘credential0’?
  169 |     res = credman_add(&credential1);
      |                        ^~~~~~~~~~~
      |                        credential0
dtls-sock/dtls-client.c:169:24: note: each undeclared identifier is reported only once for each function it appears in
dtls-sock/dtls-client.c:183:45: error: ‘_client_psk_cb’ undeclared (first use in this function)
  183 |     sock_dtls_set_client_psk_cb(&dtls_sock, _client_psk_cb);
      |                                             ^~~~~~~~~~~~~~

```

#### Versions
Current master (99e614b6d2d4155a736b744). I tested this on `native`. Compilation and functionality worked in 57533a76fdab9a948b528b549fff, where the app was first introduced, but `tinydtls` did not use Kconfig at that point. Did not bisect further.

leandrolanzieri, 2021-09-21:
I think PSK and ECC are mutually exclusive in that application, as you point out with dtls-client.c. It even seems to be the case in https://github.com/RIOT-OS/RIOT/commit/57533a76fdab9a948b528b549ffff8553f858f99, as it has:
```C

#ifdef DTLS_ECC
static const ecdsa_public_key_t other_pubkeys[] = {
    { .x = ecdsa_pub_key_x, .y = ecdsa_pub_key_y },
};

static const credman_credential_t credential = {
    .type = CREDMAN_TYPE_ECDSA,
    .tag = SOCK_DTLS_CLIENT_TAG,
    .params = {
        .ecdsa = {
            .private_key = ecdsa_priv_key,
            .public_key = {
                .x = ecdsa_pub_key_x,
                .y = ecdsa_pub_key_y,
            },
            .client_keys = (ecdsa_public_key_t *)other_pubkeys,
            .client_keys_size = ARRAY_SIZE(other_pubkeys),
        }
    },
};

#else /* ifdef DTLS_PSK */
static const uint8_t psk_id_0[] = PSK_DEFAULT_IDENTITY;
static const uint8_t psk_key_0[] = PSK_DEFAULT_KEY;

static const credman_credential_t credential = {
    .type = CREDMAN_TYPE_PSK,
    .tag = SOCK_DTLS_CLIENT_TAG,
    .params = {
        .psk = {
            .key = { .s = psk_key_0, .len = sizeof(psk_key_0) - 1, },
            .id = { .s = psk_id_0, .len = sizeof(psk_id_0) - 1, },
        }
    },
};
#endif
```
I don't think it's related to the Kconfig option, but to the application itself.

leandrolanzieri, 2021-09-21:
We can add a check for this in the application Makefile perhaps.

miri64, 2021-09-21:
Yes, `tinydtls`'s Kconfig models this as a `choice`. Not sure if that is the problem.

leandrolanzieri, 2021-09-21:
> Yes, `tinydtls`'s Kconfig models this as a `choice`. Not sure if that is the problem.

I need to double check, but I believe tinyDTLS supports both suites at the same time, here is just the way the app is coded, we could also modify it to make it work with both?

leandrolanzieri, 2021-09-21:
Sorry @miri64 , I misunderstood the problem, a fix is provided in #16874. Still I hold the idea that the application should be modified to support both suites.

miri64, 2021-09-21:
> Sorry @miri64 , I misunderstood the problem, a fix is provided in #16874. Still I hold the idea that the application should be modified to support both suites.

Yes, but let's first make the application compileable again.

PR #16874 (2021-09-21): pkg/tinydtls: fix detection of Kconfig usage
### Contribution description
As reported in #16873, the tinyDTLS example fails to build with Kconfig, because the Makefile fails to detect that Kconfig is being used, and injects CFLAGS. This makes PSK and ECC to be built, which is not supported by the example application so far.

### Testing procedure
You can follow the procedure described in #16873.

### Issues/PRs references
Fixes #16873.
closed 2021-09-21 by miri64