RIOT-OS/RIOT #16875: [Tracking]: modeling dependencies in Kconfig

fjmolinas, 2021-09-22:
#### Description

These issues want to keep track of `MODULES`/`PKG`/`CPU`/`BOARD`s that are not modelled in Kconfig. 

**no exhaustive-list** please extend:

- [ ] cpu:
  - [ ] cortex-m:
    - [x] stm32
    - [ ] nrf52 https://github.com/RIOT-OS/RIOT/pull/16837
    - [x] cc2538
    - [ ] efm32
    - [ ] ...
  - [x] native
  - [ ] esp32
  - [ ] esp8266
  - [ ] avr
  - [ ] arm7
  - [ ] riscv
  - [ ] mips
  - [ ] msp430
  - [ ] ...
   
- [ ] pkgs:
    - [ ] mynewt-core https://github.com/RIOT-OS/RIOT/pull/16780
    - [ ] uwb-core https://github.com/RIOT-OS/RIOT/pull/16780
    - [ ] uwb-dw1000 https://github.com/RIOT-OS/RIOT/pull/16780
    - [ ] ...

- [ ] sys:
  - [ ] net
  - [ ] ...

- [ ] riotboot:

- [ ] drivers:
  - [ ] ...

- [ ] boards:
   - [ ] ...
   - [ ] ...

fjmolinas, 2021-09-22:
Note AFAIK in `sys` most modules that are pending for modeling are related to `networkin`. Same goes for `pkg` and `drivers`

fjmolinas, 2021-09-22:
@maribu any change you could take a look at avr? Do you seem to be doing a lot of work there?

fjmolinas, 2021-09-22:
Since there is some discussion of eventually dropping MIPS it should be low priority.. (probably the same for MIPS)

MrKevinWeiss, 2021-11-22:
Just to note that we would like to get all the cpus in for this release, this would allow us to stop testing select apps twice and use pure kconfig for dep resolution.

MrKevinWeiss, 2021-12-13:
Just so it is written down somewhere.  The configs we are using to set things for the STDIO_USB_ACM_CDC stuff I think will fail on apps that don't use stdio.  They get hardcoded in by the <board>.config file but they really should only be brought in if there is a MODULE_STDIO. 

We will eventually have to address this (though maybe not now).

Enoch247, 2022-01-21:
I wanted to make an offer to help out with kconfig. I'm pretty confident with Makefiles and kconfig's syntax. Any place in particular I should start?

MrKevinWeiss, 2022-01-22:
Heck ya @Enoch247, if you didn't read the [kconfig docs](https://doc.riot-os.org/kconfig-in-riot.html) I would start with that.

Currently I am working on getting all boards (that are not deprecated) to run on some applications.  There are still [some boards](https://github.com/RIOT-OS/RIOT/blob/2520b5c608d393947b76b3ca8329b558c1c59dc6/.murdock#L108) that will not work on the [approved applications list](https://github.com/RIOT-OS/RIOT/blob/2520b5c608d393947b76b3ca8329b558c1c59dc6/.murdock#L135), but there are already PRs out there to fix it.

The next step would be to add more applications to the `TEST_KCONFIG_TEST_ALLOWLIST`, this will expose more modelling corner cases. Note that adding more applications will also increase the CI time so we didn't want to overdo it.  Once we are confident that the modelling is correct with kconfig we can only build test with kconfig and skip the make dependency resolution, reducing the CI time.

Though I have been talking about it for a while, I also intend on documenting some design patterns that we want to use in kconfig, specifically targeting drivers that can be selected by "default" based modules such as `SAUL_DEFAULT` or `TOUCH_DEV`.  There will be some work getting everything to the same standard once that comes in.

The final thing that is missing will be modelling the network dependencies, which will be pretty challenging as you need to understand the network stacks as well as kconfig. I image we will start working on that pretty soon.  I know we all want to complete this migration, and it seems like things are picking up.

@leandrolanzieri @fjmolinas and I will be happy to answer any questions and provide help.  The best thing you can do is document the questions and answers in a concise and add it to the docs.  I know I have asked the same things from @leandrolanzieri several times already.

Enoch247, 2022-02-04:
I finally had a chance to poke around on this. As a simple test, I tried `make -C examples/saul menuconfig` then played around with the shell's config symbols. Everything worked as expected, so I don't really have any questions so far. I'm struggling to find any concrete task to sink my teeth into here. I noticed riotboot is on the checklist above and is an area I was tinkering with earlier today, so I could work on adding the appropriate Kconfig files for that if nothing else, unless you have something else you would prefer I start with.

When I do have questions, I figured I'd document them and the answers in a gist or something, then we can figure out how to put them into something more formal once I get a bit further into it. Unless you had something else in-mind.

MrKevinWeiss, 2022-02-04:
> I figured I'd document them and the answers in a gist or something

Perfect.

I have started a [design patterns PR](https://github.com/RIOT-OS/RIOT/pull/17556), we can probably add some things in there.

I haven't looked into riotboot, but keep in mind the networking is not yet modelled which may be a requirement for at least some of riotboot.

Other than that, good luck!

aabadie, 2023-05-20:
I edited the issue with new items and marked the ones that are in progress or already done.

benpicco, 2024-01-04:
As per the last VMA we decided to [drop Kconfig](https://forum.riot-os.org/t/last-chance-for-kconfig-dependency-modelling/4068), so this issue can now be closed.

closed 2024-01-04 by benpicco