RIOT-OS/RIOT #16891: event/timeout: fixed dependency on ZTIMER_USEC preventing low-power operation

haukepetersen, 2021-09-24:
#### Description
For testing some border router stuff I just wanted to re-flash one of my very-low-power nodes with a current firmware. But it turns out,  that with current master I am not able anymore to get my low-power build to work... To recap: key to this build was to get rid of any `xtimer` and `ztimer_usec` depencency, as a running periph timer does prevent the Nordic from going to low power... 

The reason that is blocking my build now is the usage of `event/timeout` in `gcoap`. More precisely: the recent `ztimer`-enabling of `event/timeout` (#16505) introduces a fixed dependency to `ztimer_usec` - leading to `gcoap` always pulling in and enabling the periph timer...

Possible solution:
The API changes/additions to `event/timeout` already contain new `ztimer`-specific functions that let the user decide for the time backend to use. So the problem is the mapping of the existing API functions, which are however always included in the build. The key to solving this problem lies IMHO in separating the `ztimer` and non-ztimer parts of the `event/callback` API more strictly: modules using this library should either use only the `event_timeout_ztimer_x()` API and declare that dependency by using `event_timeout_ztimer`, or they should use the old, usec-based API, declaraing this by depending on `event_timeout`. Now it becomes possible to use that module without forcing `ZTIMER_USEC` into the build...

#### Steps to reproduce the issue
build anything with `gcoap`, `xtimer` or `ztimer_usec` are always pulled into the build

#### Expected results
Use `event/timeout` and  `gcoap` in low-power applications

#### Actual results
low-power not possible with `gcoap` at the moment

#### Versions
master

fjmolinas, 2021-09-27:
> The reason that is blocking my build now is the usage of event/timeout in gcoap. More precisely: the recent ztimer-enabling of event/timeout (#16505) introduces a fixed dependency to ztimer_usec - leading to gcoap always pulling in and enabling the periph timer...

I'm not sure how this could be related, that PR just enables ztimer as a backend for `event_timeout`, but `xtimer` was a forced dependency before, so there was already a running timer before no? Doesn't this suggest that `gcoap` is now depending on something using `event_timeout` which was not the case before?

fjmolinas, 2021-11-29:
> The reason that is blocking my build now is the usage of event/timeout in gcoap. More precisely: the recent ztimer-enabling of event/timeout (#16505) introduces a fixed dependency to ztimer_usec - leading to gcoap always pulling in and enabling the periph timer...

How did you resolve the dependency on `usec` for observations for `gcoap`?

kfessel, 2021-11-29:
#17141 might improve on that

fjmolinas, 2021-11-29:
> #17141 might improve on that

How? AFAIK `usec` is still required, so I'm wondering if @haukepetersen had patched this part in his branch.

kfessel, 2021-11-29:
these usec uses are just nows -> they will not block power saving (if configured correctly)

fjmolinas, 2021-11-29:
> these usec uses are just nows -> they will not block power saving (if configured for i don't care for long term usec now values short term is enough)

I don't understand, doesn't a running timer block power savings?

kfessel, 2021-11-29:
yes but having ztimer_usec does not mean to have it running (if there is no timer on it the layered pm is unblocked (this will may stop the clock which is the reason that I allways talk about ztimer_now isn't save to use for time meassurements (you need to block the pm in an other way (default all power modes but idle are blocked keeping usec running as soon as you try to meassure time with usec now  while nothing else is running on that and u activate pm_layerd you should not think of it as meassure time  but time the cpu was in idle or higher)

kfessel, 2022-02-18:
@haukepetersen i think this i fixed by #16958

haukepetersen, 2022-05-11:
Yes :-)

closed 2022-05-11 by haukepetersen