RIOT-OS/RIOT #16896: tests/thread_float: broken on AVR

benpicco, 2021-09-27:
#### Description

#### Steps to reproduce the issue

Run `tests/thread_float` on a board with an AVR8 CPU

#### Expected results

```
main(): This is RIOT! (Version: 2021.10-devel-775-gdc4744-HEAD)
THREADS CREATED

THREAD 3 start
T(3): 141.466812
T(3): 141.490616
T(3): 141.521576
T(3): 141.559769
T(3): 141.605148
T(3): 141.657516
T(3): 141.717255
T(3): 141.783844
T(3): 141.857513
T(3): 141.938278
T(3): 142.026108
T(3): 142.120789
T(3): 142.222488
T(3): 142.331390
T(3): 142.446899
T(3): 142.569565
T(3): 142.699020
T(3): 142.835480
T(3): 142.978668
THREAD 4 start
THREAD 5 start
T(5): 141.521576
T(5): 141.559769
T(5): 141.605148
T(5): 141.657516
T(5): 141.717255
T(5): 141.783844
T(5): 141.857513
T(5): 141.938278
T(5): 142.026108
T(5): 142.120789
T(5): 142.222488
T(5): 142.331390
T(5): 142.446899
T(5): 142.569565
T(5): 142.699020
T(5): 142.835480
T(5): 142.978668
T(5): 143.128845
T(5): 143.285721
make: *** [/home/benpicco/dev/RIOT/tests/thread_float/../../Makefile.include:807: term] Floating point exception (core dumped)
```

#### Actual results

```
2021-09-27 14:25:25,632 # main(): This is RIOT! (Version: 2021.10-devel-775-gdc4744-HEAD)
2021-09-27 14:25:25,633 # THREADS CREATED
2021-09-27 14:25:25,633 # 
2021-09-27 14:25:25,802 # 

2021-09-27 14:25:25,809 # main(): This is RIOT! (Version: 2021.10-devel-775-gdc4744-HEAD)
2021-09-27 14:25:25,810 # THREADS CREATED
2021-09-27 14:25:25,810 # 
2021-09-27 14:25:25,979 # 

2021-09-27 14:25:25,986 # main(): This is RIOT! (Version: 2021.10-devel-775-gdc4744-HEAD)
2021-09-27 14:25:25,987 # THREADS CREATED
2021-09-27 14:25:25,987 # 
2021-09-27 14:25:26,156 # 

2021-09-27 14:25:26,163 # main(): This is RIOT! (Version: 2021.10-devel-775-gdc4744-HEAD)
2021-09-27 14:25:26,164 # THREADS CREATED
2021-09-27 14:25:26,164 # 
2021-09-27 14:25:26,333 # 

2021-09-27 14:25:26,340 # main(): This is RIOT! (Version: 2021.10-devel-775-gdc4744-HEAD)
2021-09-27 14:25:26,341 # THREADS CREATED
2021-09-27 14:25:26,341 # 
2021-09-27 14:25:26,510 # 

2021-09-27 14:25:26,517 # main(): This is RIOT! (Version: 2021.10-devel-775-gdc4744-HEAD)
2021-09-27 14:25:26,518 # THREADS CREATED
2021-09-27 14:25:26,518 # 
2021-09-27 14:25:26,687 # 

2021-09-27 14:25:26,694 # main(): This is RIOT! (Version: 2021.10-devel-775-gdc4744-HEAD)
2021-09-27 14:25:26,694 # THREADS CREATED
```

#### Versions

benpicco, 2021-09-28:
I ran this on `avr-rss2` where I get the reboots, but on `arduino-mega2560` I just get the `?` as you.

maribu, 2021-09-28:
The `?` is expected. Citing https://manpages.debian.org/buster/avr-libc/printf.3avr.en.html:

> If the full functionality including the floating point conversions is required, the following options should be used:
> 
>     -Wl,-u,vfprintf -lprintf_flt -lm

benpicco, 2021-09-28:
I guess it's just an oversight then that the `printf_float` module does not enable that

closed 2021-09-28 by benpicco

maribu, 2021-09-28:
@benpicco Care to open a new issue regarding the `avr-rss2` failure? I think it is unrelated to `printf_float` and related to context switching / scheduling. Maybe a stack overflow or so.

I'll try to find to check if I can reproduce using the `microduino-corerf`.