RIOT-OS/RIOT #16900: drivers/dht : driver not providing full accuracy

wosym, 2021-09-27:
#### Description
DHT11 driver measures temperature and humidity. When using the Arduino example, the temperature is accurate to 0.1°C. However, when using the RIOT example (or using the driver in another application) it is only accurate to 1°C. 

Arduino:
```
23:08:38.844 -> Humidity: 49.00%  Temperature: 26.80°C 80.24°F  Heat index: 27.18°C 80.92°F
23:08:40.863 -> Humidity: 49.00%  Temperature: 26.70°C 80.06°F  Heat index: 27.09°C 80.76°F
23:08:42.915 -> Humidity: 49.00%  Temperature: 26.70°C 80.06°F  Heat index: 27.09°C 80.76°F
23:08:44.934 -> Humidity: 49.00%  Temperature: 26.70°C 80.06°F  Heat index: 27.09°C 80.76°F
23:08:46.954 -> Humidity: 49.00%  Temperature: 26.60°C 79.88°F  Heat index: 27.00°C 80.61°F
23:08:48.973 -> Humidity: 49.00%  Temperature: 26.60°C 79.88°F  Heat index: 27.00°C 80.61°F
23:08:50.992 -> Humidity: 49.00%  Temperature: 26.60°C 79.88°F  Heat index: 27.00°C 80.61°F
23:08:53.010 -> Humidity: 49.00%  Temperature: 26.60°C 79.88°F  Heat index: 27.00°C 80.61°F
23:08:55.062 -> Humidity: 49.00%  Temperature: 26.60°C 79.88°F  Heat index: 27.00°C 80.61°F
```
RIOT:
```
2021-09-27 23:07:27,746 # read
2021-09-27 23:07:27,749 # read
2021-09-27 23:07:27,749 # read
2021-09-27 23:07:27,755 # RAW values: temp:    6401 hum:   13568
2021-09-27 23:07:27,759 # DHT values - temp: 25.0°C - relative humidity: 53.0%
2021-09-27 23:07:29,770 # read
2021-09-27 23:07:29,773 # read
2021-09-27 23:07:29,773 # read
2021-09-27 23:07:29,779 # RAW values: temp:    6401 hum:   13568
2021-09-27 23:07:29,784 # DHT values - temp: 25.0°C - relative humidity: 53.0%
2021-09-27 23:07:31,794 # read
2021-09-27 23:07:31,797 # read
2021-09-27 23:07:31,797 # read
2021-09-27 23:07:31,803 # RAW values: temp:    6401 hum:   13568
2021-09-27 23:07:31,808 # DHT values - temp: 25.0°C - relative humidity: 53.0%
2021-09-27 23:07:33,819 # read
2021-09-27 23:07:33,821 # read
2021-09-27 23:07:33,822 # read
2021-09-27 23:07:33,827 # RAW values: temp:    6401 hum:   13568
2021-09-27 23:07:33,832 # DHT values - temp: 25.0°C - relative humidity: 53.0%
2021-09-27 23:07:35,843 # read
2021-09-27 23:07:35,845 # read
2021-09-27 23:07:35,846 # read
2021-09-27 23:07:35,851 # RAW values: temp:    6401 hum:   14080
2021-09-27 23:07:35,856 # DHT values - temp: 25.0°C - relative humidity: 55.0%
2021-09-27 23:07:37,867 # read
2021-09-27 23:07:37,870 # read
2021-09-27 23:07:37,870 # read
2021-09-27 23:07:37,876 # RAW values: temp:    6401 hum:   14592
2021-09-27 23:07:37,880 # DHT values - temp: 25.0°C - relative humidity: 57.0%
2021-09-27 23:07:39,891 # read
2021-09-27 23:07:39,894 # read
2021-09-27 23:07:39,894 # read
2021-09-27 23:07:39,900 # RAW values: temp:    6401 hum:   15104
2021-09-27 23:07:39,904 # DHT values - temp: 25.0°C - relative humidity: 59.0%
2021-09-27 23:07:41,915 # read
2021-09-27 23:07:41,918 # read
2021-09-27 23:07:41,918 # read
2021-09-27 23:07:41,924 # RAW values: temp:    6402 hum:   17152
2021-09-27 23:07:41,929 # DHT values - temp: 25.0°C - relative humidity: 67.0%
2021-09-27 23:07:43,939 # read
2021-09-27 23:07:43,942 # read
2021-09-27 23:07:43,942 # read
2021-09-27 23:07:43,948 # RAW values: temp:    6403 hum:   18432
2021-09-27 23:07:43,953 # DHT values - temp: 25.0°C - relative humidity: 72.0%
2021-09-27 23:07:45,963 # read
2021-09-27 23:07:45,966 # read
2021-09-27 23:07:45,966 # read
2021-09-27 23:07:45,972 # RAW values: temp:    6404 hum:   19200
2021-09-27 23:07:45,977 # DHT values - temp: 25.0°C - relative humidity: 75.0%
2021-09-27 23:07:47,987 # read
2021-09-27 23:07:47,990 # read
2021-09-27 23:07:47,991 # read
2021-09-27 23:07:47,996 # RAW values: temp:    6406 hum:   19456
2021-09-27 23:07:48,001 # DHT values - temp: 25.0°C - relative humidity: 76.0%
2021-09-27 23:07:50,012 # read
2021-09-27 23:07:50,014 # read
2021-09-27 23:07:50,015 # read
2021-09-27 23:07:50,020 # RAW values: temp:    6408 hum:   19456
2021-09-27 23:07:50,025 # DHT values - temp: 25.0°C - relative humidity: 76.0%
2021-09-27 23:07:52,036 # read
2021-09-27 23:07:52,038 # read
2021-09-27 23:07:52,039 # read
2021-09-27 23:07:52,044 # RAW values: temp:    6656 hum:   19968
2021-09-27 23:07:52,049 # DHT values - temp: 26.0°C - relative humidity: 78.0%
2021-09-27 23:07:54,060 # read
2021-09-27 23:07:54,062 # read
2021-09-27 23:07:54,063 # read
2021-09-27 23:07:54,069 # RAW values: temp:    6659 hum:   19968
2021-09-27 23:07:54,073 # DHT values - temp: 26.0°C - relative humidity: 78.0%
```

Note that while running this test I had my finger on the sensor to slowly heat it up. What is remarkable in the logs, is that after rising gradually, it suddenly jumps from 6408 to 6656 (+200). Or in a previous test: from 6665 to 6914 (+300). 
These jumps make the measured value go from 25 to 26°C or 26 to 27°C respectively.

When checking this in binary-form it becomes noticeable that the 8 LSB barely change, but the lowest bits in the MSB shift and/or get toggled. It appears the 8 LSB are not read correctly. But I could not find why this is. 

#### Steps to reproduce the issue
Run the DHT11 example. 

#### Expected results
0.1°C accurate

#### Actual results
1°C accurate

#### Versions
```
Operating System Environment
----------------------------
         Operating System: "Arch Linux" 
                   Kernel: Linux 5.13.13-arch1-1 x86_64 unknown
             System shell: GNU bash, version 5.1.8(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.1.8(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (GCC) 11.1.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 11.2.0
                  avr-gcc: avr-gcc (GCC) 11.2.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 12.0.1

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "4.1.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: "2.2.0"
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.21.2
                 cppcheck: missing
                  doxygen: missing
                      git: git version 2.33.0
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.11.0
                   python: Python 3.9.6
                  python2: Python 2.7.18
                  python3: Python 3.9.6
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

benpicco, 2021-09-28:
Does this patch work for you?

```patch
--- a/drivers/dht/dht.c
+++ b/drivers/dht/dht.c
@@ -189,8 +189,8 @@ int dht_read(dht_t *dev, int16_t *temp, int16_t *hum)
         DEBUG("RAW values: temp: %7i hum: %7i\n", (int)raw_temp, (int)raw_hum);
         switch (dev->params.type) {
             case DHT11:
-                dev->last_val.temperature = (int16_t)((raw_temp >> 8) * 10);
-                dev->last_val.humidity = (int16_t)((raw_hum >> 8) * 10);
+                dev->last_val.temperature = (int16_t)((10 * raw_temp) >> 8);
+                dev->last_val.humidity = (int16_t)((10 * raw_hum) >> 8);
                 break;
             /* DHT21 == DHT22 (same value in enum), so both are handled here */
             case DHT22:

```

wosym, 2021-09-28:
@benpicco Thanks for thinking along!

Unfortunately, it did not fix it. It was also one of the possible solutions that I had considered, but since the bug appears to already exist on line 189 (the DEBUG print), it can't be inside of that calculation. 

There isn't much else happening before that line. Just the `_read()` and the checksum calculation. So it must be somewhere in the `_read()` I reckon...

wosym, 2021-10-01:
I did some more investigating. The datasheet confused the heck out of me, so I tried to take the common sense approach. I noticed that the lowest byte was pretty much ignored. only the lower nibble was filled in, so after doing another 8 shifts right all the data disappeared. 

So... I tried shifting the lower 8 bits of the 16-bit raw value 4 times to the left. This gave me a pretty accurate value compared to what the Adafruit driver gave me in the Arduino IDE. **However** something that bugged me a lot was that all values between x.0 and x.8 were visible, but never x.9! So e.g from 26.8°C it jumped to 27.0°C. I thought it was coincidence at first, but I could not for the life of me get it to that x.9 value. While with the Adafruit library this worked without a flaw. 

I took another look at the datasheet, and tried to interpret it entirely different than I did before. I noticed it explicitly said T_upper is the integral part, while T_lower is the decimal part. So... what the current driver does is pretty wrong. It combines both numbers in a single 16-bit int... 

After correcting this, it worked flawlessly! (I'm guessing the same mistake was made for humidity though, so I will also rewrite the code for that part)

I will try to clean up my code and make a PR asap. However, I'm not sure how to proceed with the interface. I know floats are frowned upon in embedded, so I don't think I should make the driver return a float. Any suggestions? Maybe a struct that contains both the integral and decimal part?

closed 2021-11-20 by maribu

hugueslarrive, 2023-05-27:
I have opened PR #19674 since I had already fixed the code...

PR #16934 (2021-11-20): drivers/dht: correct interpreting raw values
### Contribution description

The existing driver only had an accuracy of 1°C instead of 0.1°C. This was due to an incorrect interpretation of the datasheet. The lower bits were discarded. This patch corrects the use of how the raw data should be interpreted and converted. 

The proof of the pudding is in the eating:
```
2021-10-02 00:56:24,832 # RAW values: temp: 23.6 hum: 52.0
2021-10-02 00:56:24,837 # DHT values - temp: 23.6°C - relative humidity: 52.0%
2021-10-02 00:56:26,857 # RAW values: temp: 22.9 hum: 54.0
2021-10-02 00:56:26,861 # DHT value [... cut]