RIOT-OS/RIOT #16908: tests/thread_float: crashes on

benpicco, 2021-09-28:
#### Description

The `thread_float` test crashes on `avr-rss2` (ATnega256rfr2).

#### Steps to reproduce the issue

    make -C tests/thread_float BOARD=avr-rss2 flash term

#### Expected results

```
main(): This is RIOT! (Version: 2021.10-devel-801-g6d9db)
THREADS CREATED

THREAD 3 start
T(3): 141.466812
T(3): 141.490616
T(3): 141.521576
T(3): 141.559769
T(3): 141.605148
T(3): 141.657516
T(3): 141.717255
T(3): 141.783844
T(3): 141.857513
T(3): 141.938278
T(3): 142.026108
T(3): 142.120789
T(3): 142.222488
T(3): 142.331390
T(3): 142.446899
T(3): 142.569565
T(3): 142.699020
T(3): 142.835480
THREAD 4 start
THREAD 5 start
T(3): 142.978668
T(5): 141.521576
T(3): 143.128845
T(5): 141.559769
T(3): 143.285721
T(5): 141.605148
T(3): 143.449615
T(5): 141.657516
T(3): 143.620071
T(5): 141.717255
T(3): 143.797531
T(5): 141.783844
T(3): 143.981476
T(5): 141.857513
T(3): 144.172333
T(5): 141.938278
```

#### Actual results

```
2021-09-28 15:32:17,608 # main(): This is RIOT! (Version: 2021.10-devel-801-g6d9db)
2021-09-28 15:32:17,611 # THREADS CREATED
2021-09-28 15:32:17,611 # 
2021-09-28 15:32:17,907 # IRQ_STATUS 0x80
2021-09-28 15:32:17,908 # IRQ_STATUS1 00
2021-09-28 15:32:17,910 # SCIRQS 00
2021-09-28 15:32:17,910 # BATMON 0x22
2021-09-28 15:32:17,910 # EIFR 00
2021-09-28 15:32:17,912 # PCIFR 00
2021-09-28 15:32:17,913 # *** RIOT kernel panic:
2021-09-28 15:32:17,916 # i): %f
2021-09-28 15:32:17,916 # 
2021-09-28 15:32:17,916 # 
2021-09-28 15:32:17,916 # *** halted.
2021-09-28 15:32:17,916 # 
```

#### Versions

emmanuelsearch, 2022-01-21:
it is also flaky on native (I'm currently testing on 2022.01-RC1)

```
INFO:native.tests/thread_float:Run test.flash
WARNING:native.tests/thread_float:make RIOT_CI_BUILD=1 CC_NOCOLOR=1 --no-print-directory BUILD_IN_DOCKER=1 DOCKER="sudo docker" -C ./tests/thread_float test
r
/home/emmanuel/RIOT/tests/thread_float/bin/native/tests_thread_float.elf /dev/ttyACM0 
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

Help: Press s to start test, r to print it is ready
READY
s
START
main(): This is RIOT! (Version: buildtest)
THREADS CREATED

THREAD t1 start
t1: 141.443787
t1: 141.443787
t1: 141.443787
t1: 141.443787
t1: 141.443787
t1: 141.443787
THREAD t2 start
THREAD t3 start
t3: 141.466812
t3: 141.466812
t3: 141.466812
t3: 141.466812
t3: 141.466812
t3: 141.466812
t3: 141.466812
t1: -nan
t1: 141.412857
t1: 141.412857
t1: 141.412857
t1: 141.412857
t1: 141.412857
t1: 141.412857
t1: 141.412857
t1: 141.412857
t1: 141.412857
t1: 141.412857
t1: 141.412857
t1: 141.412857
t1: 141.412857
t1: 141.412857
t3: -nan
t3: 141.436065
t3: 141.436065

Traceback (most recent call last):
  File "/home/emmanuel/RIOT/tests/thread_float/tests/01-run.py", line 81, in <module>
    sys.exit(run(testfunc))
  File "/home/emmanuel/RIOT/dist/pythonlibs/testrunner/__init__.py", line 30, in run
    testfunc(child)
  File "/home/emmanuel/RIOT/tests/thread_float/tests/01-run.py", line 67, in testfunc
    assert result == first_result, "same calculation but different result"
AssertionError: same calculation but different result
make: *** [/home/emmanuel/RIOT/makefiles/tests/tests.inc.mk:22: test] Error 1
```

fjmolinas, 2022-01-21:
@emmanuelsearch the failure you mentioned is mentioned in #495  and https://github.com/RIOT-OS/RIOT/issues/17170

maribu, 2022-01-21:
@emmanuelsearch The issue is exactly what https://github.com/RIOT-OS/RIOT/issues/17170 is about. A fix would be in glibc's implementation of ucontext by also backing up and restoring the FPU state on context switches. Likely, we should create a minimal example (that only used ucontext and nothing from RIOT) that breaks and report it upstream to glibc.

maribu, 2022-01-21:
@benpicco: have you tried increasing stack sizes?

benpicco, 2022-01-21:
> @benpicco: have you tried increasing stack sizes?

Even with 

```
CFLAGS += -DTHREAD_STACKSIZE_MAIN=4096
CFLASG += -DTHREAD_STACKSIZE_IDLE=2048
```

it still crashes.

closed 2023-01-04 by bors[bot]