RIOT-OS/RIOT #17111: [TRACKING]: ztimer migration

fjmolinas, 2021-11-02:
This issue is meant to track nontrivial users of `xtimer` or `periph_rt*`, corner cases that could conflict with a migration to `ztimer`, or need to be resolved fur such a migration to eventually happen.

### Modules depending on 64-bit xtimer Implementation #13667 

See #13667 for a tracking issue.

There are two options of addressing this:
- remove the need for 64bit long-running timers (e.g: using `ZTIMER_MSEC` or `ZTIMER_SEC`)
- add `ztimer64` which will provide the required missing API https://github.com/RIOT-OS/RIOT/pull/16928

### Modules depending on `xtimer_ticks_to_usec` API

There are two options to address this:
- migrate use cases to not rely on ticks
- implement `xtimer_ticks` on top of `ZTIMER_MSEC_BASE` add functions for conversion between ticks and usec like those in xtimer base on `ZTIMER_MSEC_BASE_FREQ`

API Users:

- [ ]  [drivers/at30tse75x](https://github.com/RIOT-OS/RIOT/tree/master/drivers/at30tse75x): used to spin can be covnerted to `ztimer_spin`
- [ ]  [drivers/at8520e](https://github.com/RIOT-OS/RIOT/tree/master/drivers/at8520e):  timeout is set in seconds (max 256) -> can be converted to `ZTIMER_SEC` or `ZTIMER_MSEC`  https://github.com/RIOT-OS/RIOT/pull/17105
- [ ]  [drivers/dose](https://github.com/RIOT-OS/RIOT/tree/master/drivers/dose): use to make sure that sleep time is above `XTIMER_BACKOFF`, not needed with ztimer.
- [ ]  [drivers/nrf24l01p](https://github.com/RIOT-OS/RIOT/tree/master/drivers/nrf24l01p): used to spin, can be converted to `ZTIMER_USEC`
- [ ]  [drivers/xbee](https://github.com/RIOT-OS/RIOT/tree/master/drivers/xbee): wait 1 second with us precision -> 32 bit is enough (one function rewrite)
- [ ]  [drivers/nvram_spi](https://github.com/RIOT-OS/RIOT/tree/master/drivers/nvram_spi): used to spin, can be converted to `ztimer_spin`
- [ ]  [drivers/rn2xx3](https://github.com/RIOT-OS/RIOT/tree/master/drivers/rn2xx3]):  timeout is set in seconds (max 256) -> can be converted to `ZTIMER_SEC` or `ZTIMER_MSEC`  https://github.com/RIOT-OS/RIOT/pull/17105
- [ ] [drivers/si11rr](https://github.com/RIOT-OS/RIOT/tree/master/drivers/si1133): a usec timeout that can be updated to not use the ticks api

- [ ] [paho-mqtt](https://github.com/RIOT-OS/RIOT/tree/master/pkg/paho-mqtt): read timeouts in ms -> use ZTIMER_MSEC

- [ ] [sys/net/gnrc/netif](https://github.com/RIOT-OS/RIOT/tree/master/sys/net/gnrc/netif): periodic wake up in us, can use `ZTIMER_USEC`
- [ ] [sys/ps](https://github.com/RIOT-OS/RIOT/tree/master/sys/ps): runtime is measured in ticks
- [ ] [sys/stdio_rtt](https://github.com/RIOT-OS/RIOT/tree/master/sys/stdio_rtt): there is some polling functionality, probably can be adapted to use usec or even msec, since polling time is 50000 ticks
- [ ] [sys/stdio_semihosting](https://github.com/RIOT-OS/RIOT/tree/master/sys/stdio_semihosting): pollrate is in ms, can be switched to `ZTIMER_MSEC`.

### Ztimer conflicts

- [ ] https://github.com/RIOT-OS/RIOT/issues/17060
- [ ] [sys/net/gnrc/link_layer/lwmac](): uses rtt, could use ZTIMER_MSEC
- [ ] [sys/net/gnrc/link_layer/gomach](): uses rtt, could use ZTIMER_MSEC
- [ ] simple deep sleep use cases where `rtc` or `rtt` are used to wake up the CPU
- [ ] active clocks preventing sleep: https://github.com/RIOT-OS/RIOT/issues/16891

### Ztimer other

- [ ] `ztimer->adjust_sleep` and `ztimer->adjust_set` automatic configuration
- [ ] `RTT_MIN_OFFSET` is not 2 for all CPU's, shuuld be addapted for all
- [ ] ztimer multiple backends pottential footgun https://github.com/RIOT-OS/RIOT/issues/16698
- [ ] CPU/BOARDs timer width are hardcoded and do not always represent the actual timer backend size `TIMER_0_MAX_VALUE`

fjmolinas, 2021-11-03:
@kaspar030 @bergzand @aabadie @maribu @jue89 @kfessel Might be interested, feel free to add anything else to the list.

fjmolinas, 2021-12-07:
Based on a discussion spanning from https://github.com/RIOT-OS/RIOT/pull/16928 there is a need to make it explicit when the modules/features that are been being pulled in are eventually blocking `ztimer` clock backend to be stopped/started to save power.

It would be needed as the first step to document which platforms are relying on the usec timer to be always running which is not always explicit when modules are using it for timestamping.

fjmolinas, 2022-05-12:
For reference in release 2024.04, these are the list of applications using `ztimer64`, mostly its `posix`, `cpp11` and some specific packages.

```
examples/ccn-lite-relay
examples/cord_ep
examples/cord_lc
examples/dtls-echo
examples/dtls-sock
examples/dtls-wolfssl
examples/gcoap_dtls
examples/gnrc_networking_mac
examples/lua_basic
examples/lua_REPL
examples/openthread
examples/posix_select
tests/cpp11_condition_variable
tests/cpp11_mutex
tests/cpp11_thread
tests/driver_esp_eth
tests/driver_ltc4150
tests/driver_pir
tests/gnrc_gomach
tests/lua_loader
tests/nimble_autoconn_ccnl
tests/pkg_arduino_sdi_12
tests/pkg_flatbuffers
tests/pkg_tflite-micro
tests/pkg_tinydtls_sock_async
tests/pkg_wolfcrypt-ed25519-verify
tests/pkg_wolfssl
tests/posix_semaphore
tests/pthread
tests/pthread_barrier
tests/pthread_cleanup
tests/pthread_condition_variable
tests/pthread_cooperation
tests/pthread_flood
tests/pthread_rwlock
tests/pthread_tls
tests/sema
tests/sntp
tests/sys_fido2_ctap
tests/unittests
tests/ztimer64_msg
```

emmanuelsearch, 2022-05-12:
(I suppose you meant 2022.04)