RIOT-OS/RIOT #17162: tests/lwip: does not compile for IPv4 on 6LoWPAN-based boards.

miri64, 2021-11-09:
#### Description
The `tests/lwip`app does not compile for 6LoWPAN boards when `LWIP_IPV4=1` is set.

```console
BOARD=iotlab-m3 LWIP_IPV4=1 make -C tests/lwip flash -j
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT/tests/lwip'
Building application "tests_lwip" for "iotlab-m3" with MCU "stm32f1".

[INFO] updating lwip /home/mlenders/Repositories/RIOT-OS/RIOT/tests/lwip/bin/pkg/iotlab-m3/lwip/.pkg-state.git-downloaded
echo 159e31b689577dbf69cf0683bbaffbd71fa5ee10 > /home/mlenders/Repositories/RIOT-OS/RIOT/tests/lwip/bin/pkg/iotlab-m3/lwip/.pkg-state.git-downloaded
[INFO] patch lwip
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/pkg/lwip
"make" -f /home/mlenders/Repositories/RIOT-OS/RIOT/Makefile.base MODULE=lwip_api -C /home/mlenders/Repositories/RIOT-OS/RIOT/tests/lwip/bin/pkg/iotlab-m3/lwip/src/api
"make" -f /home/mlenders/Repositories/RIOT-OS/RIOT/Makefile.base MODULE=lwip_core -C /home/mlenders/Repositories/RIOT-OS/RIOT/tests/lwip/bin/pkg/iotlab-m3/lwip/src/core
"make" -f /home/mlenders/Repositories/RIOT-OS/RIOT/Makefile.base MODULE=lwip_ipv4 -C /home/mlenders/Repositories/RIOT-OS/RIOT/tests/lwip/bin/pkg/iotlab-m3/lwip/src/core/ipv4
"make" -f /home/mlenders/Repositories/RIOT-OS/RIOT/Makefile.base MODULE=lwip_ipv6 -C /home/mlenders/Repositories/RIOT-OS/RIOT/tests/lwip/bin/pkg/iotlab-m3/lwip/src/core/ipv6
"make" -f /home/mlenders/Repositories/RIOT-OS/RIOT/Makefile.base MODULE=lwip_netif -C /home/mlenders/Repositories/RIOT-OS/RIOT/tests/lwip/bin/pkg/iotlab-m3/lwip/src/netif
/home/mlenders/Repositories/RIOT-OS/RIOT/tests/lwip/bin/pkg/iotlab-m3/lwip/src/core/init.c:224:2: error: #error "LWIP_ETHERNET needs to be turned on for LWIP_ARP or PPPOE_SUPPORT"
  224 | #error "LWIP_ETHERNET needs to be turned on for LWIP_ARP or PPPOE_SUPPORT"
      |  ^~~~~
```

Not sure if this was already the case on the introduction of IPv4 to that app in d536235f0560b59716d61f0575d3fdab6a1d9fbe, as the build there aborts with

```console
$ BOARD=iotlab-m3 LWIP_IPV4=1 make -C tests/lwip flash -j
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT/tests/lwip'
Building application "tests_lwip" for "iotlab-m3" with MCU "stm32f1".

/bin/sh: line 1: /home/mlenders/Repositories/RIOT-OS/RIOT/dist/tools/kconfiglib/genconfig.py: No such file or directory
make: *** [/home/mlenders/Repositories/RIOT-OS/RIOT/makefiles/kconfig.mk:104: /home/mlenders/Repositories/RIOT-OS/RIOT/tests/lwip/bin/iotlab-m3/generated/autoconf.h] Error 127
make: Leaving directory '/home/mlenders/Repositories/RIOT-OS/RIOT/tests/lwip'
```

It "builds" "again" for 9c8b7a064b304e7f308964208d4a6db0f56bca9b (with the `bkup` issue on Arch that was fixed in https://github.com/RIOT-OS/RIOT/pull/16753), but that one does not have the changes from d536235f0560b59716d61f0575d3fdab6a1d9fbe and the merge commit after reveals the first error above. Question is: does it make sense to support IPv4 on 6LoWPAN boards, as there is no spec for IPv4 for over IEEE 802.15.4 as there is for IPv6 (aka 6LoWPAN ;-)).

#### Steps to reproduce the issue
Run

```sh
BOARD=iotlab-m3 LWIP_IPV4=1 make -C tests/lwip flash -j
```

or

```sh
BOARD=iotlab-m3 LWIP_IPV4=1 LWIP_IPV6=1 make -C tests/lwip flash -j
```

(alternatively, use any other board that does not have an Ethernet-based interface)

#### Expected results
Application builds.

#### Actual results
Compilation errors with

```
  224 | #error "LWIP_ETHERNET needs to be turned on for LWIP_ARP or PPPOE_SUPPORT"
      |  ^~~~~
```

#### Versions
Might be the case ever since d536235f0560b59716d61f0575d3fdab6a1d9fbe

miri64, 2021-11-10:
While the error described here was fixed with #17174, IMHO this should be kept open, as fixing that error revealed more (linking) errors (see OP and comments of #17174)

emmanuelsearch, 2022-01-11:
@miri64 in your opinion, is this an issue/bug we should track w.r.t. the upcoming feature freeze for 2022.01 ?

miri64, 2022-01-11:
Maybe as a warning it would not be harmful to be included in the changelog. Howevre, I am not sure, if this issue is even still valid. @yarrick did one of your earlier PRs fix this?

yarrick, 2022-01-11:
`BOARD=iotlab-m3 LWIP_IPV4=1 make -C tests/lwip -j` now works (since it does not disable IPv6 anymore)

`BOARD=iotlab-m3 LWIP_IPV4=1 LWIP_IPV6=1 make -C tests/lwip -j` works

The only broken one is `BOARD=iotlab-m3 LWIP_IPV4=1 LWIP_IPV6=0 make -C tests/lwip -j` as the parts enabling sixlowpan does so independent of lwIP modules. This would fail anyway but a nicer Makefile error would be best instead of compiling everything and failing when linking.

yarrick, 2022-01-12:
We could do something like this:

```
diff --git a/pkg/lwip/Makefile.dep b/pkg/lwip/Makefile.dep
index 9903a67418..4e750a3781 100644
--- a/pkg/lwip/Makefile.dep
+++ b/pkg/lwip/Makefile.dep
@@ -34,6 +34,10 @@ endif
 ifneq (,$(filter lwip_ipv6,$(USEMODULE)))
   USEMODULE += ipv6
   USEMODULE += random
+else
+  ifneq (,$(filter lwip_sixlowpan,$(USEMODULE)))
+    $(error "6LoWPAN is enabled, IPv6 support is required")
+  endif
 endif
 
 ifneq (,$(filter lwip_udplite,$(USEMODULE)))
```

miri64, 2022-01-12:
> We could do something like this:
>
> ```
> diff --git a/pkg/lwip/Makefile.dep b/pkg/lwip/Makefile.dep
> index 9903a67418..4e750a3781 100644
> --- a/pkg/lwip/Makefile.dep
> +++ b/pkg/lwip/Makefile.dep
> @@ -34,6 [...]

Would the flip-side maybe possible? Not selecting lwip_sixlowpan, if `lwip_ipv6` is not included?

yarrick, 2022-01-12:
Skipping `lwip_sixlowpan` makes it fail when building autoinit files:
`pkg/lwip/init_devs/auto_init_at86rf2xx.c:37:13: error: implicit declaration of function 'lwip_add_6lowpan' [-Werror=implicit-function-declaration]`

We could add ifdef guards in the 6lowpan files, or supply a no-op version of that function I guess (maybe in combination with a warning). Or fail with an error in the Makefile like above but in inside
```
ifneq (,$(filter ieee802154 nrfmin esp_now cc110x,$(USEMODULE)))
  ifneq (,$(filter lwip%, $(USEMODULE)))
```
in sys/Makefile.dep