RIOT-OS/RIOT #17170: tests/thread_float fails on native

maribu, 2021-11-09:
#### Description

`tests/thread_float` since https://github.com/RIOT-OS/RIOT/pull/16901 checks mainly for correct backup and restore of the FPU state during context switches. The two threads will perform the same calculation over and over again and print the result. On `native`, both threads occasionally print a different result (namely `nan`). But obviously, the same calculation should reliably yield the same result.

#### Steps to reproduce the issue

Run `tests/thread_float` on `native` (after https://github.com/RIOT-OS/RIOT/pull/16901 is merged) over and over again. Eventually, it will fail.

#### Expected results

The test passes 100% of the time.

#### Actual results

The test fails every now and then.

#### Versions

RIOT `master` after https://github.com/RIOT-OS/RIOT/pull/16901 is merged.

kaspar030, 2021-11-09:
I think this is a duplicate: https://github.com/RIOT-OS/RIOT/issues/495

Please re-open if you disagree.

maribu, 2021-11-10:
I tend to disagree: https://github.com/RIOT-OS/RIOT/issues/495 describes an issue with the FPU state being corrupted when using the FPU in IRQs. This requires backing up and restoring the FPU state at the ISR preamble / epilogue and likely is a bug that we could fix (even though it would require assembly in the signal handlers simulating the ISRs). The issue with the FPU state getting corrupted by using makecontext() / setcontext() for context switches is IMO an issue with the standard C lib not backing up the FPU state and restoring it. Here, we can only report upstream.