RIOT-OS/RIOT #17206: boards/saml11-xpro: second UART is broken

dylad, 2021-11-15:
#### Description
The second UART available on EXT1 connector is broken on master.
I didn't bisect it yet but this only affects SAML11 not SAML10.

#### Steps to reproduce the issue
Flash `tests/periph_uart`

Connect PA8&PA9 together for loopback.
run the following commands in RIOT shell:
```
init 1 115200
send 1 "HelloRIOT!"
```

#### Expected results
"HelloRIOT!" is send & received by UART_DEV(1).

#### Actual results
Logic analyzer shows nothing.

#### Versions

dylad, 2021-11-16:
See discussion in #14311

MrKevinWeiss, 2022-07-26:
In section 4.1 of the [SAM-L11-Security-ReferenceGuide](http://ww1.microchip.com/downloads/en/AppNotes/SAM-L11-Security-ReferenceGuide-AN-DS70005365A.pdf) it goes into more detail indicating 
```
The peripherals security attribution cannot be changed during application run-time.
Any changes to the User Row fuses require a reset of the SAM L11 device.
```

Table Table 10-11 of the [SAM-L10L11-Family-DataSheet](https://ww1.microchip.com/downloads/en/DeviceDoc/SAM-L10L11-Family-DataSheet-DS60001513F.pdf) shows the UROW mapping indicating the NONSECC bit offset is 223:192. 
In section 15.6 Register Summary, it shows the SERCOM1 is bit 2 of the NONSECC register. More details are in 15.7.14 Peripheral Non-Secure Status - Bridge C.

With this information I though doing something like **WARNING DO NOT DO THIS**:
```
EDBG_ARGS="-F w,194,1" BOARD=saml11-xpro make flash term -C tests/periph_uart
```
would set that fuse to 1... It somewhat bricked the board, so that is NOT the solution.

If @dylad or @benpicco has any idea what section and offset the fuse to unlock the SERCOM1 via EDBG that would be great.
I will try Atmel Studio next and see if I can get any info from there.

MrKevinWeiss, 2022-07-26:
It should be noted that the second UART does work if you have previously set the fuse.

dylad, 2022-07-26:
> If @dylad or @benpicco has any idea what section and offset the fuse to unlock the SERCOM1 via EDBG that would be great.
I will try Atmel Studio next and see if I can get any info from there.

I vaguely remember having the same issue back then. I locked my board using edbg but Atmel Studio allows me to unlock it and also to set the fuse properly.

dylad, 2022-07-27:
I tried to look at edbg but I didn't find anything suspicious for now.
I've opened an issue upstream.

dylad, 2022-07-28:
@MrKevinWeiss The edbg issue has been fixed upstream by its author.
Would you mind retest on your side with:
`./edbg -t saml11  -F w0,194,1`

If your board is still locked, you should now be able to unlocked it with:
`./edbg -t saml11  -e` or `./edbg -t saml11  -u`

If you're happy, we can then bump edbg version in RIOT and add a notice somewhere regarding this bit to finally close this issue.

MrKevinWeiss, 2022-07-28:
I tried unlocking it before and it seems to not be so happy.  I will continue to investigate with the atmel studio (installing now) to get it working then try the new edbg.

dylad, 2022-07-28:
> I tried unlocking it before and it seems to not be so happy.

Yup the old edbg version cannot recover from this issue. But you should be able to fix it with the lastest edbg master without the need of Atmel/Microchip Studio.

dylad, 2022-07-28:
> the question is to do this by default when flashing or have a flag or just update the docs?

I don't think we shoud doing it by default. This SERCOM isn't used everytime and I don't like the idea to play with the board's fuses without the explicit consent of the end user.

Update the board documentation (for me) would be good enough + maybe add a notice in saml1x periph_conf.h file ?

dylad, 2022-07-28:
I can prepare both PRs tonight if you want.

MrKevinWeiss, 2022-07-28:
> I don't think we shoud doing it by default. This SERCOM isn't used everytime and I don't like the idea to play with the board's fuses without the explicit consent of the end user.

Agreed

> Update the board documentation (for me) would be good enough + maybe add a notice in saml1x periph_conf.h file ?

Sure thing, probably also add in a package bump:

```diff
diff --git a/dist/tools/edbg/Makefile b/dist/tools/edbg/Makefile
index 60eaf51e17..90a6a13437 100644
--- a/dist/tools/edbg/Makefile
+++ b/dist/tools/edbg/Makefile
@@ -1,6 +1,6 @@
 PKG_NAME=edbg
 PKG_URL=https://github.com/ataradov/edbg
-PKG_VERSION=e2cd361327253808f130359e2f23fa50ef44dc63
+PKG_VERSION=91bab9868833e185221592ebba71ad3e40cdc273
 PKG_LICENSE=BSD-3-Clause
 
 # manually set some RIOT env vars, so this Makefile can be called stand-alone
```

MrKevinWeiss, 2022-07-28:
I would be OK all in 1 PR.  The CI times are getting pretty fun.

PR #18384 (2022-07-29): dist/tools/edbg: bump version (and update SAML11 documentation)
### Contribution description

This PR updates EDBG version to its lastest master. It contains a fix needed by SAML11 so we can properly set fuses in User ROW memory.
Also update SAML11 board documentation: In order to use SERCOM1 on EXT1 connector. A special fuse bit must be set to enable non-secure operation on this SERCOM (As RIOT doesn't support the secure world of the SAML11 yet).

### Testing procedure
Enable SERCOM1 non-secure mode using EDBG and try to use SERCOM1 with `tests/periph_uart`
`EDBG_ARGS="-F w0,194,1" BOARD=saml [... cut]
closed 2022-07-29 by MrKevinWeiss