RIOT-OS/RIOT #17207: stm32: cmsis: some issues with files downloaded into source tree.

kaspar030, 2021-11-15:
#### Description

The stm32 cmsis headers are getting downloaded *right into the source tree*, and don't get cleaned up.
While individually they're not that large (3-12mb), they sum up to ~190mb. While probably unnoticable locally, that's causing higher peak load on CI's tmpfs. 
`git status` doesn't even show them, and `git clean -xfd` ignores them.

I think we should download (and use them from) `build/...`. That way, they could at least get clean between builds. (checking out from git-cache is almost instant for such small repos).

I've taken a brief look, that shouldn't be too hard. `gen_irqs.py` and `gen_irqs.py` can be changed to allow specifying the input and output files/folders, and after that, it's just a matter changing the makefiles, (I think).

aabadie, 2021-11-16:
This could also be solved by not using tmpfs but a local disk volume. You said that there was no performance difference between the two and using a local disk volume would give persistence of the workers ccache after a restart.

kaspar030, 2021-11-16:
> This could also be solved by not using tmpfs but a local disk volume. You said that there was no performance difference between the two and using a local disk volume would give persistence of the workers ccache after a restart.

I said there was no performance difference hammering ccache with ninja, on tmpfs vs. xfs-noatime/nvme, with completely hot caches and plenty of free RAM. Not that we'll drop tmpfs for the build directories/checkouts on CI any time soon.

aabadie, 2021-11-16:
The problem I see with moving the cmsis headers to build is related to user experience. I see your proposal as "the CI setup has an issue, let's modify RIOT to workaround the that problem", so in other word, this means CI driven development when we should think at the user experience first IMHO.

kaspar030, 2021-11-16:
```
The stm32 cmsis headers are getting downloaded right into the source tree, and don't get cleaned up.
...
git status doesn't even show them, and git clean -xfd ignores them.
```

That's at least not "best practice", no?

bergzand, 2021-11-16:
> The problem I see with moving the cmsis headers to `build/` is related to user experience.

Can you elaborate a bit why this is a problem for the user experience? Users usually don't have to care about the CMSIS headers and where they are on the fs right?

aabadie, 2021-11-16:
> Can you elaborate a bit

These files are related to STM32 so it makes sense to have them located under `cpu/stm32`. I don't see why this would be bad practice. If a developer is implementing a new driver for stm32, all required source code (versioned or generated) is located under cpu/stm32, so this avoids jumping from one directory to another. This improves user (e.g. developer) experience I think.

aabadie, 2021-11-16:
I don't see why the cmsis headers "don't get cleaned up" is an issue. They are supposed to be fetched only once for all.

maribu, 2021-11-16:
> The problem I see with moving the cmsis headers to `build/` is related to user experience.

To be honest, I was totally surprised to see that this package behaves different than other packages. My expectation was that all code outside of `build` is part of the RIOT repository and this expectation went greatly afoul the approach chosen for the STM32 vendor headers.

aabadie, 2021-11-16:
To be honest, initially in #14152 it was done this way: all cmsis were cloned in a directory at the root of the riotbase directory (that was called `cache` at this time) and this was changed after https://github.com/RIOT-OS/RIOT/pull/14152#discussion_r433749507 (sorry @fjmolinas :) ).

I don't really care of moving the headers back there (well in `build/` now) but that needs to be documented somewhere because it's not always obvious what's going on with the build system (expect for the ones involved). I'll draft a PR.

PR #17212 (2021-11-19): cpu/stm32: clone cmsis header in build/stm32

### Contribution description

This should fix the problem reported in #17207 by cloning the STM32 cmsis headers under `build/stm32` instead of inside the `cpu/stm32/include` directory.

### Testing procedure

- Green CI
- No more memory issues on the Murdock workers

### Issues/PRs references

fixes #17207
closed 2021-11-19 by aabadie