RIOT-OS/RIOT #17257: AT86RF215: How to increase increase TX power on OpenMote-b

JohaYuuki, 2021-11-23:
I am working with OpenMote-b modules with the at86rf215 chip, in mr-fsk modulation according to the IEEE 802.15.4 standard. I have been trying to change the Tx power to use it at 14 dBm, in the MakeFile I have used the statement
```
CFLAGS + = -DCONFIG_IEEE802154_DEFAULT_TXPOWER = 14
```
however the TX power does not increase by 3dBm. I attach the Makefile that I use
[Makefile.txt](https://github.com/RIOT-OS/RIOT/files/7590036/Makefile.txt)

benpicco, 2021-11-23:
Yea the dBm value displayed by RIOT is bogus. Since the TX power depends on the chosen modulation and frequency (see [AT86RF215 data sheet](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-42415-WIRELESS-AT86RF215_Datasheet.pdf), chapter 11.1 "Output Power at Several Modulations") we would need to maintain several tables in ROM just to display the right value.

Instead, dBm displayed is just [`TXPWR` - 31 dBm](https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf215/at86rf215_getset.c#L164). (It would of course be possible to implement it so that the proper calculation could be enabled with a compile-time option if desired - patches welcome!)

The added 3 dBm are then achieved by the 3 possible states of the `PACUR` value, which sets the operating current of the power amplifier. (Each step adds ~1dBm so this is not that far off).

That means that +3 dBm in RIOT are equal to PACUR = 3 and `TXPWR` = 31 which is the highest possible output power (which should yield +14 dBm with (MR)-O-QPSK)

maribu, 2023-05-17:
I think the question was answered, so that the issue can be closed. Please reopen if there are further questions.

closed 2023-05-17 by maribu