RIOT-OS/RIOT #17260: Add ADC to board p-nucleo-wb55

donsez, 2021-11-24:
#### Description

ADC are defined as A0 ... A5 arduino pins on the board `p-nucleo-wb55` 

```
#define ARDUINO_PIN_A0          GPIO_PIN(PORT_C, 0)
#define ARDUINO_PIN_A1          GPIO_PIN(PORT_C, 1)
#define ARDUINO_PIN_A2          GPIO_PIN(PORT_A, 1)
#define ARDUINO_PIN_A3          GPIO_PIN(PORT_A, 0)
#define ARDUINO_PIN_A4          GPIO_PIN(PORT_C, 3)
#define ARDUINO_PIN_A5          GPIO_PIN(PORT_C, 2)
```

```
> cd tests/periph_adc
> make BOARD=p-nucleo-wb55 flash
There are unsatisfied feature requirements: periph_adc
github/RIOT-OS/RIOT/tests/periph_adc/../../Makefile.include:947: *** You can let the build continue on expected errors by setting CONTINUE_ON_EXPECTED_ERRORS=1 to the command line.  Stop.
```

But the `FEATURES_PROVIDED += periph_adc` is not provided in `Makefile.features`.
It requires the implementation of `RIOT/cpu/stm32/periph/adc_wb.c` and to add `adc_config` into `periph_conf.h`

```c
// from Nucleo F446RE
static const adc_conf_t adc_config[] = {
    {GPIO_PIN(PORT_C, 0), 0, 0},
    {GPIO_PIN(PORT_C, 1), 0, 1},
    {GPIO_PIN(PORT_A, 1), 0, 4},
    {GPIO_PIN(PORT_A, 0), 0, 8},
    {GPIO_PIN(PORT_C, 3), 0, 11},
    {GPIO_PIN(PORT_C, 2), 0, 10},
};
```

### Useful links

donsez, 2023-06-04:
Can someone add the `periph_adc` feature (`cpu/stm32/periph/adc_wb.c`) to the `p-nucleo-wb55`board ? It exists for l*, f¨ and wl cpus.

crasbe, 2024-07-03:
I think the ADC from the WB55 is the same as in the L4 series, so the adc_l4.c can be shared (and renamed to adc_l4_wb3_wb5.c?).
This shouldn't be toooo difficult to implement, maybe I can look into it, I have the NUCLEO-WB55 at home.

From a very quick glance at the reference manuals, it looks like the L4 ADC is present in the WB55, WB35, WB50 and WB30. The WB15 and WB10 have a different type and the (very newly released) WB09 etc. have a different type as well.

WB50, WB30: https://www.st.com/resource/en/reference_manual/rm0471-multiprotocol-wireless-32bit-mcu-armbased-cortexm4-with-fpu-bluetooth-low-energy-or-802154-radio-solution-stmicroelectronics.pdf p.354ff
WB55, WB35: https://www.st.com/resource/en/reference_manual/rm0434-multiprotocol-wireless-32bit-mcu-armbased-cortexm4-with-fpu-bluetooth-lowenergy-and-802154-radio-solution-stmicroelectronics.pdf p.419ff
WB10: https://www.st.com/resource/en/reference_manual/rm0478-multiprotocol-wireless-32bit-mcu-armbased-cortexm4-with-fpu-bluetooth-low-energy-54-radio-solution-stmicroelectronics.pdf p.346ff
WB15: https://www.st.com/resource/en/reference_manual/rm0473-multiprotocol-wireless-32bit-mcu-armbased-cortexm4-with-fpu-bluetooth-low-energy-radio-solution-stmicroelectronics.pdf p.355ff
WB09: https://www.st.com/resource/en/reference_manual/rm0505-stm32wb09xe-ultralow-power-wireless-32bit-mcu-armbased-cortexm0-with-bluetooth-low-energy-and-24-ghz-radio-solution-stmicroelectronics.pdf p.219ff

crasbe, 2024-07-03:
This addition depends on #20756 to be merged, otherwise there will be a merge conflict (and the ADC wouldn't work anyway).

closed 2024-07-08 by dylad