RIOT-OS/RIOT #17298: ztimer_now() documentation is misleading

benpicco, 2021-11-30:
#### Description

The [`ztimer_now()` documentation](https://doc.riot-os.org/group__sys__ztimer.html#gabcd8174ea28626e102b40143075ed9c8) states

> Get the current time from a clock.

which leads a user to believe it functions the same as `xtimer_now()`.

However I learned that ztimer may just stop a timer if no alarm is scheduled, so the delta between two calls to `ztimer_now()` may well be zero if there was no alarm set on the timer in question.

There should also be a warning that now values are not comparable across timers, but this is a comparably minor issue.

#### Versions

kfessel, 2021-12-08:
#16698 is connected to this
#16969 would amplify the issue to do more powersaving

chrysn, 2021-12-14:
In particular, in https://github.com/RIOT-OS/RIOT/issues/16698#issuecomment-931483923 I have a concrete proposal to reduce the chances of abuse.

Text with ztimer_now could be:

> ztimer_now can only be used in very limited situations:
> * During a ztimer interrupt (although you may prefer to use the ztimer value at which the event triggered, but we currently have nothing to access that).
> * While a timer is running (which you can never be sure of -- but you may call the function, and if you later ascertain that there is still a timer you set on the queue, then the value can be used).
>
> If you want to use ztimer as a stopwatch (ie. subtract two times from each other to determine how long something took), set a timer for the maximum time you expect the task to take (which you'd also need when done with ztimer_now to ensure that it keeps running), and then use ztimer_until (https://github.com/RIOT-OS/RIOT/pull/17229) to obtain the comparable time stamps.

fjmolinas, 2021-12-14:
This IMO also needs addressing before the `ztimer` migration can be completed. `pm_layered` is a default module for most `cpu`s, `rtt` is not the most common backend and by default, all `CONFIG_ZTIMER_RTT_BLOCK_PM_MODE` is set to  `ZTIMER_CLOCK_NO_REQUIRED_PM_MODE`.

fjmolinas, 2021-12-15:
> This IMO also needs addressing before the `ztimer` migration can be completed. `pm_layered` is a default module for most `cpu`s, `rtt` is not the most common backend and by default, all `CONFIG_ZTIMER_RTT_BLOCK_PM_MODE` is set to `ZTIMER_CLOCK_NO_REQUIRED_PM_MODE`.

Discussed this offline again and although the documentation could indeed be improved I do not think this changes the default situation with regard to `xtimer`. AFAIK `ztimer` will not stop a timer just block/unblock power modes. It will block on the first `set` and `unblock` when there are no more blocked power modes. This would allow the a timer be stopped py `pm_layered` if a power mode is switched. But this will only happen if by default the power mode was unblocked, and currently **all power modes are blocked** on init, so unless the custom configuration is changing the default the timer won't be stopped.

So event if `ztimer` was to unblock or block a power mode (which is not the case), this does not behave any differently than `xtimer` in the default configuration. Even worse if a custom configuration was to unblock a power mode stopping the timers then `xtimer` would stop as well while with `ztimer` this could actually be prevented by configuring `CONFIG_ZTIMER_TIMER_BLOCK_PM_MODE`.

There is a lot of work to be done down this path for sure, but I want to go back on my statemen that this prevents migrating, since this does not impact the default behaviour with `ztimer` or `xtimer`.

kaspar030, 2021-12-15:
What @fjmolinas says.

> The [`ztimer_now()` documentation](https://doc.riot-os.org/group__sys__ztimer.html#gabcd8174ea28626e102b40143075ed9c8) states
>
>
> which leads a user to believe it functions the same as [...]

ztimer does not stop a timer. it actually prevents timers from stopping by blocking a sleep mode (e.g., if a timer is set).
xtimer does not do that, so implicitly, xtimers stop when the cpu stops.

> There should also be a warning that now values are not comparable across timers, but this is a comparably minor issue.

There [is](https://github.com/RIOT-OS/RIOT/blob/e2608e178ff3cbe9f74c0b2e079dbd7b5b9a2299/sys/include/ztimer.h#L84) now. Maybe that should be moved / copied to the `ztimer_now()` documentation.

kaspar030, 2021-12-15:
> There should also be a warning that now values are not comparable across timers, but this is a comparably minor issue.

I've added [another](https://github.com/RIOT-OS/RIOT/pull/17404) to `ztimer_now()`.

chrysn, 2021-12-15:
> ztimer does not stop a timer. it actually prevents timers from stopping by blocking a sleep mode (e.g., if a timer is set).

Isn't that an implementation detail best left unexposed? (Clocks vary by device, and once a device comes around whose clock is expensive, we should be able to change the implementation and not wait for people to change their use of ztimer again).

chrysn, 2022-02-03:
Immaterial of whether the timers are stopped implicitly (by the PM lockout being removed) or explicitly (in #17607), do we agree that comparing `ztimer_now` values across times when the underlying clock is not in use by an active ztimer_t can not be expected to yield meaningful differences?

jue89, 2022-02-03:
> do we agree that comparing `ztimer_now` values across times when the underlying clock is not in use by an active ztimer_t can not be expected to yield meaningful differences?

Yes. There is a certain chance that the MCU goes to sleep and stalls the underlying peripheral. Measures have to be taken to ensure the peripheral is clocked during time measurement.

closed 2022-02-15 by chrysn