RIOT-OS/RIOT #17305: cpu/esp32: esp_wifi_enterprise compilation error

gschorcht, 2021-12-01:
#### Description

Compiling `esp_wifi` in WPA2 Enterprise mode no longer works with the merge of PR #14754.

This problem wasn't recognized for more than a year. It shows once again that optional ESP modules that are not pulled in by default in examples and test applications must also be compiled in Murdock. For this purpose, PR #12971 tried to implement special test applications that pull in these optional ESP modules. Since these special test applications are just copies of existing applications, a more elegant way to compile optional ESP modules is to create a virtual board that uses these special ESP modules .

I am already working on such virtual board definitions and will provide a PR. Tracking Issue #12960 lists all optional ESP modules and how they are used in compilation tests.

#### Steps to reproduce the issue

Compile
```
CFLAGS='-DESP_WIFI_EAP_USER=\"riot\" -DESP_WIFI_EAP_PASS=\"riot\"' USEMODULE=esp_wifi_enterprise make BOARD=esp32-wroom-32 -C examples/gnrc_networking
```
The compilation works with commit be3667bfddc16151d91f2fa3c3b4768ceb4d953e, but doesn't work with commit 0d0bc91553830d14217bf0fe5f62dad32c7f2745.

#### Expected results

Compilation should work.

#### Actual results

Compilation doesn't work with current master.

aabadie, 2021-12-01:
> a more elegant way to compile optional ESP modules is to create a virtual board that uses these special ESP modules .

Note that introducing this virtual board might increase once more the number of jobs performed by the CI since all supported applications will be built for this new board. But maybe your proposal will deal with that ?

gschorcht, 2021-12-01:
I know. The idea came from comment https://github.com/RIOT-OS/RIOT/pull/12971#pullrequestreview-333641888. Maybe it would be better after all to have the copies of the test applications that are then only copmpiled for ESPs to reduce the compilation load in Murdock.

But, maybe it is possible to reduce the feature set so that only some applications have to be compiled for these virtual boards. Unfortunatly, a lot of features are defined by the MCU.

gschorcht, 2021-12-01:
@benpicco Could we resume the discussion about the best way to cover optional ESP modules in compilation tests? Based on our discussion in #12971, I have created a virtual ESP32 board that simply references the existing `esp32-wroom-32` board and just enables the optional ESP modules to compile them with the existing `examples` and `tests`. As @aabadie correctly pointed out, this will further increase the number of compilation jobs in Murdock. As we have seen in the last few days, each compilation in Murdock already takes 1 hour or sometimes more.

The problem described in this issue showed again that we really need such compilation tests for optional ESP modules.

gschorcht, 2021-12-01:
> But maybe your proposal will deal with that ?

One possibility to reduce the number of compilations could be to add the virtual board in `Makefile.ci` to `BOARD_INSUFFICIENT_MEMORY` for almost all application.

leandrolanzieri, 2021-12-01:
Going on the direction of #14669, I started a (very outdated by now) [branch](https://github.com/leandrolanzieri/RIOT/commits/dev/kconfig/parameterized_tests), to allow placing different configuration files on the same application for the CI to test. Configuration files are incrementally applied between each test. As most of the objects are already built only the introduced modules need to be recompiled. Together with `fixdep` providing incremental builds that could reduce the number of actual jobs.

closed 2021-12-01 by aabadie