RIOT-OS/RIOT #17311: drivers/aip31068: initialization fails in most cases when using esp_i2c_sw

HendrikVE, 2021-12-01:
#### Description
There is a 80% chance that the initialization for the driver `aip31068` fails when using `esp_i2c_sw` instead of `esp_i2c_hw`.

#### Steps to reproduce the issue
Remove the line `USEMODULE += esp_i2c_hw` from`aip31068/Makefile.dep` and test with real hardware.

#### Attachment
I used the logic analyzer software from saleae which is available [here](https://ideas.saleae.com/f/changelog/).
The attached zip-archive contains two files that can be imported to show all I2C operations for the `esp_i2c_sw`- and `esp_i2c_hw`-version of the driver. As you can see I had to reset my esp32 4 times before initialization was successful.
- [esp32_aip31068.zip](https://github.com/RIOT-OS/RIOT/files/7635000/esp32_aip31068.zip)

HendrikVE, 2021-12-01:
@gschorcht I thought a new issue is a better place to discuess this problem ;)

gschorcht, 2021-12-01:
Where can I buy this hardware?

HendrikVE, 2021-12-01:
I've got mine from [dfrobot](https://www.dfrobot.com/product-1737.html). This display combines the `aip31068` with `pca9633` in one component.

gschorcht, 2021-12-01:
Ordered, it takes 1 to 3 days. Then I will have a look.

HendrikVE, 2021-12-09:
Fixed with #17346

closed 2021-12-09 by HendrikVE