RIOT-OS/RIOT #17332: Bug: gnrc_rpl: some nodes are not reachable in the network

fastriker, 2021-12-03:
#### Description
I build up a RPL network with 8 nodes. Some nodes are not reachable via ping. The unreachable nodes also do not reach any other nodes. The reachable part of the network works as expected. With different root nodes or change between 2.4 GHz and SubGHz the reachable or unreachable nodes change. The unreachable nodes show that they are part of the RPL network.

I try to figure out the problem already for quite some time. I also wonder, if I made any configuration mistakes. I am happy to provide further information if needed or try out specific approach that can solve my issue or lead to its solution.

#### Steps to reproduce the issue

Nodes are normally reachable within one, sometimes two hops. The unmodified gnrc_networking example was used for the test.

The hardware:
- Board: Nucleo F767ZI
- Transceiver: AT86RF215 
- Custom board files from @maribu .

The following steps shows the issue:
- setup 8 nodes
- give an IP on all nodes - for example on interface 7 with  `ifconfig 7 add 2001:db7::x` where x is the node number
- initialize RPL on all nodes - for example with `rpl init 7` for interface 7
- make one node the root node - for example node 7 with `rpl root 1 2001:db7::7`
- ping each other on all nodes - for example node 1 with `ping -c 30 -i 1000 2001:db7::1 ?s40` 

I attached some log files for every node showing the problem for one specific try:
- [node 1](https://github.com/RIOT-OS/RIOT/files/7649813/ssh1.log)
- [node 2](https://github.com/RIOT-OS/RIOT/files/7649814/ssh2.log)
- [node 3](https://github.com/RIOT-OS/RIOT/files/7649815/ssh3.log)
- [node 4](https://github.com/RIOT-OS/RIOT/files/7649816/ssh4.log)
- [node 5](https://github.com/RIOT-OS/RIOT/files/7649817/ssh5.log)
- [node 6](https://github.com/RIOT-OS/RIOT/files/7649818/ssh6.log)
- [node 7](https://github.com/RIOT-OS/RIOT/files/7649819/ssh7.log)
- [node 8](https://github.com/RIOT-OS/RIOT/files/7649820/ssh8.log)

#### Expected results
Every node respond to every other nodes ping.

#### Actual results
Some nodes do not respond any ping except if send pings to themself.

#### Versions

RIOT: 2021.10 (also some random previous versions)

benpicco, 2021-12-03:
This is likely related to #17327.
`gnrc_rpl` can create some… interesting routing table entries:

```
ifconfig
Iface  7  HWaddr: 1E:36  Channel: 26  NID: 0x23 
          Long HWaddr: 5E:41:E9:71:2F:8C:9E:36 
          L2-PDU:102  MTU:1280  HL:64  RTR  
          RTR_ADV  6LO  IPHC  
          Source address length: 8
          Link type: wireless
          inet6 addr: fe80::5c41:e971:2f8c:9e36  scope: link  VAL
   -->    inet6 addr: 2001:db8::5c41:e971:2f8c:9e36  scope: global  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ff8c:9e36
          inet6 group: ff02::1a
          
          Statistics for Layer 2
            RX packets 587  bytes 41706
            TX packets 622 (Multicast: 436)  bytes 47016
            TX succeeded 1246 errors 0
          Statistics for IPv6
            RX packets 585  bytes 52908
            TX packets 579 (Multicast: 436)  bytes 56900
            TX succeeded 579 errors 0

> nib route
nib route
2001:db8::/64 dev #7
2001:db8::a425:71ad:dfbc:8e26/128 via fe80::a425:71ad:dfbc:8e26 dev #7
2001:db8::5c41:e971:2f8c:9e36/128 via fe80::a425:71ad:dfbc:8e26 dev #7 <--
2001:db8::7cb7:4dfb:f754:56fe/128 via fe80::a425:71ad:dfbc:8e26 dev #7
```

fastriker, 2021-12-06:
Thank you for your fast reply and the link to the probably related issue!

I run my setup again but this time I also print out the routing table:
- [node 1](https://github.com/RIOT-OS/RIOT/files/7660552/ssh1.log)
- [node 2](https://github.com/RIOT-OS/RIOT/files/7660553/ssh2.log)
- [node 3](https://github.com/RIOT-OS/RIOT/files/7660554/ssh3.log)
- [node 4](https://github.com/RIOT-OS/RIOT/files/7660555/ssh4.log)
- [node 5](https://github.com/RIOT-OS/RIOT/files/7660556/ssh5.log)
- [node 6](https://github.com/RIOT-OS/RIOT/files/7660557/ssh6.log)
- [node 7](https://github.com/RIOT-OS/RIOT/files/7660558/ssh7.log)
- [node 8](https://github.com/RIOT-OS/RIOT/files/7660559/ssh8.log)

For me the routing tables are looking fine, except the missing entries at node 7 for the unreachable nodes. However, I am not sure if I understood the problem completely.

cgundogan, 2021-12-06:
Just a quick comment without looking so much into the details for now:

> give an IP on all nodes - for example on interface 7 with ifconfig 7 add 2001:db7::x where x is the node number

You should set a global IPv6 address only for the root node (in your case node 7).  All other nodes will generate an IPv6 address based on their link-layer address as soon as they join the DODAG.  I could imagine that you already maxed out the IPv6 address memory space (the NIB array, what is used to printf if you type `ifconfig`).  Another issue with *large* deployments can be that the routing table has not enough memory.  Look [here](https://github.com/RIOT-OS/RIOT/blob/master/examples/gnrc_minimal/Makefile#L30) for a couple of configrations to tweak the array sizes (in case you have a large deployment).

fastriker, 2021-12-07:
Thank you very much, increasing the array size solved my issue! I am using 8 nodes currently, but they are nearly forming a complete graph.
I set IPv6 addresses for all nodes, because it was easier for me to address them. What are the reasons, that only the root node should get an address?

closed 2021-12-07 by fastriker

miri64, 2021-12-07:
> What are the reasons, that only the root node should get an address?

RPL has native support for auto-address configuration via its prefix information option, so you already get a global address. Since space is limited, the default configuration only allows for a limited number of IPv6 addresses, so if you already add a global address, you might run into a similar problem as previously mentioned, since the address array is too small. See also https://doc.riot-os.org/group__net__gnrc__netif__conf.html#ga366596e311b4a450887a1a180ae66fb3.

fastriker, 2021-12-07:
Ok, that's clear - thank you for the explanation!