RIOT-OS/RIOT #17451: sx127x: wrong RSSI values

a-podshivalov, 2021-12-27:
#### Description
Seems that the RSSI values reported in netdev_lora_rx_info_t structure are incorrect. The sx127x driver is definitely affected, and other LoRa modem drivers (sx126x and STM32WL55) could be broken too (though I can't test them now).

#### Steps to reproduce the issue
Receive a packet on the border of modem sensitivity and read the value in rssi field of packet info.

#### Expected results
The actual RSSI should be something like -120 to -130 dBm, similar to receiver sensitivity in the datasheet.

#### Actual results
The value in the rssi field of netdev_lora_rx_info_t will be a large positive (120-130).

#### Versions
All RIOT versions and all sane build environments.

The reason is that rssi field of netdev_lora_rx_info_t (as defined in drivers/include/net/netdev/lora.h) is a uint8_t, but it must contain the values in the range from, possibly, -140 to something like -10 to 10, maybe (a very strong signal, possibly the receiver and transmitter are in close proximity). In the actual device, the RSSI is indicated by a single-byte value (unsigned), and the offset is specified in the documentation. The calculations in the driver (lines 150-177) of sx127x_netdev.c use int16_t for RSSI value, and probably the simplest solution could be changing the type of rssi member in netdev_lora_rx_info_t to int16_t (probably int8_t would be unsuitable).

aabadie, 2022-01-07:
I agree that there's something wrong here. `int8_t` would probably be a good tradeoff and is what is used by the sx126x driver:

https://github.com/Lora-net/sx126x_driver/blob/ba61312213450ae94a4293d75285c1d8f30c04b3/src/sx126x.h#L551

aabadie, 2022-01-07:
I also noticed that the netdev RSSI netopt is an `int8_t`:

https://github.com/RIOT-OS/RIOT/blob/3676b635831d3f4f65efe3d8e5f78fd777b4e79f/drivers/sx127x/sx127x_netdev.c#L347-L350

@jia200x what's your opinion on this ?

jia200x, 2022-01-07:
I will be back on track on Monday. Please ping me if I don't reply by then

jia200x, 2022-01-10:
Yes, there's definitely something wrong. I also agree `int8_t` would be unsuitable, since the radio can report RSSI below the noise level.
Therefore, we should change it to `int16_t`.

PR #17497 (2022-01-17): [treewide] lora: use int16_t for RSSI value

### Contribution description
The RSSI values reported by LoRa transceiver can be less than -127.
Therefore, `int8_t` is not enough. This commit defines the RSSI of
`netdev_lora_rx_info` as `int16_t` and adapt the drivers accordingly
(sx126x, sx127x).

BTW, the `sx126x` was writing an `int16_t` into an `int8_t`. :man_shrugging:. It's indirectly fixed by this commit.

### Testing procedure
- Check that the RSSI values reported by the LoRa radios (sx126x, sx127x) are right (negative values, starting from around -140). Use `tests/driver_sx127x` and `tes [... cut]
closed 2022-01-17 by aabadie