RIOT-OS/RIOT #17492: Using Rust doc toolchain info is obsolete

fvacek, 2022-01-08:
#### Description
https://doc.riot-os.org/using-rust.html#toolchain contains steps to build `c2rust` which didn't work for me:

#### Steps to reproduce the issue
```
$ rustup install nightly-2019-12-05
$ rustup component add --toolchain nightly-2019-12-05 rustfmt rustc-dev
$ cargo +nightly-2019-12-05 install c2rust
$ git clone https://github.com/chrysn-pull-requests/c2rust -b for-riot
$ cd c2rust
$ cargo +nightly-2019-12-05 install --path c2rust
```
`cargo +nightly-2019-12-05 install c2rust` failed with error:
```
   Compiling rustc-demangle v0.1.21
error[E0658]: `match` is not allowed in a `const fn`
   --> /home/fanda/.cargo/registry/src/github.com-1ecc6299db9ec823/half-1.8.2/src/bfloat.rs:304:9
    |
304 | /         match (exp, man) {
305 | |             (0, 0) => FpCategory::Zero,
306 | |             (0, _) => FpCategory::Subnormal,
307 | |             (0x7F80u16, 0) => FpCategory::Infinite,
308 | |             (0x7F80u16, _) => FpCategory::Nan,
309 | |             _ => FpCategory::Normal,
310 | |         }
    | |_________^
    |
    = note: for more information, see https://github.com/rust-lang/rust/issues/49146
    = help: add `#![feature(const_if_match)]` to the crate attributes to enable
```
When I just cloned https://github.com/immunant/c2rust and run `cargo build` in `c2rust` directory, then I got `c2rust` binary without any error. 

#### Expected results

So I thing that sticking on `nightly-2019-12-05` is obsolete information and [doc/doxygen/src/using-rust.md](https://github.com/RIOT-OS/RIOT/blob/master/doc/doxygen/src/using-rust.md) should be updated to current state of `c2rust` development.

chrysn, 2022-01-12:
Thanks, I missed the `--locked` that's required these days (but not in the upstream C2Rust docs).

Patch incoming; for a quick fix just add `--locked` to the `cargo install --path ...` line.

closed 2022-01-17 by maribu

JKRhb, 2022-02-23:
Trying to use the Rust examples, I actually encountered the same problem with the initial c2rust installation using 

```bash
cargo +nightly-2019-12-05 install c2rust
```

Setting the `--locked` here as well solved the issue, this is not yet included the documentation, though :/

chrysn, 2022-02-24:
There shouldn't be an "initial" c2rust installation, just the later one; fix coming up.