RIOT-OS/RIOT #17555: EMULATE=1 not concurrency safe

kaspar030, 2022-01-24:
#### Description

The emulation machinery is using fixed port numbers / socket files for serial and monitor sockets, which makes them unusable in the parallel case (e.g., CI).
Could it use per-application socket files?

#17434 is blocked on this.

#### Steps to reproduce the issue

Run two instances of `EMULATE=1 make test` for e.g., microbit.

#### Expected results

Test passes.

#### Actual results

```
/tmp/dwq.0.5048392832189235/a33c1c69a48300216f746cb52394974e/dist/tools/emulator/term.sh qemu microbit /tmp/dwq.0.5048392832189235/a33c1c69a48300216f746cb52394974e/tests/xtimer_usleep /tmp/dwq.0.5048392832189235/a33c1c69a48300216f746cb52394974e/dist/tools/pyterm/pyterm '-p "/tmp/riot_tests_xtimer_usleep_microbit_uart" -b "115200" ' /tmp/riot_tests_xtimer_usleep_microbit_uart 
qemu-system-arm -machine microbit -device loader,file=/tmp/dwq.0.5048392832189235/a33c1c69a48300216f746cb52394974e/build/tests_xtimer_usleep.elf -serial telnet::5555,server,nowait,nodelay -monitor telnet::45454,server,nowait -nographic
qemu-system-arm: -monitor telnet::45454,server,nowait: Failed to find an available port: Address already in use
make[2]: *** [/tmp/dwq.0.5048392832189235/a33c1c69a48300216f746cb52394974e/Makefile.include:853: emulate] Error 1
```

#### Versions

kfessel, 2022-01-27:
Instead of socat tcp <-> pty
we could socat pty <-> pty
```
socat pty,link=/tmp/soin,raw,echo=0 pty,link=/tmp/soout,raw,echo=0
```
and the connect qemu to it
```
qemu -serial /tmp/soin
```
e.g. in term.sh soin pty with the same name as PORT ($6) +_in

-- qemu can also generate pty but it auto allocates the pty name (pyt has no parameters)

kaspar030, 2022-02-15:
> and the connect qemu to it

well, that would work, *but*, the current machinery is just setting socat as terminal, and starts qemu first. Not sure if that order is (easily) fixable.

kaspar030, 2022-02-15:
apparrently qemu's `pty` takes an `id=<n>` parameter, allowing to control the name of the pty somewhat. maybe that can be used.

kfessel, 2022-02-16:
sadly pty provides not control which path it goe to
id is the backend id to identify which device goes where in qemu it is not the path 
```
-chardev pty,id=id
    Create a new pseudo-terminal on the host and connect to it. pty does not take any options.
    pty is not available on Windows hosts.
```
```
pty
    [Linux only] Pseudo TTY (a new PTY is automatically allocated)
```

https://www.qemu.org/docs/master/system/invocation.html#hxtool-6

kfessel, 2022-03-04:
may be Closed

please check if you can employ the changes by #17679

kaspar030, 2022-03-04:
Yeah, https://github.com/RIOT-OS/RIOT/pull/17434 has been waiting in the queue since #17679 got merged... let's see

kaspar030, 2022-06-05:
fixed by https://github.com/RIOT-OS/RIOT/pull/17679

closed 2022-06-05 by kaspar030