RIOT-OS/RIOT #17573: boards/6lowpan-clicker: Kconfig build different to Makefile build

jenswet, 2022-01-26:
#### Description
The "normal" Makefile build selects/builds the module `newlib_syscalls_mips_uhi`.

The Kconfig build doesn't and therefore selects the default `newlib_syscalls_default`.

#### Steps to reproduce the issue
Building the test from #17477 brought up this bug.
Just remove the board from the blacklist and try to build the test once with and once without Kconfig.

#### Expected results
```
make clean all -j8 BOARD=6lowpan-clicker BUILD_IN_DOCKER=1       
Deprecated board:  6lowpan-clicker
Deprecated cpu:  mips_pic32mx
rm -rf /Users/jens/Uni/MA/RIOT-fork/tests/pkg_etl/bin/6lowpan-clicker/pkg-build/etl
Launching build container using image "riot/riotbuild:latest".
docker run --rm --tty --user $(id -u) -v '/Users/jens/Uni/MA/RIOT-fork:/data/riotbuild/riotbase:delegated' -e 'RIOTBASE=/data/riotbuild/riotbase' -e 'CCACHE_BASEDIR=/data/riotbuild/riotbase' -e 'BUILD_DIR=/data/riotbuild/riotbase/build' -e 'RIOTPROJECT=/data/riotbuild/riotbase' -e 'RIOTCPU=/data/riotbuild/riotbase/cpu' -e 'RIOTBOARD=/data/riotbuild/riotbase/boards' -e 'RIOTMAKE=/data/riotbuild/riotbase/makefiles'      -e 'BOARD=6lowpan-clicker'  -w '/data/riotbuild/riotbase/tests/pkg_etl/' 'riot/riotbuild:latest' make 'BOARD=6lowpan-clicker'   -j8 all
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
Deprecated board:  6lowpan-clicker
Deprecated cpu:  mips_pic32mx
Building application "tests_pkg_etl" for "6lowpan-clicker" with MCU "mips_pic32mx".

"/usr/bin/make" -C /data/riotbuild/riotbase/pkg/etl
"/usr/bin/make" -C /data/riotbuild/riotbase/boards/6lowpan-clicker
"/usr/bin/make" -C /data/riotbuild/riotbase/core
"/usr/bin/make" -C /data/riotbuild/riotbase/cpu/mips_pic32mx
"/usr/bin/make" -C /data/riotbuild/riotbase/cpu/mips_pic32mx/p32mx470f512h
"/usr/bin/make" -C /data/riotbuild/riotbase/drivers
"/usr/bin/make" -C /data/riotbuild/riotbase/sys
"/usr/bin/make" -C /data/riotbuild/riotbase/cpu/mips32r2_common
"/usr/bin/make" -C /data/riotbuild/riotbase/sys/auto_init
"/usr/bin/make" -C /data/riotbuild/riotbase/drivers/periph_common
"/usr/bin/make" -C /data/riotbuild/riotbase/cpu/mips32r2_common/newlib_syscalls_mips_uhi
"/usr/bin/make" -C /data/riotbuild/riotbase/cpu/mips32r2_common/periph
"/usr/bin/make" -C /data/riotbuild/riotbase/cpu/mips_pic32_common
"/usr/bin/make" -C /data/riotbuild/riotbase/cpu/mips_pic32_common/periph
"/usr/bin/make" -C /data/riotbuild/riotbase/sys/isrpipe
"/usr/bin/make" -C /data/riotbuild/riotbase/sys/malloc_thread_safe
"/usr/bin/make" -C /data/riotbuild/riotbase/sys/stdio_uart
"/usr/bin/make" -C /data/riotbuild/riotbase/sys/test_utils/interactive_sync
"/usr/bin/make" -C /data/riotbuild/riotbase/sys/tsrb
   text    data     bss     dec     hex filename
 151140    2728    5564  159432   26ec8 /data/riotbuild/riotbase/tests/pkg_etl/bin/6lowpan-clicker/tests_pkg_etl.elf
```

#### Actual results
```
 make clean all -j8 BOARD=6lowpan-clicker BUILD_IN_DOCKER=1 TEST_KCONFIG=1
Deprecated board:  6lowpan-clicker
Deprecated cpu:  mips_pic32mx
=== [ATTENTION] Testing Kconfig dependency modelling ===
Deprecated board:  6lowpan-clicker
Deprecated cpu:  mips_pic32mx
=== [ATTENTION] Testing Kconfig dependency modelling ===
make: Nothing to be done for 'clean'.
Launching build container using image "riot/riotbuild:latest".
docker run --rm --tty --user $(id -u) -v '/Users/jens/Uni/MA/RIOT-fork:/data/riotbuild/riotbase:delegated' -e 'RIOTBASE=/data/riotbuild/riotbase' -e 'CCACHE_BASEDIR=/data/riotbuild/riotbase' -e 'BUILD_DIR=/data/riotbuild/riotbase/build' -e 'RIOTPROJECT=/data/riotbuild/riotbase' -e 'RIOTCPU=/data/riotbuild/riotbase/cpu' -e 'RIOTBOARD=/data/riotbuild/riotbase/boards' -e 'RIOTMAKE=/data/riotbuild/riotbase/makefiles'      -e 'BOARD=6lowpan-clicker' -e 'TEST_KCONFIG=1'  -w '/data/riotbuild/riotbase/tests/pkg_etl/' 'riot/riotbuild:latest' make 'BOARD=6lowpan-clicker' 'TEST_KCONFIG=1'   -j8 all
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
Deprecated board:  6lowpan-clicker
Deprecated cpu:  mips_pic32mx
=== [ATTENTION] Testing Kconfig dependency modelling ===
"/usr/bin/make" -C /data/riotbuild/riotbase/pkg/etl
Building application "tests_pkg_etl" for "6lowpan-clicker" with MCU "mips_pic32mx".

"/usr/bin/make" -C /data/riotbuild/riotbase/boards/6lowpan-clicker
"/usr/bin/make" -C /data/riotbuild/riotbase/core
"/usr/bin/make" -C /data/riotbuild/riotbase/cpu/mips_pic32mx
"/usr/bin/make" -C /data/riotbuild/riotbase/cpu/mips_pic32mx/p32mx470f512h
"/usr/bin/make" -C /data/riotbuild/riotbase/drivers
"/usr/bin/make" -C /data/riotbuild/riotbase/sys
"/usr/bin/make" -C /data/riotbuild/riotbase/cpu/mips32r2_common
"/usr/bin/make" -C /data/riotbuild/riotbase/cpu/mips_pic32_common
"/usr/bin/make" -C /data/riotbuild/riotbase/sys/auto_init
"/usr/bin/make" -C /data/riotbuild/riotbase/drivers/periph_common
"/usr/bin/make" -C /data/riotbuild/riotbase/cpu/mips32r2_common/periph
"/usr/bin/make" -C /data/riotbuild/riotbase/cpu/mips_pic32_common/periph
"/usr/bin/make" -C /data/riotbuild/riotbase/sys/isrpipe
"/usr/bin/make" -C /data/riotbuild/riotbase/sys/malloc_thread_safe
"/usr/bin/make" -C /data/riotbuild/riotbase/sys/newlib_syscalls_default
"/usr/bin/make" -C /data/riotbuild/riotbase/sys/stdio_uart
/data/riotbuild/riotbase/sys/newlib_syscalls_default/syscalls.c:122:26: error: ‘heaps’ defined but not used [-Werror=unused-const-variable=]
  122 | static const struct heap heaps[NUM_HEAPS] = {
      |                          ^~~~~
/data/riotbuild/riotbase/sys/newlib_syscalls_default/syscalls.c:106:14: error: ‘heap_top’ defined but not used [-Werror=unused-variable]
  106 | static char *heap_top[NUM_HEAPS] = {
      |              ^~~~~~~~
"/usr/bin/make" -C /data/riotbuild/riotbase/sys/test_utils/interactive_sync
cc1: all warnings being treated as errors
make[3]: *** [/data/riotbuild/riotbase/Makefile.base:146: /data/riotbuild/riotbase/tests/pkg_etl/bin/6lowpan-clicker/newlib_syscalls_default/syscalls.o] Error 1
make[2]: *** [/data/riotbuild/riotbase/Makefile.base:31: ALL--/data/riotbuild/riotbase/sys/newlib_syscalls_default] Error 2
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/data/riotbuild/riotbase/Makefile.base:31: ALL--/data/riotbuild/riotbase/sys] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [/data/riotbuild/riotbase/Makefile.include:703: application_tests_pkg_etl.module] Error 2
make: *** [/Users/jens/Uni/MA/RIOT-fork/makefiles/docker.inc.mk:318: ..in-docker-container] Error 2
```

#### Versions
latest riot/riotdocker

MrKevinWeiss, 2022-07-25:
The mips boards are deprecated, and should be removed before the kconfig migration is complete.

maribu, 2022-09-18:
With the PR removing MIPS support and dropping this board on its way in, I think we can categorize this as `wontfix` and close it. That will also get rid of all the special handling and blacklisted the MIPS platform required.

closed 2022-09-18 by maribu