RIOT-OS/RIOT #17580: doc: doc.riot.org and circle-ci  use different doxygen versions doxygen 1.8.17 vs 1.9.1

kfessel, 2022-01-27:
#### Description
circle-ci builds doc using old doxygen 1.8.17
[image]

doc.riot.org is build using a more current 1.9.1
[image]

#### Steps to reproduce the issue
look at:
https://15379-7370241-gh.circle-artifacts.com/0/doc/index.html
and 
https://doc.riot-os.org/index.html
#### Expected results
doc.riot.org and circle-ci  use the same doxygen version 
#### Actual results
doc.riot.org and circle-ci  use different doxygen versions
#### Versions

aabadie, 2022-01-27:
I think the issue comes from version used to generated doc.riot-os.org. The version used by circleci is the same as the one used by the static tests, e.g from Docker image riot/static-test-tools:

```
$ docker run --rm riot/static-test-tools:latest doxygen --version
1.8.17
```

kfessel, 2022-02-11:
debian stable is at version 1.9.1

aabadie, 2022-02-12:
The Docker image is based on Ubuntu Focal, latest LTS (next LTS will 22.04):

https://github.com/RIOT-OS/riotdocker/blob/14fa5cda36f0f09282f3d1efdca79b7c1cb83dca/riotdocker-base/Dockerfile#L1

IMHO, we should use the same version to generate doc.riot-os.org. If we wait for 22.04 and use the same Docker image to generate the doc on doc.riot-os.org, everything will use the same Doxygen version, probably higher 1.9.1.

maribu, 2023-01-05:
Does anyone know what the state of this is?

aabadie, 2023-01-05:
> Does anyone know what the state of this is?

The problem is potentially still there since the published online documentation is generated on a separate server with a different version than the one used for previews (Docker based).
But we can assume that both versions are the same now the riotdocker image is based on 22.04.

closed 2023-05-17 by maribu