RIOT-OS/RIOT #17620: ESP-32 crashing when importing PWM and using it in micropython

herculescoding93, 2022-02-06:
hi so i got an esp 32 today and i installed micropython on it, i ran a couple test with the motor using pwm and it worked fine but after 1 hour i tried again and i get this error which i dont know what it is, could anyone help me pleaseee

ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12344
ho 0 tail 12 room 4
load:0x40080400,len:4124
entry 0x40080680
W (61) boot.esp32: PRO CPU has been reset by WDT.
W (61) boot.esp32: WDT reset info: PRO CPU PC=0x40094585
W (63) boot.esp32: WDT reset info: APP CPU PC=0x40093dea

gschorcht, 2022-09-18:
@herculescoding93 Hm, this problem is not related to RIOT-OS and it is very difficult to guess what could have gone wrong in Micropython. Have you already asked the people from mircopython?

The only thing I can tell from the log is that your ESP32 has been blocked for some reason and therefore reset by the system watchdog. Both cores were running and the PC of the PRO core was at 0x40094585 and the PC of the APP core was at 0x40093dea when they were reset. Both addresses are in the flash. You could use the `nm -s <your_elf_file> | grep 0x40009 | sort` to figure out in which functions these addresses are. If you have an ESP32-WROVER-KIT, it would be quite easy to debug, with other ESP32 it is a bit tricky to use the JTAG interface for debugging.

maribu, 2023-01-05:
Issue reporter unresponsive for almost a year. I'm closing this now.

Please reopen if the issue is still present.

closed 2023-01-05 by maribu