RIOT-OS/RIOT #17717: examples/mallinfo: mallinfo is deprecated in glibc 2.33

kfessel, 2022-02-28:
#### Description

`struct mallinfo` became deprcated in glibc 2.33

https://elixir.bootlin.com/glibc/glibc-2.33/source/malloc/malloc.h#L118
vs
https://elixir.bootlin.com/glibc/glibc-2.32/source/malloc/malloc.h#L101

resulting in warnig, that is upgraded to error in "examples/mallinfo" when compiling for native

```
RIOT/examples/mallinfo/main.c:23:5: error: ‘mallinfo’ is deprecated [-Werror=deprecated-declarations]
   23 |     mi = mallinfo();
      |     ^~
In file included from RIOT/examples/mallinfo/main.c:2:
/usr/include/malloc.h:118:24: note: declared here
  118 | extern struct mallinfo mallinfo (void) __THROW __MALLOC_DEPRECATED;
      |                        ^~~~~~~~
```

#### Steps to reproduce the issue

compile "examples/mallinfo" with glibc-2.33

#### Expected results

I don't know ignore warning or use struct mallinfo2 

#### Actual results

It does not compile 

#### Versions

current master 

#### related PR/ISSUE

#17715 #17714

chrysn, 2022-03-03:
Where is that example in the first place? I found it neither in master nor in a pull request. The users I found in the code base already check for `NO_MALLINFO` (which is set unless newlib or picolibc is used), or are in code so platform specific I suspect glibc isn't an option there anyway.

kfessel, 2022-03-03:
Oops doing "git status" lists it as untracked - seems like it is local to my worktree

closed 2022-03-03 by kfessel