RIOT-OS/RIOT #17722: feather-m0: `make flash` reports "device unsupported"

miri64, 2022-03-01:
#### Description
When trying to flash `hello-world` to `BOARD=feather-m0`, `bossa` (the default programmer for that board) reports `Device unsupported`.

#### Steps to reproduce the issue
With the respective board plugged into USB, run
```
BOARD=feather-m0 make -C examples/hello-world flash
```

#### Expected results
The hello world application should be flashed to the board.

#### Actual results
Flashing fails with

```
/home/mlenders/Repositories/RIOT-OS/RIOT/dist/tools/bossa-1.9/bossac -p /dev/ttyACM0 -o 0x2000 -e -i -w -v -b -R /home/mlenders/Repositories/RIOT-OS/RIOT/examples/hello-world/bin/feather-m0/hello-world.bin

Device unsupported
make: *** [/home/mlenders/Repositories/RIOT-OS/RIOT/examples/hello-world/../../Makefile.include:796: flash] Error 1
```

#### Versions
Current master (a17ff53ecfd23ed14b02e68f1018bfc97c69db2f) and definitely since 2021.10. I tried earlier releases, but they require cherry-picking 8f10f22b1fe51e6aee041fbd9722, with cherry-picking I hit another merge-conflict wall in the 2020er releases, so I stopped there... I am pretty sure, that I was able to flash this board in early 2020 though.

dylad, 2023-03-03:
I don't own the board and I am not using bossac since quite some time.
But if you still have access to it, could you try:
`BOSSA_FLAGS_OPTS="-i -d" BOARD=feather-m0 make -C examples/hello-world flash`
This might help to figure out what's going here.

crasbe, 2025-03-06:
I have a Feather M0 Proto board now and can reproduce the issue:

```
~/RIOTstuff/riot-doxygen/RIOT/tests/sys/shell$ /home/cbuec/RIOTstuff/riot-doxygen/RIOT/dist/tools/bossa-1.9/bossac -p /dev/ttyACM0 -o 0x2000 -e -i -w -v -b -d -R /home/cbuec/RIOTstuff/riot-doxygen/RIOT/tests/sy
s/shell/bin/feather-m0/tests_shell.bin
Set binary mode
Send auto-baud
Set binary mode
No device found on /dev/ttyACM0
```

HOWEVER... my board comes with an UF2 bootloader. If you double tap the reset button, the typical bootloader Mass Storage window opens:
[image]

Perhaps your board is the same?

<hr/>

The switch to the new bootloader must've been somewhere around 2018. My proposal would be to change the standard bootloader for the Feather M0 to the newer UF2 bootloader and add a bit of information to the documentation on how to update to the newer UF2 bootloader: https://learn.adafruit.com/installing-circuitpython-on-samd21-boards/installing-the-uf2-bootloader

crasbe, 2025-03-07:
Okay apparently the UF2 bootloader should be BOSSA compatible.

The error message that I got is also related to the fact that I use `usbipd` with the WSL container. The USB device is "ejected" and not automatically reconnected to the WSL.

I'll have to check with an actual Linux system.

crasbe, 2025-03-07:
Okay I have a hypothesis what`s wrong now.

First of all, I can reproduce the issue after RIOT has been programmed for the first time:
```
~/riot-feather/RIOT/tests/sys/shell$ BOARD=feather-m0 make flash
Building application "tests_shell" for "feather-m0" with CPU "samd21".

"make" -C /home/chris/riot-feather/RIOT/pkg/cmsis/ 
...
"make" -C /home/chris/riot-feather/RIOT/sys/ztimer
   text	   data	    bss	    dec	    hex	filename
  25796	    176	   4892	  30864	   7890	/home/chris/riot-feather/RIOT/tests/sys/shell/bin/feather-m0/tests_shell.elf
stty -F /dev/ttyACM0 raw ispeed 1200 ospeed 1200 cs8 -cstopb ignpar eol 255 eof 255
sleep 2
/home/chris/riot-feather/RIOT/dist/tools/bossa-1.9/bossac -p /dev/ttyACM0 -o 0x2000 -e -i -w -v -b -R /home/chris/riot-feather/RIOT/tests/sys/shell/bin/feather-m0/tests_shell.bin

Device unsupported
make: *** [/home/chris/riot-feather/RIOT/tests/sys/shell/../../../Makefile.include:846: flash] Fehler 1
```

Double tapping the reset button to put it into the bootloader state makes it possible to flash:
```
~/riot-feather/RIOT/tests/sys/shell$ BOARD=feather-m0 make flash
Building application "tests_shell" for "feather-m0" with CPU "samd21".

"make" -C /home/chris/riot-feather/RIOT/pkg/cmsis/ 
...
"make" -C /home/chris/riot-feather/RIOT/sys/ztimer
   text	   data	    bss	    dec	    hex	filename
  25796	    176	   4892	  30864	   7890	/home/chris/riot-feather/RIOT/tests/sys/shell/bin/feather-m0/tests_shell.elf
stty -F /dev/ttyACM0 raw ispeed 1200 ospeed 1200 cs8 -cstopb ignpar eol 255 eof 255
sleep 2
/home/chris/riot-feather/RIOT/dist/tools/bossa-1.9/bossac -p /dev/ttyACM0 -o 0x2000 -e -i -w -v -b -R /home/chris/riot-feather/RIOT/tests/sys/shell/bin/feather-m0/tests_shell.bin
Device       : ATSAMD21x18
Version      : v1.1 [Arduino:XYZ] Apr 10 2020 13:40:42
Address      : 0x0
Pages        : 4096
Page Size    : 64 bytes
Total Size   : 256KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
BOD          : true
BOR          : true
Erase flash

Done in 0.842 seconds
Write 25972 bytes to flash (406 pages)
[==============================] 100% (406/406 pages)
Done in 0.157 seconds
Verify 25972 bytes of flash
[==============================] 100% (406/406 pages)
Verify successful
Done in 0.246 seconds
Set boot flash true
```

This means that the bossa compatibility mode is generally working. When the magic bytes are sent via USB, the board resets but it seems to return to the application. This means, that the magic number put into the SRAM is incorrect and the bootloader won't stay in upload mode. This is the file in question: https://github.com/RIOT-OS/RIOT/blob/master/boards/common/samdx1-arduino-bootloader/reset.c

What's interesting here is that there is a define for `BOOTLOADER_UF2`, which is *not* set for the Feather-M0.
https://github.com/RIOT-OS/RIOT/blob/999220c493decf6b94f07dde6fdbe01b69fcd2a2/boards/common/samdx1-arduino-bootloader/reset.c#L35-L41

Flashing the Feather with that define set (the first time you have to do double tap the reset button of course) does work:
```
~/riot-feather/RIOT/tests/sys/shell$ CFLAGS+=-DBOOTLOADER_UF2 BOARD=feather-m0 make flash
Building application "tests_shell" for "feather-m0" with CPU "samd21".

"make" -C /home/chris/riot-feather/RIOT/pkg/cmsis/ 
...
"make" -C /home/chris/riot-feather/RIOT/sys/ztimer
   text	   data	    bss	    dec	    hex	filename
  25796	    176	   4892	  30864	   7890	/home/chris/riot-feather/RIOT/tests/sys/shell/bin/feather-m0/tests_shell.elf
stty -F /dev/ttyACM0 raw ispeed 1200 ospeed 1200 cs8 -cstopb ignpar eol 255 eof 255
stty: /dev/ttyACM0: Eingabe-/Ausgabefehler
make: [/home/chris/riot-feather/RIOT/tests/sys/shell/../../../Makefile.include:852: preflash] Fehler 1 (ignoriert)
sleep 2
/home/chris/riot-feather/RIOT/dist/tools/bossa-1.9/bossac -p /dev/ttyACM0 -o 0x2000 -e -i -w -v -b -R /home/chris/riot-feather/RIOT/tests/sys/shell/bin/feather-m0/tests_shell.bin
Device       : ATSAMD21x18
Version      : v1.1 [Arduino:XYZ] Apr 10 2020 13:40:42
Address      : 0x0
Pages        : 4096
Page Size    : 64 bytes
Total Size   : 256KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
BOD          : true
BOR          : true
Erase flash

Done in 0.842 seconds
Write 25972 bytes to flash (406 pages)
[==============================] 100% (406/406 pages)
Done in 0.156 seconds
Verify 25972 bytes of flash
[==============================] 100% (406/406 pages)
Verify successful
Done in 0.273 seconds
Set boot flash true
```

<hr/>

That means we have a clear path now, the `boards/feather-m0/Makefile.include` just has to have the `BOOTLOADER_UF2` flag set and the documentation has to be extended to mention that that option should be disabled for old boards.
Other boards like the Adafruit Grand Central M4 already has the flag set.

closed 2025-03-07 by miri64